springfall2008/batpred #1666: Multiple warnings: "Warn: Out of range index 1 within item"

discodancerstu, 2024-11-27:
Predbat displaying multiple warnings.

System is 2 x Hybrid Inverters, 2 x batteries and GivTCP 3.

Predbat log and apps.yaml as requested attached.
[predbat.log](https://github.com/user-attachments/files/17934114/predbat.log)
[appsyaml.txt](https://github.com/user-attachments/files/17934127/appsyaml.txt)

springfall2008, 2024-11-30:
**battery_scaling** and **inverter_time** should be set in apps.yaml one per inverter.

You should also comment back in the other settings below this comment which is actually now incorrect as some are used for charge curve calculation:

```yaml
  # If not using REST then instead set the Control here (one for each inverter)
  # - you can delete this section if using REST
  ...
```

The warning reducing idle time and device_id can be ignore I think that's a bug on my side

discodancerstu, 2024-11-30:
Thanks Trefor, I can't see how to set *battery_scaling* and *inverter_time*
for each inverter, please can you advise?

Also to confirm I should have this section as per below?

[image: image.png]

discodancerstu, 2024-11-30:
[image]

discodancerstu, 2024-11-30:
Since doing this, the following warnings and errors are now showing. 
[image]

gcoan, 2024-11-30:
You've got badly formatted YAML

If you look at your apps.yaml, each entry should be indented by two spaces

soc_max and reserve are both indented correctly and then everything after that is missing a space

see https://springfall2008.github.io/batpred/apps-yaml/#warning-appsyaml-file-format

closed 2024-12-15 by springfall2008