springfall2008/batpred #1668: Predbat car charging slot does not turn off for higher rates

thewhale21, 2024-11-28:
**Describe the bug**
Predbat car charging slot does not turn off for higher rates 

**Expected behaviour**
when higher rate slot is found, Predbat car charging slot should turn to off

**Predbat version**

8.7.2

**Environment details**

- Inverter and battery setup
- Giv Energy Hybrid with 2 batteries
- Standard HAOS installer or Docker
- HAOS
- Anything else?

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

[image]
[image]

**Log file**
https://pastebin.com/hDLWM3zx

springfall2008, 2024-11-30:
I'm afraid the logfile is incomplete, do you have the log for one complete cycle at 22:30?

thewhale21, 2024-12-02:
Can be closed, After going through logs i can see that predbat thought my octo cosy day rate is a low rate slot so decided to charge the car. Will alter low rate threshold

thewhale21, 2024-12-06:
@springfall2008 Not sure if this is intended behaviour, but it turns out that if car charging now is set and true, it will keep the car charging through the high rate slot and overrides the scheduled charge window. 

I would expect that if car charging now is true, planned charging slot can still be false. as otherwise this will create a feedback loop. and gives the behavior i experienced.

closed 2024-12-06 by thewhale21