springfall2008/batpred #1671: Automated Re-sync of Givenergy Inverter time

AGLAISTER, 2024-11-29:
Since 3 days ago I've had an issue where sensor.givtcp_ce2235g284_invertor_time was getting out of sync with HA system time eventually causing predbat to detect the error where the time was that far out of sync

What I did was set up an automation that is triggered on an update of sensor.givtcp_ce2235g284_invertor_time and then detects when sensor.givtcp_ce2235g284_invertor_time was +/- 2 minutes out of sync with HA time

**Describe the solution you'd like**
Add the value template to predbat for Givenergy systems to detect the difference and then trigger turning on of switch.givtcp_ce2235g284_sync_time

A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
I created this automation 

Automation YAML

alias: Check Inverter Time Offset
description: >-
  Triggers when the inverter time differs from the current time by more than 2
  minutes.
triggers:
  - entity_id: sensor.givtcp_ce2235g284_invertor_time
    trigger: state
conditions:
  - condition: template
    value_template: >
      {% set inverter_time = states('sensor.givtcp_ce2235g284_invertor_time') |
      as_datetime %} {% set current_time = states('sensor.time') | as_datetime
      %} {% if inverter_time is not none and current_time is not none %}
        {{ (inverter_time - current_time).total_seconds() | abs > 120 }}
      {% else %}
        false
      {% endif %}
actions:
  - data:
      message: |
        The inverter time is out of sync by more than 2 minutes.
    action: notify.notify
  - type: turn_on
    device_id: 47656f2c4b843dd8bb9363ddf89dd36f
    entity_id: 130020cb1a3dfaff6996ea21165a8315
    domain: switch
mode: single

**Additional context**
This required GivTCP To be at least v3.0.4 as earlier versions do not have the sync switch

gcoan, 2024-11-30:
I do not think this is a good idea and not convinced this is needed.

Why is your inverter time getting out of sync with your HA time?

To me this points to either an underlying problem with your inverter clock that is faulty, or (more likely), an intermittent network problem with GivTCP not being able to talk to your inverter reliably, and would normally show up in the GivTCP log.

My inverter is maybe only ever 20-30 seconds adrift from GivTCP.  Occasionally it does get out of step due to a GivTCP comms issue and automatically Predbat restarts GivTCP (the commands are in the template apps.yaml) and this fixes the comms issue.

AGLAISTER, 2024-11-30:
I have raised a support request but so far nothing had been
uncovered/determined

When you say you don't think this is a good idea can you elaborate why you
think that?

I have an automation doing this every 30 minutes at the moment and having
seen the speak to the geek video on memory wear I would like to know if
this had anything to do with your view?

Thanks Alan

gcoan, 2024-11-30:
> When you say you don't think this is a good idea can you elaborate why you think that?

I'm saying that I don't think re syncing the time is a good idea because it's covering up an underlying problem.

The inverter clock shouldn't drift so this automation shouldn't be necessary.

I can't remember when I last manually synchronised the inverter time from the GivEnergy portal.  Maybe when the GMT/BST clock change routine runs this forces a clock sync (I don't know), but I just checked my two inverters and  one is about 8 seconds slow and the other is about 10 seconds slow.  My experience is that the inverters keep accurate time.

AGLAISTER, 2024-11-30:
> experience is that the inverters keep accurate time so if yours doesn't
> then something else is wrong.

Thanks Geoffrey,

I do have an open support call with givenergy so hopefully it will be
sorted by them

I'm find for this to be closed

closed 2024-11-30 by springfall2008