springfall2008/batpred #1675: invertor time drift

AGLAISTER, 2024-11-30:
**Describe the bug**
sensor.givtcp_ce2235g284_invertor_time is drifting about 2 minutes per 30 minutes.
call/issue registered with Givenergy with Paul Landreagon and Mike smith who asked me to send these logs to you as well

I've updated from Givtcp v2.4.9 to v3.0.4 and added an automation to resync time which does get sensor.givtcp_ce2235g284_invertor_time back to system time but it then drifts out again

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Predbat version**
8.7.2 

**Environment details**

- Inverter and battery setup
Gen 1 inverter & 9.5kw battery
- Standard HAOS installer
- Givtcp v3.0.4
- Mqttbroker 6.4.1

**Screenshots**

[image]
[image]

As can be seen in the screen shots the sensor shows a 25 minute time difference but the portal shows the right time
[givtcplog.txt](https://github.com/user-attachments/files/17967174/givtcplog.txt)
[predbatdetaillog.log](https://github.com/user-attachments/files/17967175/predbatdetaillog.log)
[Inverter Register Record Data Download 2024-11-30.csv](https://github.com/user-attachments/files/17967176/Inverter.Register.Record.Data.Download.2024-11-30.csv)
[predbataddonlog.txt](https://github.com/user-attachments/files/17967177/predbataddonlog.txt)

2 days ago when I was getting a lot of errors and restarts of givtcp even when the clock was just 10 minutes out - but now it seems to be showing 25 minutes out but no restart of  givtcp

gcoan, 2024-12-01:
I don't think this is a a Predbat issue, if GivTCP is not reporting the inverter time correctly then it will be a GivTCP issue.

Was wondering if you had your poll frequency for GivTCP set too low, but looking at the logfile, GivtCP appears to be running every 30 seconds OK.

Suggest that you firstly delete all the PKL (cache) files in the GivTCP folder and restart GivTCP, and see if that resolves the issue.

If the inverter time in HA is not updating still then maybe there is a problem with GivTCP/MQTT.  You could try shutting down everything, HA and all the add-on's and then restarting, and see if that gets the control updating OK.

If that fails, shutdown GivTCP, uninstall MQTT to delete all the givtcp_ sensors in HA, shutdown and restart everything again, and re-install MQTT then restart GivTCP.  That should re-initialise the sensors in HA.

If that fails then I'm out of ideas and you'll have to ask the issue on the GivTCP github.   Have you tried reverting to GivTCP v2 and see if that has the same issue?

AGLAISTER, 2024-12-01:
Hi Geoffrey,

Thanks for getting back to me

I also suspect it's more than likely a givtcp issue.

It first showed up on v2.4.9 out of no where last week, although this may
have been after I updated ha core but I cannot remember for sure.

As the logs on the inverter seem to show the right time, predbat seems fine
and it seems mainly to be the givtcp inverter time going out I suspect
givtcp as well.

I'll try the things you suggest as soon as I can get time to for now I have
an automation handling the desync

Interestingly the sensor I created for the time difference shows a very
random pattern it seemed to have calmed down over night then a few
incidents where the automation resyncd the this evening it went crazy and
was resync every few minutes so I changed my automation only to run if the
clocks drifted by 5 minutes

The difference is time now - givtcp inverter time (in seconds)

I'd be interested in your thoughts on this

springfall2008, 2024-12-26:
Was this ticket the one that turned out to be a hardware issue?

AGLAISTER, 2024-12-26:
Hi Trefor,

HW is the prime suspect at the moment, what I can say is there is no
evidence I've found the issue is with predbat so this can be closed

Thanks Alan

closed 2024-12-26 by gcoan