springfall2008/batpred #1682: Issue with potential Excessive writes to the inverter

Timeflies1980, 2024-12-02:
**Describe the bug**
Multiple (excessive?) writes to the inverter

**Expected behaviour**
Less writes?

**Predbat version**

8.8.1
**Environment details**

- 3rd Gen 5Kw Givenergy Invter
- Standard HAOS install
- Anything else?

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)
[predbat.1.log](https://github.com/user-attachments/files/17978487/predbat.1.log)

springfall2008, 2024-12-03:
Potential fix on 'main' for testing

dandwhelan, 2024-12-03:
[Predbat-Github-0312.zip](https://github.com/user-attachments/files/17991113/Predbat-Github-0312.zip)
Last night I enabled Low Power mode to see if the write would lower, I think they went up if anything. Since then I've turned off Low Power and a 2nd optimise and it's been pretty steady, but loads might have been going on overnight. Uploaded the logs so you can compare numbers. TL:DR 143 Writes in 2 days.

Timeflies1980, 2024-12-03:
> Potential fix on 'main' for testing

Just updated to main.  Will report back.

springfall2008, 2024-12-04:
Low power mode is behaving a bit oddly, I'll add some more debug on it for my run tonight

dandwhelan, 2024-12-04:
Overnight using the main feed build
[image]
Without low mode on. Let me know if you want logs or to enable anything else overnight

Timeflies1980, 2024-12-04:
Hi All, An update after updating to Main.  It does seem to have improved slightly but still seeing around 100 writes overnight.  That said I do have quite a dynamic night most nights with lots of charging/load elements (Cars, Batteries, Dishwasher, washing machine, tumble, etc) so maybe this is to be expected.  Also as mentioned elsewhere this is still way be low a write every 5 mins over the the timeframe of a day.
[image]

springfall2008, 2024-12-04:
Discharge fix on 'main' for testing

mpartington, 2024-12-10:
This might be in the weeds, but I can't work out why in the middle of my charge window Predbat briefly decides to change to a freeze mode. This adds 6 additional writes as it changes back next run. Could be a bigger impact on other plans. Normal log attached

[image]

[predbatx.log](https://github.com/user-attachments/files/18079926/predbatx.log)

mpartington, 2024-12-14:
Same again at the same time.... What is special about 4.30?  i can't find any local automation and the night before it didn't glitch
[image]
[image]

No logs as debug fills then up too quickly, but I do have the debug files
[predbat_debug_04_30_00.txt](https://github.com/user-attachments/files/18135028/predbat_debug_04_30_00.txt)
[predbat_debug_04_35_00.txt](https://github.com/user-attachments/files/18135027/predbat_debug_04_35_00.txt)
[predbat_debug_04_40_00.txt](https://github.com/user-attachments/files/18135026/predbat_debug_04_40_00.txt)