springfall2008/batpred #1684: Octopus all_rates attribute warning

zaheerm, 2024-12-03:
**Describe the bug**

Log shows:

```
2024-12-03 09:30:04.507396: Warn: No Octopus data in sensor sensor.electricity_<meter>_previous_rate attribute 'all_rates'
2024-12-03 09:30:04.507507: Warn: No Octopus data in sensor event.electricity_<meter>_current_day_rates attribute 'all_rates' / 'rates' / 'raw_today'
2024-12-03 09:30:04.507691: Error: metric_octopus_import is not set correctly or no energy rates can be read
2024-12-03 09:30:04.547027: Warn: record_status Error: metric_octopus_import not set correctly or no energy rates can be read
2024-12-03 09:30:04.547354: Error: Exception raised 
2024-12-03 09:30:04.547909: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 959, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/predbat.py", line 529, in update_pred
    self.fetch_sensor_data()
  File "/config/fetch.py", line 825, in fetch_sensor_data
    raise ValueError
ValueError
```

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Predbat version**

Latest (zip file shows 8.8.1)

**Environment details**

- GivTCP inverter and battery
- Standard HAOS installer 
- Octopus integration latest version

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

gcoan, 2024-12-03:
Is this a new install or an upgrade?

Are the Octopus sensors that Predbat is looking for populated correctly and what's in your apps.yaml, is Predbat configured correctly to find the Octopus sensors?

If its a new install, you follow the instructions about enabling specific events that Predbat needs

zaheerm, 2024-12-03:
[image]
is a screenshot of my home-assistant current_day_rates event sensor.

gcoan, 2024-12-03:
Thanks for the extra info.  The apps.yaml looks correct. 

If you look at the sensor sensor.octopus_energy_electricity_MPAN_METERSERIAL_current_rate what do you see, is it populated OK?

Can you share your logfile please, am guessing it's either a problem with not picking up your Octopus details properly.

gcoan, 2024-12-03:
> is a screenshot of my home-assistant current_day_rates event sensor.

Looks like the event is firing OK.  Let's see what's in the actual import sensor

zaheerm, 2024-12-03:
https://gist.github.com/zaheerm/f550ffb3f766ce2eec4855faee9
[image]
c8c53 is the log file

zaheerm, 2024-12-03:
[image]
when expanding attributes

springfall2008, 2024-12-03:
There's something wrong here as the Event shown doesn't have the mpan in its name?

gcoan, 2024-12-03:
I think the problem is that your import energy sensor is named in a way that Predbat doesn't expect.

From the logfile it finds your import sensor:

2024-12-03 09:02:10.818563: Regular expression argument metric_octopus_import matched ^(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_current_rate)$ with sensor.electricity_METER_NUMBER_current_rate

Whereas on mine the sensor name is sensor.octopus_energy_electricity_METER_NUMBER_current_rate

Predbat uses the import sensor to then find the names of other sensors it needs, and it's not finding them.  Guessing that there is a mis-match and the other sensors have the octopus_energy_ prefix whereas current_rate doesn't.

Did you rename this sensor?

Simplest thing to try is to rename the sensor you have to the name it should normally be.  If you want to rename it then you'll also need to rename the other sensors that Predbat is looking for so they all have matching name structure

zaheerm, 2024-12-03:
Good spot, All my sensors seem to be sensor.electricty_ while my events seem to be event.octopus_energy_electricity. I may set the sensors and events manually in predbat config and give that a go rather than doing renames.

gcoan, 2024-12-04:
👍

I think that's your problem.  Without checking the code I think Predbat expects the events and sensors to be named consistently.  You may have to do some renaming.

Curious as to why your new install ended up like this.  Maybe something changed in the Octopus Integration

zaheerm, 2024-12-07:
Thanks @gcoan I ended up renaming the sensors to make them consistent once i spotted nothing else was using them. This solved my issue, shall I close?

gcoan, 2024-12-07:
@zaheerm glad you've got it working OK.  Weird that your entities had different names.

Are you seeing any warnings in the predbat log about all_rates not existing on the sensor still?

There's another issue that has been raised #1699 about all_rates warning, which from what I can see doesn't exist as an attribute so wondered if you were seeing the same warning as well?

At least we got yours working now

zaheerm, 2024-12-07:
I am not seeing any more warnings at all. I saw that issue come in and looked for any similar patterns here but don't see it.

gcoan, 2024-12-07:
Ok thanks for confirming, sounds like can close this issue now then

closed 2024-12-07 by zaheerm