springfall2008/batpred #1686: after update to 8.8.1 frequent charging for 5-15 minutes during the day at expensive rates and soc above 80%

eddysteurs, 2024-12-04:
**Describe the bug**
A clear and concise description of what the bug is.
since the upgrade to 8.8.1 i see very frequent "charging" for 5 to 15 minutes during the day and this on expensive rates, whil the battery is still almost full (>80%) and there are cheaper time slots later in the day
**Expected behaviour**
A clear and concise description of what you expected to happen.

**Predbat version**

8.8.1

**Environment details**

- Inverter and battery setup:   solis via soliscloud
- Standard HAOS installer or Docker : addon
- Anything else?
using nordpool pricing for Belgium
**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.
[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)
[predbat log 20241204.txt](https://github.com/user-attachments/files/18005798/predbat.log.20241204.txt)

springfall2008, 2024-12-04:
I don't quite understand the behaviour but I'd hazard a guess that reducing metric keep to something lower e.g. 1 kWh will probably workaround this

eddysteurs, 2024-12-05:
Which metric specifically you mean?

            Eddy Steurs
            エディスタルス
            Asterlaan 16, 2950 Kapellen
             Phone +32 473 630 377
             Email: ***@***.******@***.***>

springfall2008, 2024-12-05:
Sorry this one

[image]

eddysteurs, 2024-12-05:
I will raise it from 3 to 4 but sure if it will help
Reason for being already at 3 is that the solis soc is at 2,3, so i rounded the soft Soc to 3
Eddy Steurs

image.png (view on web)<https://github.com/user-attachments/assets/26e8aa45-46c0-42ba-b9a2-4ccbdb0127ea>

springfall2008, 2025-01-01:
If this is still an issue when you see a bad plan please grab the debug YAML and attach it here

closed 2026-08-20 by chalfontchubby