springfall2008/batpred #1688: Solaredge hold for car and freeze charge/discharge

PieBeard, 2024-12-04:
**Is your feature request related to a problem? Please describe.**
With a SolarEdge system when Predbat plans to hold charge for car what actually occurs is that the battery is discharged into the car and then as the SOC drops the plan reevaluates and charges the battery more increasing cycles on the battery.

**Describe the solution you'd like**
Whilst trying to work out some automations and for a backup to Predbat not charging properly related to Issue #1632 I've found that you can prevent battery discharge by selecting other options from the entity select.solaredge_i1_storage_command_mode (mine is called select.solaredge_multi_i1_storage_command_mode)

by selecting: 
"Solar Power Only (off)" battery charge and discharge is turned off and the house will run of grid or solar power
"Charge from Clipped Solar Power" and the battery will only charge from excess solar above the inverters capacity this seems to draw power for the house from the grid with about 40w coming from the battery, nothing much really.
"Charge from Solar Power" is any excess solar the house isn't using is used to charge the battery and when house load is above Solar power it draws from the grid with only about 40w coming from the battery again.

So in essence the Hold for Car could select any of these and they could be used for the equivalent of freeze charging too.

Just a note that in SolarEdge Systems there is a default mode which can be set and is usually maximise self consumption.  The timeout for this is usually set at 3600s from when the service selection changes the storage_command_mode.  Given that the SolarEdge Modbus Multi is a local connection there is no risk of excess API calls and so service calls at every slot even if there is no change in what it is doing helps prevent this automatic reset to the default mode.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

LupoLoopy, 2024-12-08:
Ties into notes here on 'unexpected' behaviours with SolarEdge:

https://www.facebook.com/groups/1477599886299106/posts/1621646515227775/?__cft__[0]=AZUsGEczym4iv6XCRDLyvY4uHq48rdg8MME3u3bHdxPDYOududt043NssxEAb4P2jY6ULYjuF8EPuC6QrHj98yhiqtibQttFGDyFZOfIAfb1c2p2IZnuzgGI6vGg3hpKTxPlXRSmdC_AkdM5tnQl8_fsikf_zuBPdlIzGFT3s6ZgV2Jof07t_1tHFSK0QtrxvWg&__tn__=%2CO%2CP-R

cyberkryten, 2024-12-11:
I have had the charge issue recently.  

However, Predbat has been working fine for months manipulating the battery charge/discharge rates to achieve the hold/freeze states, so I don't think anything needs to be changed in the control modes on SolarEdge to achieve the desired functionality.  It can already stop the battery discharging by setting the discharge rate to zero.

Also, I tried some of the above modes when I got the inverter (early 2024) whilst trying to write my own system (before I found Predbat) and they behaved oddly at times, though I have an 8kW inverter and 3.68kW export limit so had specific use cases.  I can't remember the exact details, but it's worth going through the SolarEdge PDFs which have more detail because the App descriptions don't quite match what it does (IIRC).

Is your system resetting to the default?  I've found Predbat correctly controls the Storage Command Mode and updates it so that a revert to the default doesn't happen.  The exception is the times when the inverter resets itself (was once per day, now twice per day since the September firmware update) as when it comes back online, there's no 'last mode' option and it comes back in the Default Mode, but Predbat then spots this and sets the correct mode again.

closed 2026-08-20 by chalfontchubby