springfall2008/batpred #1690: Hold charge not holding charge

chris-y, 2024-12-05:
**Describe the bug**
When Predbat puts a "hold charge" into the schedule the battery actually discharges 

**Expected behaviour**
Hold charge should ensure the battery doesn't discharge. I'd expect this to be a charge at 0A for my custom inverter.

**Predbat version**
8.8.1

**Environment details**

- Solis inverter set up as a custom for control

**Screenshots**
I didn't screenshot the plan in advance as I didn't know I needed to. Octopus Intelligent had my car scheduled overnight and Predbat had these slots as hold charge at 100%.

Checking back in the script history, during hold charge, the inverter was not set to do anything: 

```
        chargeCurrent: '072'
        dischargeCurrent: '00'
        chargeStartTime: '00:00'
        chargeEndTime: '00:00'
        dischargeStartTime: '00:00'
        dischargeEndTime: '00:00'
```

I would have expected the times to be set, and the charge current to be at 0A.

This is the graph where you can see the car has ended up charging from the battery, causing the battery to not charge up to 100% for the end of the off peak slot: (the pink line is the battery level)

[image]

**Log file**
I can't scroll back the logs far enough

chris-y, 2024-12-06:
This is the sort of thing I'm seeing in the plan - car set to charge after the house battery charges, then it drains it rather than holding charge (in this plan it appears it hasn't even considered doing a hold charge when the car is charging, so it'll definitely drain)

[image]

chris-y, 2024-12-07:
I'm not sure if this is the same issue or different. 

My plan looks like this, so it is supposed to be charging: 

[image]

However it has set the start and ends times to midnight (ie. don't charge):

[image]

You can see the start and ends times are 00:00 for a bit around the time when this service was called: 

[image]

The relevant part of the log is here. It looks like it maybe called the service before setting the times?

[image]

springfall2008, 2024-12-07:
Are you saying the call to the service should be after adjusting the start/end times not before?

chris-y, 2024-12-07:
> Are you saying the call to the service should be after adjusting the start/end times not before?

Yes.

springfall2008, 2024-12-07:
Fix on main for testing

chris-y, 2024-12-07:
As far as I can tell, that's fixed the order issue, thanks! 

Whether it has fixed the charge hold problem may take a while to figure out. I might have to force a charge hold and see what happens.

Log:

 15993	2024-12-07 23:30:20.300763: Inverter 0 Calling service charge_start_service domain charge with data {'device_id': '', 'target_soc': 19, 'power': 3600}
15992	2024-12-07 23:30:20.300667: Inverter 0 Skipping service discharge_stop_service domain discharge with data {'device_id': ''} as already called
15991	2024-12-07 23:30:20.300377: Inverter 0 Wrote 23:59:00 to idle_end_time, successfully now 23:59:00
15990	2024-12-07 23:30:18.239107: Inverter 0 set new demand (idle) end time to 23:59:00 was 23:30:00
15989	2024-12-07 23:30:18.238933: Inverter 0 Wrote 03:30:00 to idle_start_time, successfully now 03:30:00
15988	2024-12-07 23:30:16.171496: Inverter 0 set new demand (idle) start time to 03:30:00 was 08:00:00
15987	2024-12-07 23:30:16.171347: Adjust demand (idle) time computed is 03:30:00-23:59:00
15986	2024-12-07 23:30:16.170846: Adjust idle time, charge 23:30:00-03:30:00 discharge 00:00:00-00:00:00
15985	2024-12-07 23:30:16.170388: Inverter 0 Wrote 8.0 to timed_charge_current, successfully now 8.0
15984	2024-12-07 23:30:14.108640: Inverter 0 Wrote 400 to charge_rate, successfully now 400
15983	2024-12-07 23:30:12.047852: Inverter 0 current charge rate is 3600.0W and new target is 400W
15982	2024-12-07 23:30:12.047620: Low Power mode: minutes left: 230 absolute: 240 SOC: 0.96 Target SOC: 1.82 Charge left: 0.86 Max rate: 3600.0 Min rate: 215.0 Best rate: 400.0
15981	2024-12-07 23:30:12.044808: Charge window will be: 2024-12-07 23:30:00+00:00 - 2024-12-08 03:30:00+00:00 - current soc 10 target 19
15980	2024-12-07 23:30:12.044522: Include original charge start 12-07 23:30:00, keeping this instead of new start 12-07 23:30:00

chris-y, 2024-12-08:
I forced a charge freeze and it seems to have started charging at 74A

chargeCurrent: '074'
        dischargeCurrent: '00'
        chargeStartTime: '23:30'
        chargeEndTime: '03:30'
        dischargeStartTime: '00:00'
        dischargeEndTime: '00:00'

23:40:08 to 23:40:18 the Amps sensor was set to 74
23:40:18 it was set to 0

[image]

The service was called at 23:40:14

[image]

So it looks like there's another ordering issue here. 

My Amps sensor is set in the config as such: 
```
timed_charge_current:
    - input_number.solis_battery_charge_rate_predbat
  timed_discharge_current:
    - input_number.solis_battery_discharge_rate_predbat
```

This is the bit of the log relating to those two times: (sorry it's a bit cut off, let me know if you need any more - I have the full log saved)
```
2024-12-07 23:40:10.805029: Inverter 0 Wrote 74.0 to timed_charge_current, success>2024-12-07 23:40:10.805453: Adjust idle time, charge 23:30:00-03:30:00 discharge 0>2024-12-07 23:40:10.805858: Adjust demand (idle) time computed is 03:30:00-23:59:002024-12-07 23:40:10.806036: Inverter 0 current discharge rate is 3600.0W and new t>2024-12-07 23:40:12.867877: Inverter 0 Wrote 0 to discharge_rate, successfully now>2024-12-07 23:40:14.939534: Inverter 0 Wrote 0.0 to timed_discharge_current, succe>2024-12-07 23:40:14.939695: Freeze charging with soc 11%                           2024-12-07 23:40:14.939843: Inverter 0 Skipping service discharge_stop_service dom>2024-12-07 23:40:14.939928: Inverter 0 Calling service charge_start_service domain>2024-12-07 23:40:16.104789: No export window planned                               2024-12-07 23:40:16.105247: Inverter 0 Adjust force export to False, change times >2024-12-07 23:40:16.105354: Adjust idle time, charge 23:30:00-03:30:00 discharge 0>2024-12-07 23:40:16.105728: Adjust demand (idle) time computed is 03:30:00-23:59:002024-12-07 23:40:16.105924: Car charging from battery is off, next slot for car 0 >2024-12-07 23:40:16.106012: Inverter 0 Skipping service discharge_stop_service dom>2024-12-07 23:40:16.106061: Within charge freeze setting target soc to current soc>2024-12-07 23:40:16.106126: Inverter 0 adjust target soc for hold to 11% based on >2024-12-07 23:40:16.106195: Inverter 0 Current charge limit is 19% and new target >2024-12-07 23:40:18.180206: Inverter 0 Wrote 11 to charge_limit, successfully now >2024-12-07 23:40:20.242190: Inverter 0 Wrote 0.0 to timed_charge_current, successf>2024-12-07 23:40:20.242354: Current SOC 11% is same as Target SOC 11. Grid Charge >2024-12-07 23:40:20.242420: Inverter 0 count register writes 3                     2024-12-07 23:40:20.398033: Total inverter register writes now 853                 2024-12-07 23:40:20.585663: Completed run status Freeze charging                   2024-12-07 23:40:20.657748: Info: record_status Freeze charging target 11%         2024-12-07 23:40:20.698515: Saved current settings to /addon_configs/6adb4f0d_pred
```

chalfontchubby, 2026-08-20:
Thanks for the report. This issue has been open for over a year with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still a problem for you, please let us know and we'll keep it open. Otherwise, we'll close it in a week (around 2026-08-28).

raddzz, 2026-08-21:
I believe this is still active, with a solis inverter it requires the charge/discharge times to be written as well as the current setting to 0 to freeze charge/discharge. As far as I can tell predbat only writes the current.

springfall2008, 2026-10-29:
Closing this one. It was reported on 8.8.1 with a custom Solis setup, and the order-of-writes problem found in the thread was fixed at the time. Logs were requested on 21 Aug to check whether the remaining hold-charge behaviour still happens, and none have arrived. A great deal has changed in the Solis support since then.

@raddzz, if you still see a hold charge (freeze charge) not holding on a Solis, please open a new issue with a full log file, `predbat_debug.yaml` and `apps.yaml` (all downloadable from the Predbat web page), and we'll pick it up from there.

closed 2026-10-29 by springfall2008