springfall2008/batpred #1692: Time till iboost starts

mogons57, 2024-12-05:
**Is your feature request related to a problem? Please describe.**
I have iboost setup to switch on (and heat water) when the electricity prices are low (which works well) but it would be nice to be able to spot when this is going to happen and then use an automation to switch off the boiler in anticipation, so that the best use could be made of the cheaper electricity. Often I'm finding that my hot water tank is already quite full when the iboost switches on so I'm not making the most of the cheap power. 

**Describe the solution you'd like**
A "time_till_iboost" sensor that says how long it is till an iboost session starts. For example if it is currently 18:00 and predbat has decided that iboost should start at 12:30 the "time_till_iboost" sensor should be 04:30 (i.e. 4 hours and 30 mins). If no iboost session is coming then the sensor could be set to a large number (say 48 hours) or set to unavailable (if that is possible).

A test could then be used in an automation along the lines of 
if (time_till_iboost < 3 hours) then switch off boiler
boiler would then switch back on when iboost session stops

**Describe alternatives you've considered**
I've though about trying to get this out of the predbat plan but could not work out how to do that.

**Additional context**
An example of what time_till_iboost might look like on the run up to and during some iboosting 

time     iboost    time_till_iboost 
21:00    -            1:30    
21:30    -            1:00
22:00    -            0:30
22:30    1.5         0
23:00    1.5         0
23:30    -            0:30
00:00    1.5         0
00:30    -            1:00
01:00    -            0:30
01:30    1.5         0
02:00    -            48:00 / unavailable
02:30    -            48:00 / unavailable

gcoan, 2024-12-05:
I think you should be able to get the data out of the existing predbat.iboost_best sensor https://springfall2008.github.io/batpred/output-data/#iboost-solar-diverter-data which has the times of the planned iboost in its attributes

mogons57, 2024-12-06:
In case anyone else wants to do this, the code below may not be that efficient but seems to work:

```
{% set iboost_times = state_attr("predbat.iboost_best","results") %}
{% set times =  iboost_times.keys()|list %}
{% set iboost_energy =  iboost_times.values()|list  %}
{% set ni = namespace(x=0) %}
{% set data = namespace(h_bool=False) %}
{% set iboost_starts = "" %}
{% for ni in range(0,times|count()-1) if data.h_bool == false  %}
{## {{ times[ni]}} {{ iboost_energy[ni]}}, {{iboost_energy[ni+1]-iboost_energy[ni]}} ##}
{% if iboost_energy[ni+1]-iboost_energy[ni] > 0 %}
{% set data.h_bool = true %}{{ (as_timestamp (times[ni])-as_timestamp (now()) ) / 3600 | round (0) }}
{% else %}
{% endif %}
{% endfor %}
{% if data.h_bool == false %}
{{100}}
{% endif %}
```

I've used this as a template sensor to calculate the time till the next iboost starts. If no iboost is imminent then it returns 100h and if currently boosting it will produce a small negative answer.

I'm then testing if this value is less than (say) 4 hours and if so telling the boiler not to fire up (even if it normally would).

closed 2024-12-06 by mogons57

gcoan, 2024-12-06:
@mogons57 thanks for the template code, I'll add it to the documentation to help others out

I notice there is a commented out line in the middle of the template code, can this be safely removed?

mogons57, 2024-12-06:
Yes that can come out, was just used during  development.