springfall2008/batpred #1698: install issue -  not resolve load_today value sensor.givtcp_{geserial}_load_energy_today_kwh

dicconel, 2024-12-06:
First install won't initialise with error -  not resolve load_today value sensor.givtcp_{geserial}_load_energy_today_kwh

**Predbat version** v8.8.2

- GivAIO with gateway and string inverter (growatt)
- HAOS running in Proxmox on dell optiplex 3040

[predbat.log](https://github.com/user-attachments/files/18038466/predbat.log)

dicconel, 2024-12-06:
[prdbat_app_yaml.txt](https://github.com/user-attachments/files/18038541/prdbat_app_yaml.txt)

Yaml file in its current state.

gcoan, 2024-12-06:
@dicconel do you have a single AIO or multiple?

Your problem is with the setting of geserial, which isn't set currently in apps.yaml.  Predbat uses this to find all the rest of the sensor names.

If you have a single AIO then the template code that is currently commented out in apps.yaml should find your inverter serial number correctly and geserial2 is not needed to be set.

If you have multiple AIO's then please follow the install instructions https://springfall2008.github.io/batpred/inverter-setup/#givenergy-with-givtcp and in particular that you need to set geserial to your AIO gateway serial number.  geserial2 is again not needed

dicconel, 2024-12-08:
@gcoan 

one AIO.  

I tried the geserial/geserial2 with/without the manual/auto serial number to no dice. 

However, your link to setting up the invertor did help!  I had changed the name to a more friendly one but not updated that in the prdbat yaml.

However, it's now throwing a "record_status Error: Exception raised can only concatenate str (not "int") to str" error.

gcoan, 2024-12-08:
@dicconel yes if you change the inverter prefix in givtcp then it won't find it.

Can you share the top bit of your apps.yaml where you have the geserial defined and some of the sensors defined, and a fragment of your predbat logfile so we can see the actual failure.  It still sounds like you have a mismatch in your apps.yaml

If you have a single AIO then geserial should be configured to point to your AIO and geserial2 is not needed at all, all the geserial2 lines can be commented out of apps.yaml

dicconel, 2024-12-08:
@gcoan 

Spotted the error when I was looking back thro' the yaml before pasting it. I'd missed a leading 0 in battery scaling (ie .85 rather than 0.85) I think. Also, had missed changing the sensor name in REST so that appeared once I'd fixed the str error but all seems to be working now anyway. At least in appearance. 

Now to dive into the settings for getting my optimisation optimised! 

@springfall2008 @gcoan Thanks for all the help and for hand holding a newbie!

closed 2024-12-08 by dicconel

gcoan, 2024-12-08:
glad you got sorted.  Enjoy Predbat !