springfall2008/batpred #1699: No Octopus data in sensor sensor.octopus_energy_electricity_XXXXXX_YYYYYY_previous_rate attribute 'all_rates'

LupoLoopy, 2024-12-06:
**Describe the bug**
Repeated warning in the log as follows:
2024-12-06 14:15:09.348795: Warn: No Octopus data in sensor sensor.octopus_energy_electricity_XXX_YYY_previous_rate attribute 'all_rates'

Sensor name is correct but no such attribute exists.

**Expected behaviour**
For Predbat not to plumb a sensor for a non-existent attribute, assuming this isn't an isolated issue with my Octopus integration install.
**Predbat version**

8.8.2. Also happened on 8.8.1. Not sure when it started.

**Environment details**

- Solar Edge inverter and battery
- Standard HAOS install

**Screenshots**
Log file as described. Screenshot of sensor attached 
[image]

gcoan, 2024-12-06:
You are the second person to report this issue, see #1684 and the fix suggested there

It looks like maybe the Octopus integration has renamed some of the sensors?

LupoLoopy, 2024-12-06:
I did see that issue, but I am pretty certain mine is slightly different. My sensor is correctly named/discovered, as per the screenshot and the warning log line. I will go recheck the build guide to see if I've missed anything, as the other fault reporter did, but it's been working fine up until now.

gcoan, 2024-12-06:
Sorry your issue is different, I saw the all_rates comment and didn't read it properly, thought it was the same issue.

So you had Predbat working previously OK and now it isn't?

It looks like it is a Predbat bug.  I am running an older version of Predbat and an older version of the Octopus integration and I don't have 'all rates' attribute on this sensor either, so looks like Predbat is looking for something that does't exist

The rates are attributes of the event, not the sensor, and the attribute name is 'rates' not 'all rates'

LupoLoopy, 2024-12-07:
> So you had Predbat working previously OK and now it isn't?

It works, but the warning log is full of these errors. I supposed that it was an issue worth logging. Save my SSD some writes at least!

LupoLoopy, 2025-04-04:
Okay. I'm revisiting this, a long time after the fact, as I've had to sanitise some of the other SolarEdge vestigial configuration lines creating warnings in the log.

I still have this issue. Do others using the BottlecapDave Octopus integration have the 'all rates' attribute on their '..._previous_rate' entity?

gcoan, 2025-04-04:
Mine looks exactly the same as the screenshot in the opening entry on this issue 

No .all rates. attribute

closed 2026-08-29 by springfall2008