springfall2008/batpred #1700: Custom inverter start/end times set after service call

chris-y, 2024-12-07:
I'm not sure if this is the same issue or different to #1690

My plan looks like this, so it is supposed to be charging: 

[image]

However it has set the start and ends times to midnight (ie. don't charge):

[image]

You can see the start and ends times are 00:00 for a bit around the time when this service was called: 

[image]

The relevant part of the log is here. It looks like it maybe called the service before setting the times?

[image]

This zoomed in section of the end time seems to confirm this is the case: 
[image]

The time that was set was after the service was called.

springfall2008, 2024-12-07:
How is your service getting the start and end times as they are not passed in the service call?

chris-y, 2024-12-07:
It's picking them up from sensor.predbat_mine_0_charge_start_time and sensor.predbat_mine_0_charge_end_time, which Predbat sets

springfall2008, 2024-12-08:
I put a fix for this on 'main' for testing

chris-y, 2024-12-08:
That has fixed this issue, thanks!
I've put a note on the other issue #1690 about the current setting having a similar problem.

closed 2024-12-08 by springfall2008