springfall2008/batpred #1703: v8.8.2 appears to stop my battery exporting when rates are favourable.

samtate, 2024-12-08:
**Describe the bug**
Updating to v8.8.2 compared to v8.7.0 appears to stop my battery from exporting overnight when rates are favourable. I am on EON Next Drive which has 7 hours of 6.7p import overnight, and a constant 16.5p export. When my predbat version is 8.7.0, the battery charges from 4% to 100%  (10kWh) in about 3 hours and then spends the remaining 4 hours charging and discharging to export to the grid for arbitrage.

On v8.8.2, the battery simply charges to 100% and then is held there for the cheap period: no export takes place. This means my daily cost is actually more.

**Expected behaviour**
Predbat should use arbitrage when the import is much less than the export to generate extra income.

**Predbat version**

v8.8.2

**Environment details**

GivEnergy Inverter and 9.5kWh battery
Installed Predbat addon in Home Assistant

gcoan, 2024-12-08:
can you share log files and your predbat plan please to aid investigation. Ideally from both the old and new versions

springfall2008, 2024-12-08:
There's a new release (v8.8.3) where with debug_enable set True a debug YAML will be written to the 'debug' directory. If you can generate and attach the YAML together with an image of where the plan is bad I can reproduce it.

samtate, 2024-12-09:
Yaml and screenshots below
[image]
[image]
[image]

YAML: https://file.io/aFhycVs58Pys

samtate, 2024-12-09:
For comparison, the overnight period on v8.7.1 which is more what I'd expect: 
[image]

springfall2008, 2024-12-10:
This is because 'calculate_export_oncharge' is an Expert mode feature and the default has been changed to off, to enable it turn on expert_mode and then turn on this toggle.

gcoan, 2024-12-10:
It also looks like you have one large charge slot in the night.  Turn combine_charge_slots off to create a series of charge slots and then Predbat can use some of those for exporting as well 

https://springfall2008.github.io/batpred/customisation/#battery-margins-and-metrics-options

samtate, 2024-12-16:
Thanks both, I changed those settings and now it works perfectly again on recent versions

closed 2024-12-16 by samtate