springfall2008/batpred #1708: Export Slots On Intelligent Octopus Go - 8.8.3

PianSom, 2024-12-08:
**Describe the bug**
My plan is no longer allowing for its usual export. 

**Expected behaviour**
I would expect export in the middle of the night, and before tomorrow's Power-up. Charging also appears to be slower than expected (and slower than observed historically)

**Predbat version**
8.8.3

**Environment details**
AIO single
Docker setup for Predbat and HA

**Screenshots**
[image]

**Log file**
Changed from .yaml to .txt
[predbat_debug_21_10_00.txt](https://github.com/user-attachments/files/18054045/predbat_debug_21_10_00.txt)

springfall2008, 2024-12-09:
From what I can tell your plan is fine, there is no spare energy to export as you are running out of battery. That said you might want to turn off 'switch.predbat_combine_export_slots' to get the best export when you have spare battery.

However it looks like your battery_charge_power_curve is bad. I don't know how it got to be 0.36 for all values 1-85 did you set it or was it computed?

I would try copying this curve but deleting the values 85 and and below and that may fix your issue?

"battery_charge_power_curve:
  1: 0.36
  2: 0.36
  3: 0.36
  4: 0.36
  5: 0.36
  6: 0.36
  7: 0.36
  8: 0.36
  9: 0.36
  10: 0.36
  11: 0.36
  12: 0.36
  13: 0.36
  14: 0.36
  15: 0.36
  16: 0.36
  17: 0.36
  18: 0.36
  19: 0.36
  20: 0.36
  21: 0.36
  22: 0.36
  23: 0.36
  24: 0.36
  25: 0.36
  26: 0.36
  27: 0.36
  28: 0.36
  29: 0.36
  30: 0.36
  31: 0.36
  32: 0.36
  33: 0.36
  34: 0.36
  35: 0.36
  36: 0.36
  37: 0.36
  38: 0.36
  39: 0.36
  40: 0.36
  41: 0.36
  42: 0.36
  43: 0.36
  44: 0.36
  45: 0.36
  46: 0.36
  47: 0.36
  48: 0.36
  49: 0.36
  50: 0.36
  51: 0.36
  52: 0.36
  53: 0.36
  54: 0.36
  55: 0.36
  56: 0.36
  57: 0.36
  58: 0.36
  59: 0.36
  60: 0.36
  61: 0.36
  62: 0.36
  63: 0.36
  64: 0.36
  65: 0.36
  66: 0.36
  67: 0.36
  68: 0.36
  69: 0.36
  70: 0.36
  71: 0.36
  72: 0.36
  73: 0.36
  74: 0.36
  75: 0.36
  76: 0.36
  77: 0.36
  78: 0.36
  79: 0.36
  80: 0.36
  81: 0.36
  82: 0.36
  83: 0.36
  84: 0.36
  85: 0.36
  86: 1.0
  87: 0.94
  88: 0.92
  89: 0.88
  90: 0.88
  91: 0.84
  92: 0.79
  93: 0.66
  94: 0.6
  95: 0.49
  96: 0.42
  97: 0.36
  98: 0.24
  99: 0.21
  100: 0.21
"

[image]

PianSom, 2024-12-09:
Oh, excellent spot! Yes, a stuffed up charge curve would do it. Put it back to manual and all is normal, thanks!

As to how it happened - I have no idea. I have had it on `auto` for months now. Very odd.

PianSom, 2024-12-09:
PS I had switch.predbat_combine_export_slots off for a few months at the start of the year. While it may provide a better plan, I found (and find) that it produces multiple in-half-hour charge/discharge cycles. I cannot believe that it healthy for battery chemistry (not that I have any basis for that belief!) so prefer it `on`.

springfall2008, 2024-12-10:
Probably a bug in the discharge curve calculation somewhere, something I need to look into another time.

closed 2024-12-10 by PianSom