springfall2008/batpred #1709: High number of register writes on agile tariff

kernow, 2024-12-08:
**Describe the bug**
A large number of register writes are being performed when on the Octopus Agile tariff, approx 170 in 24 hours. I've linked tot he raw logfile below, Let me know if you would like to see more from the older logfile.

**Predbat version**

v8.8.2

**Environment details**

- Inverter and battery setup
Single Hybrid GivEnergy inverter, single 9.5 kwh battery

**Log file**
[Raw Logfile](https://gist.githubusercontent.com/kernow/c9902b705db9164408f96fbb0341b2ff/raw/2109f940f92b020eaf23e62ab0f0c3da4f1202c0/gistfile1.txt)

kernow, 2024-12-08:
I've updated to the latest version and generated a [plan for you here](https://gist.githubusercontent.com/kernow/c9902b705db9164408f96fbb0341b2ff/raw/03f90f265a81b90be9bc6348a5fce7547040064e/plan)

kernow, 2024-12-08:
I just realised this logfile probably doesn't go back far enough as there haven't been that many writes in the last 7 hours. I will see if I can find the logs from when there was a log of writes going on...

kernow, 2024-12-08:
@springfall2008 is there a way to get older logs or are they purged after a certain age? I can't see anything older that a few hours on my system.

Ian7027, 2024-12-09:
Morning,

Here's my log file from this morning, around 80 writes since midnight.  Hope this might help.

[Predbat_Agile_Writes_GE_AIO.log](https://github.com/user-attachments/files/18058817/Predbat_Agile_Writes_GE_AIO.log)

discodancerstu, 2024-12-09:
Hi, I'm also experiencing high number of writes, system is 2 G3 hybrid inverters, each with a 9.5 battery.
[predbat.log](https://github.com/user-attachments/files/18066573/predbat.log)

kernow, 2024-12-10:
I've got a better [logfile here with more writes](https://gist.githubusercontent.com/kernow/c9902b705db9164408f96fbb0341b2ff/raw/73699f73cf084c7f9bd54d604050e828e79c60f9/Another%2520log)

PhilDye, 2024-12-10:
I'm also seeing a higher-than-expected count, also on Agile; 
[image]

I see a **lot** of these entries, flipping back and forth with only 10s between them;

```
2024-12-10 09:20:53.239699: Inverter 0 Wrote scheduled_discharge_enable to True successfully and got on
2024-12-10 09:21:03.322458: Inverter 0 Wrote scheduled_discharge_enable to False successfully and got off
```

[predbat.1.log](https://github.com/user-attachments/files/18085215/predbat.1.log)

Inverter: All-In-One, via GivTCP v2.4.9
Predbat version: v8.8.4

SwiftRR, 2024-12-10:
I have an AC3.0 inverter with two 8.2 kWh batteries on Agile importing.
The number of writes depends so much an activity.

When Storm Darragh came through, I was getting a lot of daily counts up to 160. Predbat was regularly alternating between charging and exporting overnight to minimise costs.
But in more benign days, such as today, I have just 33. The inverter is mainly just charging at night and discharging under demand during the day. With forecast this week, I expect this to continue.
See chart below where my average is 56/day after 9 days. The chart shows the differences from day to day, with most writes on 2 days out of the 9.
Is this just expected behaviour?
 
[image]

Rob

springfall2008, 2025-01-01:
I've made some changes that may help with this, please let me know if 8.9.1 helps

kernow, 2025-01-02:
Upgraded to the newest version at 9pm yesterday and have had 42 writes in that time, about 15.5 hours. Not sure how many writes is considered low/acceptable? I'll keep an eye over a longer period and report back.

discodancerstu, 2025-01-05:
Hi Trefor,

I'm still seeing a really high number of writes, since 1 Jan 25 (over a
dual hybrid inverter setup) the daily average is 361.

gcoan, 2025-01-06:
bear in mind the inverter writes is across all inverters so you'd need to half that number

but it is still high.  I am on Agile as well with dual hybrid inverters and long term average is 64 a day per inverter (yesterday was 47)

can you share your predbat plan

discodancerstu, 2025-01-06:
See attached, is this what  you want?

I have dual G3 inverters and 2 batteries.  Number of writes is always very
high, significantly more than yours.

I don't run any automations to balance the batteries, it is entirely run by
PredBat.

gcoan, 2025-01-06:
Your predbat plan didn't arrive in github, can you try again please?

Wanted to see what predbat was planning and whether there was any chopping and changing that might cause the high number of inverter writes

Here is my daily inverter writes for both inverters for the last week, as you can see, generally quite a bit lower than yours:
[image]

you said you don't have any automations to balance the inverters, are you using predbat to balance them and prevent cross charging ?  Is switch switch.predbat_balance_inverters_enable turned on?

I stopped using predbat to balance the inverters some time before the inverter write issue was highlighted as I was uncomfortable with the number of commands it was sending to the inverters, could be every minute.  I just live with the cross charging that occurs although have made some suggestions of how to reduce it #1031

discodancerstu, 2025-01-07:
Good morning, PB plan as follows:
[image]

And settings:
[image]

How did you create a sensor with daily writes?

gcoan, 2025-01-07:
Your plan is none too exciting, but the cause of the high inverter writes is that you are using the balance inverter settings in Predbat.

What this setting does with two inverters is to change the charge rates on the two inverters to try to stop the cross charging from occurring.

I noticed this a while ago in github issue #1397 and that was part of why I stopped using the Predbat balance inverter settings.

For me I get most of my cross charging occurring during the evening when the batteries are in Demand mode, I have suggested on #1031 that if the batteries are set to Pause Charge when there's no sun then this stops a lot of the cross charging.

I'm going to add this to the documentation, but you can create a daily inverter write sensor by creating a utility meter in configuration.yaml.  You can't do it via the HA User Interface as the source entity for the UM is validated that its in the sensor domain, but if you create it in YAML then it all works OK.

Add to configuration.yaml:

```
utility_meter:
# Predbat daily inverter writes utility meter
  predbat_daily_inverter_writes:
    source: predbat.inverter_register_writes
    name: Predbat Daily Inverter Writes
    unique_id: predbat_daily_inverter_writes
    cycle: daily
```

discodancerstu, 2025-01-07:
Thanks Geoffrey, how can I see/monitor where and when the cross charging is?

gcoan, 2025-01-07:
> Thanks Geoffrey, how can I see/monitor where and when the cross charging is?
> 

I'm purging my entity history now so can only go back a couple of days of history, but here's an example from earlier today of cross charging occurring between my two inverters G and H:

[image]

It generally happens when the inverters are in Demand (Eco/Idle) mode.

There are two clear signs of cross charging:

- Sign 1 is one SoC is rising whilst the other SoC is falling.  Obviously you could get that from solar on one inverter array and not the other, but generally this is a strong indicator of cross charging.

- Sign 2 is battery power on one inverter being positive and the other being negative, i.e. one is charging and the other is discharging

In my case 99.9% of the cross charging I see is from inverter G discharging and inverter H charging. 

I did create an automation to detect this cross charging and to issue a pause charge on inverter H.  This stops the cross charging and then when next runs it will notice that inverter H is in pause mode when it shouldn't be (according to the plan), and will cancel the pause.  And then likely I could get cross charging happening again.

With my gen 1 inverters the cross charging is usually quite gentle and the automation stops it getting wild, but it still happens and I often see that G's battery is empty and H's is full

My automation:

```
alias: Detect and stop cross charging G to H
description: ""
triggers:
  - entity_id:
      - sensor.g_xxxx_battery_power
    for:
      minutes: 1
    above: 0.4
    trigger: numeric_state
conditions:
  - condition: numeric_state
    entity_id: sensor.h_yyyy_battery_power
    below: -0.2
actions:
  - action: select.select_option
    metadata: {}
    data:
      option: PauseCharge
    target:
      entity_id: select.h_yyyy_battery_pause_mode
  - action: script.notify_all_devices
    metadata: {}
    data:
      critical: "N"
      url: ""
      title: “Cross charging detected"
      message: >-
        G battery power {{ states('sensor.g_xxxx_battery_power') }}kW, H
        battery power {{ states('sensor.h_yyyy_battery_power') }}kW
    enabled: false
mode: single
```

discodancerstu, 2025-01-08:
Here is my history from today:

[image]

Does this suggest a lot of XC?

kernow, 2025-01-08:
6 days after installing this update I am averaging 37.1 writes per day.

gcoan, 2025-01-09:
> Here is my history from today:
> 
> Does this suggest a lot of XC?

You are not getting XC because predbat is detecting the SoC difference and is hammering the battery power up and down (bottom graph) on the inverter to stop the XC.

That's what's causing your high number of inverter writes.

Try turning predbat's inverter balancing off and see what it looks like, are your inverters reasonably stable or do you get power shuffling back and forth at high charge and discharge rates all the time which clearly wastes energy in the conversion out and in of the batteries.

Trefor does advise being careful about using the balancing option https://springfall2008.github.io/batpred/caution/#flash-memory

discodancerstu, 2025-01-09:
Thanks Geoffrey, the number of daily writes (across 2 inverters) is now
less than 400, so hope we're in a good place.

ChrisMontford, 2025-04-24:
I have started to look at the register writes, and I am experiencing a similar problem. When Predate initiates charging the number of writes goes up alarmingly.

[image]

The register write counted started at 16,500.

gcoan, 2025-04-24:
> I have started to look at the register writes, and I am experiencing a similar problem. When Predate initiates charging the number of writes goes up alarmingly.

@ChrisMontford we're going to need a load more information about what your setup is, what predbat was doing, what the plan was, the logfile, etc to have any hope of investigating your report.

In the documentation there is a handy daily register writes code example https://springfall2008.github.io/batpred/output-data/#inverter-data - you might add that to make it easier to track what is happening each day.

You'd be better creating a new issue than tagging on this one that is quite stale. 

I'll close this issue as the original problem was resolved in January

closed 2025-04-24 by gcoan