springfall2008/batpred #1713: problems after any predbat updates.

Dave-Parry, 2024-12-09:
Whenever I do a predbat update, I get problems where predbat entities are not available.
[image]
Restarting adding or a restart of HA also does not clear it. 
I cannot downgrade the predbat version as the version selector is not available. To clear I have to reload a complete backup.
predbat auto updates is turned off.

**Predbat version**

was V8.8.2, don't know now as I cant see the version.
Also I did not update the system, not sure if HA did the update.

**Environment details**

- Inverter and battery setup Hybrid inverter and 2 batts
- Standard HAOS installer
- Anything else?

[predbat logs.txt](https://github.com/user-attachments/files/18065360/predbat.logs.txt)

Dave-Parry, 2024-12-09:
Looking at the logs again, it seems it may not be caused by an update, as the logs say V8.8.2 which it was on before the problem. The logs do say "Error: Exception raised 'attributes'" so maybe that is the cause. If so how do I fix this?

Dave-Parry, 2024-12-10:
Trefor, will close this as it may not be a predbat issue. I think it is either a problem with givTCP, MQQT or maybe a HA problem. I suspect that HA may have a problem as reloading the system partially cures it. Will try reinstalling HA and reloading a backup.

Dave-Parry, 2024-12-18:
Sorry trevor, happened again.
Here is the log file.
[Predbat log.txt](https://github.com/user-attachments/files/18184019/Predbat.log.txt)

springfall2008, 2024-12-21:
I pushed a potential fix to 'main' that will skip this error

Dave-Parry, 2024-12-21:
Thanks Freeform, Will try in the morning and let you know how I get on.
Don’t want to tempt fate tonight as there’s quite a few minus rates tonight!!

Regards
David Parry
***@***.***

Dave-Parry, 2024-12-22:
Trefor, thanks that seems to have sorted it out. Did a core update and a ha system update as well as updating to predbat  Main this morning. All without a problem. Also sorry about "freeform" as your name in above post. Damn auto correct ! didn't spot it before posting.

closed 2024-12-23 by springfall2008