springfall2008/batpred #1715: Nonsensical 4am exports

BuhJuhWuh, 2024-12-10:
**Describe the bug**
Last night, and tonight, Predbat has scheduled a spurious export at 4am in the middle of its charging period.  Switching to 'control charge' reverts the schedule to a more sensible charge-only strategy, picking only the cheapest set of slots required to fill the battery.  Switching back to 'control charge and discharge' consistently inserts the 4am export again.  Last night I inadvertently left it on the latter, and the export did take place, so it doesn't seem to be one of those oddities that self corrects as the time of calculation gets closer to the real time.

**Expected behaviour**
Only export when advantageous to do so.  Import prices are well above export at the moment, so this doesn't make sense.

**Predbat version**
v8.8.4 - may have been present in earlier versions, but only noticed this yesterday.

**Environment details**
- Inverter and battery setup - GivEnergy 5kW Gen 3 Inverter with PV and 9.5kWh Gen 2 battery
- Standard HAOS
- Anything else? - Tariff - Octopus Agile and Outgoing Fixed (constant 15p/kWh)

**Screenshots**
[image]

**Log file**
(Not convinced this is telling anything that useful...)
[6adb4f0d_predbat_2024-12-10T21-02-10.834Z.log](https://github.com/user-attachments/files/18085834/6adb4f0d_predbat_2024-12-10T21-02-10.834Z.log)

BuhJuhWuh, 2024-12-11:
Another issue (I _think_ separate, just discovered while looking into the other one) - here is the plan it came up with on 'control charge' - much more sensible. (Screengrabbed from today's plan on the pre-Nordpool repeated data, so looks identical apart from the date.)

[image]

But here is what happened - note the discharges to the house at 0030 and 0130 - as if it hasn't been able to set the freeze mode correctly, and has then had to compensate afterwards by charging again. 

[image]

My log file appears to be scattered liberally with 
`Warn: Inverter 0 Trying to write pause_mode to Disabled didn't complete got PauseDischarge` which seems to roughly fit(?). Any clues as to whether this is a bug or something else happening at my end gratefully received. Again, I think this is separate to the scheduled 4am export issue.

BuhJuhWuh, 2024-12-11:
As far as I can work out, the latter issue was a comms issue, where Predbat was failing to write PauseDischarge, causing the inverter to default to Eco/Demand mode, after which Predbat would then initiate a charge again to get back the lost energy. This can be seen in part 1 of the image below, with all the blips.

I then reset the inverter and rebooted HA/add-ons, and things start to look more sensible in part 2. Hopefully that's that part sorted. (Log from point of restart below.)

[image]
[logs.txt](https://github.com/user-attachments/files/18100487/logs.txt)

However, Predbat is again planning an export at 4am for no apparent reason...

springfall2008, 2024-12-11:
Can you try the version on 'main' ?

BuhJuhWuh, 2024-12-11:
Thanks Trefor. Not sure, to be honest - is there a straightforward means of doing so from within HA?  (I usually just get the updates when they appear there.)  Or if there are instructions somewhere for how to do so, I'm willing to give it a try.

BuhJuhWuh, 2024-12-11:
Looks like it is still planning a 4am export. Each time it has been that characteristic charge>discharge at 0400 followed by discharge at 0430.
[image]

BuhJuhWuh, 2024-12-12:
Tried going back a couple of versions. For what it's worth:

- 8.8.4/main - 4am exports and repeated failure to freeze charge (half-hourly blips & occasional longer unwanted discharges to home);
- 8.7.2 - sorry, didn't check the 4am export; still fails to freeze charge (half-hourly blips); 
- 8.6.1 - early days, but seems to be fine for both issues.

[image]

When it is failing to freeze, Predbat still behaves as if it is freezing, but the inverter behaviour does not reflect this: 

[image]

BuhJuhWuh, 2024-12-13:
4am export issue seems to be resolved in 8.8.6, so will close this one. Still struggling with freeze charge failures, but will move that to a new topic.

closed 2024-12-13 by BuhJuhWuh