springfall2008/batpred #1721: After updating Home Assistant: An Exception Occurred - Cannot read history for sensor...

mechanimal82, 2024-12-12:
**Describe the bug**
After updating Home Assistant to 2024.12.0/1/2 Predbat fails to start and create a plan. Status shows: "An Exception Occurred". Restoring a backup to 2024.11.3 and Predbat works normally.

I noticed an issue with reading the history for the iBoost so have actually disabled this from my apps.yaml and also from within the Predbat controls. I also rebooted the entire system after doing this... interestingly though it's still trying to read the history for this. But also seems to fail to read the load history also, but checking the state of the sensor I can see a value for these sensors, as well as a history and the correct UOM.

**Predbat version**

8.8.x

**Environment details**

- Fox Inverter and batteries
- Tesla T-Smart Immersion
- HAOS installed on Proxmox

**Log file**
[6adb4f0d_predbat_2024-12-12T10-13-54.188Z.log](https://github.com/user-attachments/files/18109651/6adb4f0d_predbat_2024-12-12T10-13-54.188Z.log)
[6adb4f0d_predbat_2024-12-12T09-42-07.392Z.log](https://github.com/user-attachments/files/18109666/6adb4f0d_predbat_2024-12-12T09-42-07.392Z.log)
[apps.yaml.txt](https://github.com/user-attachments/files/18109712/apps.yaml.txt)

gcoan, 2024-12-12:
when you say you rebooted the system, did you do a full shutdown of HA and all the add-on's?

By default a reboot just restarts HA and that might not be enough

mechanimal82, 2024-12-12:
> when you say you rebooted the system, did you do a full shutdown of HA and all the add-on's?
> 
> By default a reboot just restarts HA and that might not be enough

Yep. Full system shutdown of the VM and restart. IT Crowd style :)

gcoan, 2024-12-12:
> Yep. Full system shutdown of the VM and restart. IT Crowd style :)

Great, thanks for confirming.

There have been some other people reporting similar 'unavailable sensor' issues with Predbat and the latest HA version.  My money is on it being a HA problem rather than Predbat, but pinning this down is difficult.

mechanimal82, 2024-12-12:
Agreed, seems HA as restoring a backup gets PredBat working again. I just couldn't see many report and wondered if there may be some breaking changes that PredBat needs to adjust to work around or whether the fault is within HA. Ill stay on 2024.11 for a bit and see what comes up

springfall2008, 2024-12-13:
This error is usually while HA restarts, did it not clear leaving it a few minutes or re-starting predbat add on?

mechanimal82, 2024-12-13:
> This error is usually while HA restarts, did it not clear leaving it a few minutes or re-starting predbat add on?

The error didn't clear, and no plan was being generated for a day or so when I first upgraded HA... I was hoping it'd be something that would get fixed, so each .x release of 2024.12 I've tried:

1. again updating HA (doing a full backup)
2. restarting the entire VM
3. wait 15-20 minutes
4. see the error not clearing and no plan generating
5. restore backup

It must be something in this latest release for HA, but it's only PredBat that seems to fail. I don't know if the way the history stats are being read has changed or something. I've looked at the git repository for the modbus controller (Nathan Marlor's Fox modbus code) but there are no issues reported there.

Maybe I should try a completely clean install of Predbat after upgrading HA. Let me give that a try

mechanimal82, 2024-12-13:
[predbat.log](https://github.com/user-attachments/files/18124092/predbat.log)

Nope. That didn't work. 

1. Updated HA
2. removed predbat and addon folder
3. Rebooted HA system
4. installed Predbat from scratch
5. Started Predbat
6. replaced apps.yaml
7. restarted predbat
8. Exception error occurred again

log file attached

springfall2008, 2024-12-14:
Are you saying in HA there is something called 'sensor.load_energy_today' which has a history?

mechanimal82, 2024-12-15:
> Are you saying in HA there is something called 'sensor.load_energy_today' which has a history?

That's one of the sensors provided by the Fox modbus integration. Is this the sensor causing the issue? Looking at it, it resets itself each day, so is not a cumulative sensor. It does have a UoM of kWh and appears and seems to work OK when using HA 2024.11.3,

springfall2008, 2024-12-16:
Can you show the details of load_energy_today sensor, e.g. is the required_unit kWh?

mechanimal82, 2024-12-16:
[image]

tmh88, 2024-12-17:
i had the same issue yesterday at 17.20 unfortunately didn't realize until this morning when the batteries where flat. its been working brilliantly until then, no updates on PB or HA since the 14th. i have Iboost enable toggled off as i use it to heat my hot tub in the summer. so data is there and not changed since the end of august. commenting out the sensor in the apps.yaml is the only way to remove the error

 ## log
2024-12-17 18:10:00.884258: Info: record_status Error: Exception raised 
2024-12-17 18:10:00.884493: Error: 
2024-12-17 18:15:00.501712: --------------- PredBat - update at 2024-12-17 18:15:00+00:00 with clock skew 0 minutes, minutes now 1095
2024-12-17 18:15:00.503246: Saved current settings to /addon_configs/6adb4f0d_predbat/predbat_config.json
2024-12-17 18:15:00.503493: Using cached GITHub data for https://api.github.com/repos/springfall2008/batpred/releases age 95.0 minutes
2024-12-17 18:15:00.503606: Predbat /config/predbat.py version v8.8.9 currently running, latest version is v8.8.9 latest beta v8.8.9
2024-12-17 18:15:00.503679: Autoupdate: Currently up to date
2024-12-17 18:15:00.570341: Configuration: forecast_hours 48 num_cars 0 debug enable is False calculate_plan_every 10
2024-12-17 18:15:00.571137: Predbat mode is set to Control charge
2024-12-17 18:15:00.800932: Error: Unable to fetch history for sensor.spa_heater_power_daily
2024-12-17 18:15:00.873038: Warn: record_status Error: Unable to fetch history from sensor.spa_heater_power_daily
2024-12-17 18:15:00.873235: Error: Exception raised 
2024-12-17 18:15:00.875937: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 965, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/predbat.py", line 535, in update_pred
    self.fetch_sensor_data()
  File "/config/fetch.py", line 760, in fetch_sensor_data
    self.iboost_energy_today, iboost_energy_age = self.minute_data_load(self.now_utc, "iboost_energy_today", self.max_days_previous, required_unit="kWh")
  File "/config/fetch.py", line 407, in minute_data_load
    raise ValueError
ValueError

2024-12-17 18:15:00.988645: Info: record_status Error: Exception raised 
2024-12-17 18:15:00.988872: Error: 

## Sensor
[image]

## Sensor History
[image]

springfall2008, 2024-12-21:
> i had the same issue yesterday at 17.20 unfortunately didn't realize until this morning when the batteries where flat. its been working brilliantly until then, no updates on PB or HA since the 14th. [...]

I think this is different, your spa heater hasn't been updated in so long I think HA deleted the history

springfall2008, 2024-12-21:
> ![image](https://private-user-images.githubusercontent.com/854263/396029177-d2442a68-9388-4a10-9236-13f62518eedf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ4MTM4NzEsIm5iZiI6MTczNDgxMzU3MSwicGF0aCI6Ii84NTQyNjMvMzk2MDI5MTc3LWQyNDQyYTY4LTkzODgtNGExMC05MjM2LTEzZjYyNTE4ZWVkZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyMVQyMDM5MzFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mMjZlMzJjOWNkZWFmZDhhY2NlY2IyZjA3YWRmNWNhMzc4OWRkM2YxM2IzY2QzZjY1M2U3YmUwZWQ5YTMyY2Q5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.yNH9IzdBfaKStzsXVh5jtMjzzJHNNscUwQrnKGE0vVc)

Can you show the history of this sensor on a chart?

tmh88, 2024-12-21:
> Can you show the history of this sensor on a chart?

i'm not sure how to show it on a chart, how do you do that?

mechanimal82, 2025-01-06:
Was actually an issue caused by the Modbus update in Home Assistant. This is now resolved in the Fox modbus Controller, so am closing

closed 2025-01-06 by mechanimal82