springfall2008/batpred #1722: Constant logfile errors - State attributes for predbat exceed maximum size of 16384 bytes.

TheSargey, 2024-12-12:
**Describe the bug**
I have the following errors appearing every five minutes in home-assistant.log:

2024-12-12 10:15:18.012 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for predbat.pv_energy exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-12-12 10:15:18.104 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for predbat.import_energy exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-12-12 10:15:18.371 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for predbat.metric exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-12-12 10:15:18.819 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for predbat.base10_metric exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-12-12 10:15:19.073 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for predbat.base10_import_energy exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-12-12 10:15:19.440 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for predbat.best10_metric exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-12-12 10:15:19.665 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for predbat.best10_import_energy exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-12-12 10:15:20.469 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for predbat.load_power_best exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-12-12 10:15:20.948 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for predbat.best_load_energy exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-12-12 10:15:22.444 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for predbat.plan_html exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2024-12-12 10:16:20.472 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for predbat.load_energy exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored

**Expected behaviour**
I'd like to understand how to stop this from occuring please.

**Predbat version**
8.8.1

**Environment details**

- GivEnergy Gen1
- Standard HAOS

gcoan, 2024-12-12:
This is already covered in the FAQ's and guidance as to how to suppress these warnings

https://springfall2008.github.io/batpred/faq/#predbat-is-causing-warning-messages-about-exceed-maximum-size-in-the-home-assistant-core-log

TheSargey, 2024-12-12:
Morning Geoff, my sincere apologies, I'd scanned the forum but neglected to RTFM.

Many thanks for your help as always.

Sarge

gcoan, 2024-12-12:
No worries @TheSargey you are not the first who hasn't found this in the FAQ's!

I will be updating the advice about the cause and what to do about these messages, but for now these can be safely suppressed as they are just warnings

TheSargey, 2024-12-13:
Morning Geoff,

Just a quick one, I've added the exceptions in the configuration file but one error was still reporting - I think you'll need to add this into your list on the above link:

- "State attributes for predbat.import_energy exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored"

Cheers,

Sarge

closed 2024-12-13 by TheSargey

gcoan, 2024-12-13:
Thanks @TheSargey this extra exclusion was on my to-do list to add it https://community.givenergy.cloud/d/3696-first-night-live-on-predbat/2650, have done so now