springfall2008/batpred #1727: Predbat v8.5.5/8.5.6 seems to break 'switch.predbat_set_charge_low_power'

campbellcopeland, 2024-12-13:
**Describe the bug**
After upgrading to v8.5.5, full spead charging happens at the end of the OCtopus Go offpeak window and not slower charging throughout

**Expected behaviour**
similar charging pattern to v8.5.4

**Predbat version**

v8.5.5 / v8.5.6

**Environment details**

- Inverter and battery setup : GivEnergy AIO
- Standard HAOS installer or Docker : HAOS
- Anything else?

**Screenshots**
[image]

[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

springfall2008, 2024-12-13:
Do you have combine charge enabled? If not maybe turn it on.

campbellcopeland, 2024-12-13:
Both combine charge and discharge enabled. Consistently ignoring low power setting on v8.5.5 onwards

campbellcopeland, 2024-12-13:
Turning that on results in same behaviour as v8.5.4. I’ll stick with that for now 👍

closed 2024-12-14 by springfall2008