springfall2008/batpred #1728: after upgrade to 8.8.6 (from 8.8.5) no more charging slots occur in the plan

eddysteurs, 2024-12-13:
**Describe the bug**
after I updated to 8.8.6 no more chargig slots appear in the plan and soc remains at minimum (12%) for the next 36  hrs

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Predbat version**
8.8.6

**Environment details**

- Inverter and battery setup: solis and pylontecg
- Standard HAOS installer or Docker : HAOS 2024.12.3
- Anything else?

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.
[image]
[image]
[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

BuhJuhWuh, 2024-12-13:
I had this initially on updating to 8.8.6, when the nordpool data was showing a strangely flat/non-peak-y day ahead - so could have been it just wasn't meeting the threshold to bother charging? Once the new Agile data arrived at 4pm mine sorted itself out to the usual overnight charge. Has yours done the same now?

pgit2345, 2024-12-13:
I'm getting this too. Just upgraded to latest predbat and HA 2024.12.1 to .3

It drops to 4% overnight which is lower than best soc keep and doesn't plan any charges, it's waiting for solar it seems.

There's a couple of more expensive slots in the morning, so I'd have thought some charge would happen - especially to not leave it at 4%

springfall2008, 2024-12-13:
It's possible I messed something up with the Soc keep, what did you set keep to?

springfall2008, 2024-12-13:
Btw most useful thing is if you enable debug mode and attach the yaml file from the debug directory then I can recreate your plan locally

pgit2345, 2024-12-13:
[predbat_debug_20_55_00.yaml.txt](https://github.com/user-attachments/files/18131833/predbat_debug_20_55_00.yaml.txt)

Best soc keep is 0.7 kWh

Ta

eddysteurs, 2024-12-14:
best soc keep is 3.0 kwh

[predbat_debug_14_15_00.yam.txt](https://github.com/user-attachments/files/18135984/predbat_debug_14_15_00.yam.txt)

pgit2345, 2024-12-14:
Mine is planning to charge around 4.30am, but will wait to see what happens post 4pm when the rates are announced.  

It's going to drop to 4% and stay there until 0430, but given the rates are expected to get cheaper every hour or so until 0430 that seems to make sense.

springfall2008, 2024-12-14:
I've put back the metric keep method to the old one for now

https://github.com/springfall2008/batpred/releases/tag/v8.8.8

I don't think the new method worked.

pgit2345, 2024-12-14:
Yep, looking more normal now. 🤞😀

closed 2024-12-15 by springfall2008