springfall2008/batpred #1731: Trying to export when soc 10%

crspytopgn, 2024-12-13:
**Describe the bug**
My battery is trying to export even though import costs are higher than export

**Expected behaviour**
Not export at all

**Predbat version**

8.8.6

**Environment details**

Solax x1 hybrid g4
Standard HAOS installer 

**Screenshots**
[image]

**Log file**
[predbat_debug_22_20_00.zip](https://github.com/user-attachments/files/18132566/predbat_debug_22_20_00.zip)

gcoan, 2024-12-13:
What is your meteic_min_improvement_export set to?

Per the release notes it's recommended to set this to 5.0 which is the new default value

> In an attempt to reduce random export slots which have little impact the default metric_min_improvement_export has been changed to 5, meaning export must be worth at least 5p/30 minute period to be triggered. At a 3kw export rate this would be 5p for 1.5kWh which is around 3.3p per kWh. You can change this setting in expert mode.

crspytopgn, 2024-12-14:
> What is your metric_min_improvement_export set to?
> 
> Per the release notes it's recommended to set this to 5.0 which is the new default value
> 
> > In an attempt to reduce random export slots which have little impact the default metric_min_improvement_export has been changed to 5, meaning export must be worth at least 5p/30 minute period to be triggered. At a 3kw export rate this would be 5p for 1.5kWh which is around 3.3p per kWh. You can change this setting in expert mode.

It was set to 0.1 so I've just changed it to 5.0

gcoan, 2024-12-14:
that should hopefully fix the extra exports.

My metric was already set to 4.0 but the default was 0.1 previously

springfall2008, 2024-12-14:
I found a bug here, the default metric_min_improvement_discharge was changed in a way where the value was being rounded to an integer so the older 0.1 value because 0. Fix on main for the next release.

closed 2024-12-15 by springfall2008