springfall2008/batpred #1732: Not sure what is going on - Charging

searlecom, 2024-12-14:
**Describe the bug**

The charging plan does not make sense to me and the battery will not charge when setting a manual charging slot.  I had to override the charging slot by using the website to set the charging slot to force a charge to get me over the peak rate hump.

I am new to Predbat and this was working the other day.

**Expected behaviour**

Last night my inverter did not charge my battery.  SOC went down to 20%, I would have expected it to charge overnight

**Predbat version**

v8.8.6

**Environment details**

- Sunsynk 8.8K Ecco and 20KW of SSLB01
- Standard HAOS install
- No solar - Updated the Yaml file to set solar forecast to 0, done this today.

**Screenshots**
[image]
[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

[predbat.log](https://github.com/user-attachments/files/18136369/predbat.log)
[automations_yaml.txt](https://github.com/user-attachments/files/18136370/automations_yaml.txt)
[apps_yaml.txt](https://github.com/user-attachments/files/18136371/apps_yaml.txt)
[predbat_debug_15_05_00_yaml.txt](https://github.com/user-attachments/files/18136376/predbat_debug_15_05_00_yaml.txt)

springfall2008, 2024-12-14:
The plan looks okay to me although your load seems very low , is it accurate?

Has it ever been able to charge, maybe the inverter controls have an issue?

searlecom, 2024-12-14:
Yes, it has been charging.

I set it to manual charge at 23:14.

It looks like the inverter timer SOC and charge from grid switch has been changed but not the timer start time (prog1):

[image]

[predbat.log](https://github.com/user-attachments/files/18138143/predbat.log)

searlecom, 2024-12-14:
I have just set the timers on the inverter as below:

[image]

So next time PredBat calls for charge it will switch "charge from grid" on and set the SOC to 100 on timer 1, thus the battery should charge.

So I have now set a force charge and that is now working.

Turn it off again and it stopped charging.

So from what I am seeing, it looks like the Prog1 timer time is not being set to the correct time on a manual charge at least, not sure about the auto charge.

It might be worth running a check on the timers "on time" (as set on the inverter), when a charge is needed and if the setting is incorrect then change it to allow charging in the needed window.  Or even better, if you can, if a Sunsynk inverter, set prog1 to start at 00:00 and prog2 to start at 24:00 thus it will be able to charge all day but will use the Charge from Grid switch as per my config.  However I am not sure how that will impact any export as I do not have solar yet but I don't thing the timers control the export but I could be wrong on that.

Any thoughts and ideas are welcome from your end good sir.

searlecom, 2024-12-15:
Pleased to report setting the timers as above has resulted in a full battery overnight.

I will leave the ticket open so you can look into the timer options if needed.  It might help someone else going forward, assuming I am setting things up as the system was designed to be.

closed 2026-08-28 by chalfontchubby