springfall2008/batpred #1739: Custom inverter discharge end time out by one

chris-y, 2024-12-15:
**Describe the bug**
The custom inverter discharge end time predbat_mine_0_discharge_end_time is set one minute into the future. 
eg. I'm seeing 10:01 instead of 10:00

This is causing overlap with the charge time when a discharge occurs before a charge.

**Expected behaviour**
The discharge end time should be set to the expected time, not one minute afterwards.

**Predbat version**

8.8.6

**Screenshots**
[image]

[image]

springfall2008, 2024-12-15:
What version are you on, the overlap was meant to be fixed a while back?

springfall2008, 2024-12-15:
Hmm, looks like a bug the overlap was meant to be fixed a while back

springfall2008, 2024-12-15:
Can you capture a logfile when this is set?

chris-y, 2024-12-15:
17343	2024-12-15 22:56:00.003303: Inverter 0 Skipping service charge_stop_service domain charge with data {'device_id': ''} as already called
17342	2024-12-15 22:56:00.003032: Car charging from battery is off, next slot for car 0 is 12-15 23:00:00 - 12-15 23:30:00
17341	2024-12-15 22:55:58.675217: Inverter 0 Calling service discharge_start_service domain discharge with data {'device_id': '', 'target_soc': 13, 'power': 3600}
17340	2024-12-15 22:55:58.675114: Inverter 0 Skipping service charge_stop_service domain charge with data {'device_id': ''} as already called
17339	2024-12-15 22:55:58.674917: Inverter 0 set force export to True
17338	2024-12-15 22:55:58.674715: Inverter 0 Wrote Timed Export to inverter_mode, successfully now Timed Export
17337	2024-12-15 22:55:26.581044: Sleeping (workaround) as start/end of discharge window was just adjusted
17336	2024-12-15 22:55:26.580992: Inverter 0 Turning on scheduled export
17335	2024-12-15 22:55:26.580931: Inverter 0 Wrote scheduled_discharge_enable to True successfully and got on
17334	2024-12-15 22:55:26.580744: Switch sensor.predbat_MINE_0_scheduled_discharge_enable is now on
17333	2024-12-15 22:55:24.518588: Inverter 0 Wrote 23:01:00 to discharge_end_time, successfully now 23:01:00
17332	2024-12-15 22:55:22.452212: Inverter 0 Set new export end time to 23:01:00 was 00:00:00
17331	2024-12-15 22:55:22.452053: Inverter 0 Wrote 23:01:00 to idle_start_time, successfully now 23:01:00
17330	2024-12-15 22:55:20.389260: Inverter 0 set new demand (idle) start time to 23:01:00 was 12:00:00
17329	2024-12-15 22:55:20.389143: Adjust demand (idle) time computed is 23:01:00-23:59:00
17328	2024-12-15 22:55:20.388799: Adjust idle ti

springfall2008, 2024-12-16:
What about earlier in the log does it mention charging times? I think what's happening here is scheduled charge enable is off as it triggered the discharge service and its assuming charge times don't apply in this case.

springfall2008, 2024-12-16:
Are you using inverter type 'GS' as 'can_span_midnight' as False disables the +1 minute thing anyhow?

chris-y, 2024-12-16:
I've attached the full log of that pass.

I'm using inverter type 'MINE' but 'can_span_midnight' is True, so maybe I just need to disable that?

[predbat2255.log](https://github.com/user-attachments/files/18148046/predbat2255.log)

chris-y, 2024-12-16:
> What about earlier in the log does it mention charging times? I think what's happening here is scheduled charge enable is off as it triggered the discharge service and its assuming charge times don't apply in this case.

OK, I can split those out, I've seen in the past it hasn't called the charge service again when it occurs immediately after a discharge though.  This is what happened which prompted me to raise this (battery didn't charge overnight - although I'm not sure if that was due to the overlap or some failure elsewhere) - didn't call it again with the discharge removed.

chris-y, 2024-12-19:
I've just realised what is causing my charge failure. Discharging was set from 0000-2330, and charging from 2330-0530 - so after midnight it was set to charge and discharge. The discharge start time shouldn't be 0000?

I've just turned off "can span midnight" to see if it helps.

chris-y, 2024-12-20:
With "can span midnight" off, Predbat didn't call the service again just after midnight to set the next day's charging (I ran it manually just now and it set the correct times). The off by one on the discharge end time is fixed however. 

So looks like;
* Discharge start time needs checking as it is being set to midnight
* Service needs to be called at midnight if can_span_midnight is False

chris-y, 2024-12-22:
Here's the log for it not running again after midnight when can span midnight is false: 

 13762	2024-12-22 00:00:30.293526: No export window planned
13761	2024-12-22 00:00:30.293470: Inverter 0 Skipping service charge_start_service domain charge with data {'device_id': '', 'target_soc': 100, 'power': 3600} as already called
13760	2024-12-22 00:00:30.293378: Inverter 0 Skipping service discharge_stop_service domain discharge with data {'device_id': ''} as already called
13759	2024-12-22 00:00:30.293183: Inverter 0 Updated start and end charge window to 00:00:00 - 05:30:00 (old 23:30:00 - 23:59:00)
13758	2024-12-22 00:00:30.293002: Inverter 0 Wrote 05:30:00 to charge_end_time, successfully now 05:30:00
13757	2024-12-22 00:00:28.228092: Inverter 0 Wrote 00:00:00 to charge_start_time, successfully now 00:00:00
13756	2024-12-22 00:00:26.167502: Inverter 0 Wrote 23:59:00 to idle_end_time, successfully now 23:59:00
13755	2024-12-22 00:00:24.102898: Inverter 0 set new demand (idle) end time to 23:59:00 was 00:00:00
13754	2024-12-22 00:00:24.102740: Inverter 0 Wrote 05:30:00 to idle_start_time, successfully now 05:30:00
13753	2024-12-22 00:00:22.039951: Inverter 0 set new demand (idle) start time to 05:30:00 was 00:00:00
13752	2024-12-22 00:00:22.039813: Adjust demand (idle) time computed is 05:30:00-23:59:00
13751	2024-12-22 00:00:22.039416: Adjust idle time, charge 00:00:00-05:30:00 discharge 00:00:00-00:00:00
13750	2024-12-22 00:00:22.038941: Low Power mode: minutes left: 320 absolute: 330 SOC: 3.072 Target SOC: 9.6 Charge left: 6.53 Max rate: 3600.0 Min rate: 1187.2727272727273 Best rate: 1500.0
13749	2024-12-22 00:00:22.028902: Charge window will be: 2024-12-22 00:00:00+00:00 - 2024-12-22 05:30:00+00:00 - current soc 32 target 100
13748	2024-12-22 00:00:22.028725: Base export window [ ]
13747	2024-12-22 00:00:22.028559: Base charge window [ 22-12 23:30:00 - 22-12 23:59:00 @ 0p 100%, 23-12 23:30:00 - 23-12 23:59:00 @ 0p 100% ]
13746	2024-12-22 00:00:21.865779: Found 1 inverters totals: min reserve 0.96 current reserve 0.96 soc_max 9.6 soc 3.07 charge rate 1.2750000000000001 kW discharge rate 3.5999999999999996 kW battery_rate_min 0.0 w ac limit 3.6 export limit 3.6 kW loss charge 4 % loss discharge 4 % inverter loss 4 %
13745	2024-12-22 00:00:21.865632: Note: Inverter does not support reserve - disabling reserve functions

springfall2008, 2024-12-23:
Fix on 'main' for the midnight issue to test

chris-y, 2024-12-24:
> Fix on 'main' for the midnight issue to test

That worked, thanks!

closed 2024-12-26 by springfall2008