springfall2008/batpred #1742: SolaX: TypeError: strptime() argument 1 must be str, not dict

M0LTE, 2024-12-15:
**Describe the bug**

Solax inverter:

```
predbat  | 2024-12-15 22:30:34.808234: Info: record_status Error: Exception raised strptime() argument 1 must be str, not dict
predbat  | 2024-12-15 22:30:34.808415: Error: strptime() argument 1 must be str, not dict
predbat  | Traceback (most recent call last):
predbat  |   File "/config/hass.py", line 173, in timer_tick
predbat  |     item["callback"](None)
predbat  |   File "/config/predbat.py", line 970, in run_time_loop
predbat  |     raise e
predbat  |   File "/config/predbat.py", line 965, in run_time_loop
predbat  |     self.update_pred(scheduled=True)
predbat  |   File "/config/predbat.py", line 556, in update_pred
predbat  |     status, status_extra = self.execute_plan()
predbat  |                            ^^^^^^^^^^^^^^^^^^^
predbat  |   File "/config/execute.py", line 186, in execute_plan
predbat  |     inverter.adjust_charge_window(charge_start_time, charge_end_time, self.minutes_now)
predbat  |   File "/config/inverter.py", line 2097, in adjust_charge_window
predbat  |     self.press_and_poll_button(entity_id)
predbat  |   File "/config/inverter.py", line 2131, in press_and_poll_button
predbat  |     time_pressed = datetime.strptime(self.base.get_state_wrapper(entity_id, refresh=True), TIME_FORMAT_SECONDS)
predbat  |                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
predbat  | TypeError: strptime() argument 1 must be str, not dict
```

This appears to be inside this:
```
    def press_and_poll_button(self, entity_id):
        """
        Call a button press service (Solis) and wait for the data to update
        """
```

implying the code should not even be called for SolaX inverters.

**Expected behaviour**

Exception not to happen

**Predbat version**

8.8.9

**Environment details**

- Solax x1 hybrid + master battery + 2x slave battery
- Bottlecap dave modbus
- Docker

springfall2008, 2024-12-16:
What is in your apps.yaml for 'charge_discharge_update_button' ?

springfall2008, 2024-12-21:
You should probably turn off the time_button_press inverter option, but I'll change the code to ignore it if commented out

springfall2008, 2024-12-21:
Fix on main to catch this crash

M0LTE, 2024-12-21:
@springfall2008 I pulled the updated inverter.py:

```
predbat  | 2024-12-21 16:42:54.009266: Error: Exception raised Inverter.press_and_poll_button() missing 1 required positional argument: 'entity_id'
predbat  | 2024-12-21 16:42:54.016187: Error: Traceback (most recent call last):
predbat  |   File "/config/predbat.py", line 911, in update_time_loop
predbat  |     self.update_pred(scheduled=False)
predbat  |   File "/config/predbat.py", line 556, in update_pred
predbat  |     status, status_extra = self.execute_plan()
predbat  |                            ^^^^^^^^^^^^^^^^^^^
predbat  |   File "/config/execute.py", line 287, in execute_plan
predbat  |     inverter.adjust_force_export(True, discharge_start_time, discharge_end_time)
predbat  |   File "/config/inverter.py", line 1735, in adjust_force_export
predbat  |     self.press_and_poll_button()
predbat  | TypeError: Inverter.press_and_poll_button() missing 1 required positional argument: 'entity_id'
predbat  |
predbat  | 2024-12-21 16:42:54.035835: Info: record_status Error: Exception raised Inverter.press_and_poll_button() missing 1 required positional argument: 'entity_id'
predbat  | 2024-12-21 16:42:54.036026: Error: Inverter.press_and_poll_button() missing 1 required positional argument: 'entity_id'
predbat  | Traceback (most recent call last):
predbat  |   File "/config/hass.py", line 173, in timer_tick
predbat  |     item["callback"](None)
predbat  |   File "/config/predbat.py", line 917, in update_time_loop
predbat  |     raise e
predbat  |   File "/config/predbat.py", line 911, in update_time_loop
predbat  |     self.update_pred(scheduled=False)
predbat  |   File "/config/predbat.py", line 556, in update_pred
predbat  |     status, status_extra = self.execute_plan()
predbat  |                            ^^^^^^^^^^^^^^^^^^^
predbat  |   File "/config/execute.py", line 287, in execute_plan
predbat  |     inverter.adjust_force_export(True, discharge_start_time, discharge_end_time)
predbat  |   File "/config/inverter.py", line 1735, in adjust_force_export
predbat  |     self.press_and_poll_button()
predbat  | TypeError: Inverter.press_and_poll_button() missing 1 required positional argument: 'entity_id'
```

M0LTE, 2024-12-21:
I think line 2122 just wants changing to `def press_and_poll_button(self):`

M0LTE, 2024-12-21:
I have now added `time_button_press: False` to my `apps.yml`- it was previously not defined

philicibine, 2024-12-22:
> charge_discharge_update_button

I am seeing this same error.  I am running a solis inverter with the solax modbus integration.  I guess as its solis.. i do need this button?  Any idea how i can fix?

M0LTE, 2024-12-22:
I think the fix for the SolaX issue has introduced a bug which might affect
all

philicibine, 2024-12-22:
Have tried rolling back the version but still seeing this error

```
Error: Inverter.press_and_poll_button() missing 1 required positional argument: 'entity_id'
```

springfall2008, 2024-12-26:
Fixed on main now

closed 2024-12-26 by springfall2008