springfall2008/batpred #1744: Solcast Warning in log

JohanAlvedal, 2024-12-16:
**Describe the bug**
I'm not sure what I've done, if I have it, if so I've missed it. What I know I have done is that I have set PV Scaling to 0.55 as my PVs basically do not give anything due to frost and snow. I get this warning in the log:

2024-12-16 06:25:02.578485: Warn: PV Forecast data adds up to 42.15 kWh but total sensors add up to 7.85 kWh, this is unexpected and hence data maybe misleading (factor 5.4)
--
**Predbat version**

8.8.9

**Environment details**
- Huawei Inverter Luna 2000 battery
- Standard HAOS 

**Screenshots**
[image]

**Predbat debug yaml file**
[predbat_debug_06_25_00.yaml.txt](https://github.com/user-attachments/files/18145298/predbat_debug_06_25_00.yaml.txt)

springfall2008, 2024-12-16:
I don't think pv_scaling is related. 

Are you using solcast integration or Predbat's direct to Solcast method?

JohanAlvedal, 2024-12-16:
> I don't think pv_scaling is related.
> 
> Are you using solcast integration or Predbat's direct to Solcast method?

Solcast integration

springfall2008, 2024-12-21:
Are you getting correct data or is it wrong?

JohanAlvedal, 2024-12-21:
It is correct

closed 2026-08-28 by chalfontchubby