springfall2008/batpred #1745: Debug mode switch

JohanAlvedal, 2024-12-16:
It would be good if you got a warning at the debug enable switch or if it went off by itself after a while, now I saw that I forgot mine which resulted in one and a half day 5 minute debug logs!!

Or on restart that it empties the folder.

springfall2008, 2024-12-16:
I've added the status of the debug_enable to the HTML plan in the next release.

Were you using this to generate a debug YAML? If so maybe it could be good just to have a quick way to generate that instead?

mpartington, 2024-12-16:
If possible, I think it needs a way of auto deleting the yaml debug files. I've left it on whilst testing the various main versions on export, but its a 3.7 Mb file every 5 mins, so thats over 1Gb data every 24 hours.

Appreciate it's not meant to be on permanently,  but it could kill HA back ups (to Google) and disk space quite quickly.

A way to snapshot a debug yaml would be useful

gcoan, 2024-12-16:
Either a snapshot or manage the number of file generations of the debug yaml files to limit the storage requirements

springfall2008, 2024-12-21:
It will only keep one every 5 minutes and then they get overwritten as there is no date stamp just time

gcoan, 2024-12-22:
They do accumulate quite a bit of space, 2.3Mb each, and they definitely slow Predbat down in execution.

@JohanAlvedal I just put together a simple automation to turn off the debug mode if its been left on for 20 minutes:

```
alias: "Predbat: Auto turn-off debug mode"
description: ""
triggers:
  - trigger: state
    entity_id:
      - switch.predbat_debug_enable
    to: "on"
    for:
      minutes: 20
conditions: []
actions:
  - action: script.notify_all_devices
    metadata: {}
    data:
      title: "INFO: Predbat debug has been on for 20 minutes"
      critical: "N"
      url: ""
      message: switch.predbat_debug_enable turned off
  - action: switch.turn_off
    target:
      entity_id: switch.predbat_debug_enable
    data: {}
mode: single
```

The notify_all_devices is a script I have to send a mobile alert so you could remove this if not required

JohanAlvedal, 2024-12-22:
> They do accumulate quite a bit of space, 2.3Mb each, and they definitely slow Predbat down in execution.
>
> @JohanAlvedal I just put together a simple automation to turn off the debug mode if its been [...]

Tnx.

springfall2008, 2024-12-23:
BTW: I've added a button to the WebUI to create and download a debug YAML so you don't need to enable debug mode anymore:

[image]

gcoan, 2024-12-23:
> BTW: I've added a button to the WebUI to create and download a debug YAML so you don't need to enable debug mode anymore

That's useful to know, I didn't appreciate from the release notes that this creates and downloads the debug YAML file.  I assumed from the UI that it just downloads the latest file that had previously been created.

Might be worth changing the UI so it appears separately as create and download?

springfall2008, 2024-12-23:
Good point, I could probably make that clearer

springfall2008, 2025-01-12:
This is now done, 
[image]

closed 2025-01-12 by springfall2008