springfall2008/batpred #1746: Not sure SolaX inverter control is complete - needs to touch two entities

M0LTE, 2024-12-16:
**Describe the bug**

To properly control the SolaX inverters charge/discharge/self-use state, 

I have configured the following, but I think the mapping is too simplistic, since two entities need adjusting to fully control the inverter, not one:

```
  charge_start_service:
    service: select.select_option
    entity_id: select.solax_manual_mode_select
    option: "Force Charge"
  charge_stop_service:
    service: select.select_option
    entity_id: select.solax_manual_mode_select
    option: "Stop Charge and Discharge"
  discharge_start_service:
    service: select.select_option
    entity_id: select.solax_manual_mode_select
    option: "Force Discharge"
```

How I think it should map:

```
Predbat Mode      |  select.solax_charger_use_mode  |  select.solax_manual_mode_select
Demand            |  Self Use Mode                  |  anything
Charging          |  Manual Mode                    |  Force Charge
Freeze charging   |  Manual Mode                    |  Stop Charge and Discharge
Hold charging     |  Manual Mode                    |  Stop Charge and Discharge
No Charge         |  Manual Mode                    |  Stop Charge and Discharge
Exporting         |  Manual Mode                    |  Force Discharge
Freeze exporting  |  Manual Mode                    |  Stop Charge and Discharge
Hold exporting    |  Self Use Mode                  |  anything
Calibration       |  n/a                            |  n/a
```

It's possible I've just misconfigured it- I'm a new user.

M0LTE, 2024-12-16:
Concrete example which just happened:

Predbat status changed from Demand to Charging

solax_charger_use_mode was set to Self Use Mode by me, earlier, manually, for Demand mode. 

Predbat changed solax_manual_mode_select to Force Charge, but left solax_charger_use_mode at Self Use Mode. 

As well as changing solax_manual_mode_select to Force Charge, it should have also changed solax_charger_use_mode to Manual Mode.

Without doing this, setting solax_manual_mode_select to Force Charge has no effect on the behaviour of the inverter- it remained in self use mode.

M0LTE, 2024-12-16:
I'm going to try using scripts for this. 

Something like

```
  # Service for start/stop
  charge_start_service:
    service: script.inverter_force_charge
  charge_stop_service:
    service: script.inverter_self_use
  discharge_start_service:
    service: script.inverter_force_discharge
```

with simple scripts defined in HASS which make the necessary adjustments to the inverter settings for each mode

will report back

M0LTE, 2024-12-21:
The above workaround did work well for my situation.
I may publish a refined SolaX temp,late, since the one in the repo is relatively broken.

M0LTE, 2024-12-21:
Ok, RTFM, you can call multiple services.

closed 2024-12-21 by M0LTE

nikotime, 2025-01-21:
> Ok, RTFM, you can call multiple services.

Hey @M0LTE , I was wondering whether you could share your apps.yaml for your Solax inverter with you mentioned changes? I have done mine with simple automations reading from sensors but would love to do it properly!

headbouyJB, 2025-03-03:
This tripped me up too - the template yaml for Solax has only one entity, and didn't actually act on the inverter in the way it is expected. 
Reckon that template linked in the docs could do with an update!