springfall2008/batpred #1748: Export not working, high night time load estimates and car battery SOC not worming

jasonhenley, 2024-12-16:
**Describe the bug**
A clear and concise description of what the bug is.
Night time, from 20:30 estimates of load seem high, causing no or low exporting. Sometimes I have to manually switch on export. Car SOC is linked to a sensor on HA, but is not being picked up in PB - shows 100%
**Expected behaviour**
A clear and concise description of what you expected to happen.

**Predbat version**

8.89

**Environment details**

- Inverter and battery setup
- 9.5KWH batter
- 3000 invertor (not AIO)
- Standard HAOS installer or Docker
- Standard HAOS
[predbat_debug_21_05_00.yaml.txt](https://github.com/user-attachments/files/18157195/predbat_debug_21_05_00.yaml.txt)
[predbat_debug_21_05_00.yaml.txt](https://github.com/user-attachments/files/18157197/predbat_debug_21_05_00.yaml.txt)

- Anything else?

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

langersuk, 2024-12-16:
I have also noticed that Export is not working, rolled back to 8.8.4 and it is fine; seems to stopped the export from 8.8.5 onwards
[image]

[image]

springfall2008, 2024-12-21:
The plan I generated by on your YAML and 8.8.11 below has no export before the charging window as it predicts you run out of battery first. It also has no export during the charge period as switch.predbat_calculate_export_oncharge is turned off (expert mode feature). 

So the only question is your night-time load skewed due to your car? It appears like you have not set **car_charging_energy** meaning that previous car charging will be included in your load predictions? I suggest you find a sensor for your car charging energy.

"""
car_charging_energy - Set in apps.yaml to point to a Home Assistant entity which is the incrementing kWh data for the car charger. This has been pre-defined to a regular expression that should auto-detect the appropriate Wallbox and Zappi car charger sensors, or edit as necessary in apps.yaml for your charger sensor.
This can be set to a list of car charging energy sensors, one per line if you have multiple EV car chargers.
TIP: You can also use car_charging_energy to remove other house load kWh from the data Predbat uses for the forecast, e.g. if you want to remove Mixergy hot water tank heating data from the forecast such as if you sometimes heat on gas, and sometimes electric depending upon import rates.

input_number.predbat_car_charging_energy_scale - A Home Assistant entity used to define a scaling factor (in the range 0.1 to 1.0) to multiply the car_charging_energy sensor data by if required (e.g. set to 0.001 to convert Watts to kW).
"""

[image]

closed 2026-08-20 by chalfontchubby