springfall2008/batpred #1749: I'm at a loss, Predbat is fubar

savethemanual, 2024-12-16:
**Describe the bug**
After updating to the latest version of Predbat, 8.8.9, it's now failing to start properly and complaining about lots of things, which despite crawling through the logs, I cannot establish the issue. One symptom is that Predbat thinks my SOC is 0%, it's complaining about car charging despite not having a car and setting it to no car in apps.yaml. Inverter charge rate defaults to 2.6kW. AC limit is 450,000 kW?!
I also recently updated to the latest HA, but rolling back to a backup didn't help.

**Expected behaviour**
BAU, detailed plan, etc.

**Predbat version**

8.8.9

**Environment details**

- Sunsynk ECCO 8kW
- Standard HAOS installer Addon
- Sunsynk Deye Addon to import inverter entities
- HA Core 2024.11.3
- Apps.yaml attached 
[apps.export.log](https://github.com/user-attachments/files/18157772/apps.export.log)

**Log file**
Attached
[predbat.log](https://github.com/user-attachments/files/18157734/predbat.log)

**Predbat debug yaml file**
Can provide if necessary

gcoan, 2024-12-17:
you can comment the export_triggers out in your apps.yaml if you are not using them

first thing to always check if Predbat goes weird is the apps.yaml file format because this can really screw up Predbat 

Looking at yours it looks like you have wrongly indented the charge start, stop and discharge start service.  No discharge stop service either, is that a separate problem?

[image]

The commands should be indented in alignment with everything else,i.e. two leading spaces

savethemanual, 2024-12-17:
Well that's embarrassing; amateur hour!
Thanks for replying 🙏

closed 2024-12-20 by springfall2008