springfall2008/batpred #1750: Battery not charging correctly overnight

richadlam, 2024-12-17:
**Describe the bug**
Battery is not charging correctly overnight.
no config changes have been made other than updating plugin / addon versions.
[image]

[image]

last working correctly around v8.6.0

also, have been getting a message from HA "The entity no longer has a state class"
[image]

**Expected behaviour**
Battery charges to 100% overnight during off peak import rate

**Predbat version**

v8.8.9

**Environment details**

HAOS v14.0
Core v2024.12.3
SolarEdge Modbus Multi v3.0.3
Octopus Energy v13.2.1
myenergi v0.0.29
Standard HAOS installer

[apps.yaml.txt](https://github.com/user-attachments/files/18163260/apps.yaml.txt)

**Log file**
[predbat.log.txt](https://github.com/user-attachments/files/18163264/predbat.log.txt)

richadlam, 2024-12-17:
to add to this, i've just watched as again the battery has failed to start charging when the plan says it should be.
[image]

i've managed to capture the logs for the time this happened. the charging window should have started at 2024-12-17 23:00:00
[predbatlog evening of 17-12-24.txt](https://github.com/user-attachments/files/18172445/predbatlog.evening.of.17-12-24.txt)

looking at the log, it seems that predbat thinks it activated the inverter to "charge from solar and grid" mode, however the "device_id" is missing from the command, and a warning to say the service call failed:

`2024-12-17 23:00:21.633797: Charge window will be: 2024-12-17 23:00:00+00:00 - 2024-12-18 01:30:00+00:00 - current soc 2 target 100
2024-12-17 23:00:21.634396: Adjust idle time, charge 23:00:00-01:30:00 discharge 00:00:00-00:00:00
2024-12-17 23:00:21.635005: Adjust demand (idle) time computed is 01:30:00-23:59:00
2024-12-17 23:00:21.635189: Inverter 0 set new demand (idle) start time to 01:30:00 was 09:50:00
2024-12-17 23:00:25.715372: Inverter 0 Wrote 01:30:00 to idle_start_time, successfully now 01:30:00
2024-12-17 23:00:25.715605: Inverter 0 set new demand (idle) end time to 23:59:00 was 23:00:00
2024-12-17 23:00:29.796380: Inverter 0 Wrote 23:59:00 to idle_end_time, successfully now 23:59:00
2024-12-17 23:00:29.796862: Inverter 0 Calling service charge_stop_service domain discharge with data {'device_id': ''}
2024-12-17 23:00:31.662047: Inverter 0 Calling service charge_start_service domain charge with data {'device_id': '', 'target_soc': 100, 'power': 5000}
2024-12-17 23:00:34.783070: Warn: Service call select/select_option failed with response None
2024-12-17 23:00:34.792600: Car charging from battery is off, next slot for car 0 is 12-17 23:00:00 - 12-18 00:30:00
2024-12-17 23:00:34.793143: Setting charging SOC to 100 as per target
2024-12-17 23:00:34.793393: Inverter 0 adjust target soc for charge to 100.0% based on requested all inverter soc 100%
2024-12-17 23:00:34.793591: Inverter 0 Current charge limit is 4% and new target is 100%
2024-12-17 23:00:38.874512: Inverter 0 Wrote 100 to charge_limit, successfully now 100
2024-12-17 23:00:38.875055: Current SOC 2% is less than Target SOC 100. Grid charging enabled with charge current set to 65.00
2024-12-17 23:00:38.875212: Inverter 0 count register writes 0`

the inverter state (solaredge_panels_storage_command_mode) was never altered in the HA logbook until I manually triggered it
[image]
[image]

I hope the added detail helps to diagnose the issue quickly! please let me know if you need any more detail

thanks!

jrviz, 2024-12-18:
I have Solaredge batteries and have seen the same problem over the last few days. No charging at all the night before last, a small amount of charging last night (up to 20%). The only changes I have made in the last few days is to apply several predbat updates. Happy to provide whatever log information would be useful.

PieBeard, 2024-12-20:
Adding my logs for last night and some images that might also contain the same issues which have been ongoing.

Plan of what was supposed to be happening 
[image]
What predate thought was happening,
[image]
What was happening
[image]
what predate was trying to do
[image]

Attached are the entity logs and then predate logs.
[Entity Log.txt](https://github.com/user-attachments/files/18210571/Entity.Log.txt)
[Predbat Log 20th Dec.txt](https://github.com/user-attachments/files/18210573/Predbat.Log.20th.Dec.txt)

springfall2008, 2024-12-21:
I'm wondering if this relates to the service being used timing out after a period of time? Predbat assumes once the charge service is called that charging will continue until the stop service is called.

richadlam, 2024-12-21:
> I'm wondering if this relates to the service being used timing out after a period of time? Predbat assumes once the charge service is called that charging will continue until the stop service is called.

Possibly, however often charging doesnt start at all for instance, the battery is at 0% at 23:30 when the cheap electricity rate starts, and predbat wants it to charge, and changes its state to charging, hoever the inverter doesnt get a call to start charging

last night the battery got 21 minutes of charge from 03:00, then the inverter reset to defaut settings and the house load consumed thast charge. the plan said it should be charging from 23:30

previously, predbat would notice this change in state due to the reset, oor that the inverter hadnt started charging and change it's state again, ensuring the charge happens

jrviz, 2024-12-21:
Similarly, I had a plan which should have started at 23:30 but there was no charging until 03:50 (in the log file above). I have not read into the detail of the log actions before but one difference I can see between 23:30 and 03:50 is that it is only in the later period there is an inverter write to set the inverter charge rate, there isn't one at 23:30.  The correct charge window is set. start_charge_service is called at both 23:30 and 03:50 but not at any time in between.

Charging was working without issues up until the session at 23:30 on the 15th. I have taken Predbat updates as they appeared and there have been several over this period. Could it be that something has chnaged which leads to Predbat concluding charging is not to be started? (although there have been some Home Assistant core/OS changes as well).

springfall2008, 2024-12-21:
Do you perhaps have charge_low_power on and and is it confusing things?

What does your charge_start_service do as it should start charging?

jrviz, 2024-12-21:
> Do you perhaps have charge_low_power on and and is it confusing things?
> 
> What does your charge_start_service do as it should start charging?

Is that "set_charge_low_power_mode" ? I do not have that on.

The solaredge charge_start_service  (from apps.yaml) is:
  charge_start_service:
    service: select.select_option
    entity_id: "select.solaredge_i1_storage_command_mode"
    option: "Charge from Solar Power and Grid"

As noted above the difference I can see between a successful start and a failed start is the issuing of a charge_rate write to the inverter prior to the charge_start_service. I'm not sure what triggers those inverter writes for charge rate?

richadlam, 2024-12-21:
My settings are the same as jrviz.

I can't see how to run that service from predbat, however when I manually change my storage command mode to charge from solar and grid then the battery starts to charge within 5 mins.

jrviz, 2024-12-24:
I had a look at the entities not being available but it was always going to be the solaredge multi integration where this was an issue. I initially hardwired these values in apps.yaml to see if this would fix the charging (unlikely and it didn't!) but I then fixed the entities by changing the storgae control mode in the solaredge integration to another value and then back to remote control and the entities became available again.

Last night I had charging start at the expected time (23:30) but then at 00:30 charging stopped. The solaredge integration had a change of storage command mode to "maximise self consumption". This is the command predbat issues to stop charging but I can't see this happening in the log where the stop command is logged as being skipped. (I wanted to get some charge in last night so I switched over to manual charging).

There are (at least) three actors here which can change the battery charging state. SE integration, SE online platform and Predbat. Right now I'm suspecting that it isn't predbat that is the problem so am examining ways to check out the other candidates.

For predbat is there more debug available which shows when there is a write to the inveter or the stop/start command that is issued?

springfall2008, 2024-12-26:
I suspect I've fixed the midnight issue in the latest release, can you re-test?

jrviz, 2024-12-26:
Is that a newer version than 8.8.14? I have set up ready to go for tonight and predbat has a plan to start at 23:30

jrviz, 2024-12-26:
Today I have also set up a rates_import_override period and saw charging of batteries start and stop OK (using 8.8.14)

springfall2008, 2024-12-26:
The fix was in 8.8.14, I forgot to mention it in the release notes. The service API was not called again after midnight meaning a charge than spanned midnight that was split into two sections would not work.

jrviz, 2024-12-26:
> The fix was in 8.8.14, I forgot to mention it in the release notes. The service API was not called again after midnight meaning a charge than spanned midnight that was split into two sections would not work.

What do you mean by split into two sections? If I have a charge period of 23:30 to 05:30 would I expect this change to make a difference?

springfall2008, 2024-12-26:
> If I have a charge period of 23:30 to 05:30 would I expect this change to make a difference?

Right, if the inverter is set as 'can't span midnight' then it would be split in two.

The warning seems to related to a select operation, is that in your charge_stop_service, what is it set as?

jrviz, 2024-12-26:
Yes, that was the call to charge_stop_service at the end of my test rates_import_override period. It did stop though.

How would I know if the inverter is set as can't span midnight.

I'll see how the run goes tonight and look at the logs. I'm still a bit suspicious of the SolarEdge online platform having changed behaviour and it is changing the battery mode (held as an entity in the SE Multi Integration) from "Remote Control" to "Maximise Self Consumption" for some reason. Remote Control is needed for the integration to work with Predbat or anything else that wants to control the charge state. If this happens then that also explains the reported entities that become unavailable

I have set up an automation to monitor state change on that entity and force it back to remote control and send me a notification if it happens.

PieBeard, 2024-12-27:
Still something odd going on with the charging.  Two nights ago battery did actually charge to 100% but different to how the plan stated it. Given it being Christmas didn't get chance to get dogs etc.  Last night charging didn't work correctly. Think Service calls didn't go through properly.

[image]
[image]
[image]
[image]
 
Here are logs and debug
[predbat-5.log](https://github.com/user-attachments/files/18258954/predbat-5.log)
[predbat_debug241227.yaml.txt](https://github.com/user-attachments/files/18259060/predbat_debug241227.yaml.txt)

jrviz, 2024-12-27:
Last night the battery charged for about an hour and then stopped as command mode changed back to "Maximise Self Consumption" which stops charging. I changed back manually several times during the night and it kept switching back. As I have stated above I think it is the SE online platform change of behaviour and thinking it knows best and changing the mode. I found this discussion on the SE Multi integration. https://github.com/WillCodeForCats/solaredge-modbus-multi/discussions/705

The Predbat commands are getting through but the online is overriding it. From the discussion referenced here there are a couple of strategies: 1. Change the Predbat command to use the Storage Default Mode service or 2.  Use an automation to monitor the Command Mode and change it to "Charge from Solar Power and Grid" while binary_sensor.predbat_charging is true.

I'll try both today. I have done a quick test with strategy 1 but I think I saw this being overridden. I'll test this again as I like this as the commands can all stay in apps.yaml Otherwise I'll set up strategy 2 with a separate automation.

jrviz, 2024-12-27:
There is a possible third strategy and that is Predbat issues (for this inverter) the start_charge command at the start of each 30min slot or every time the plan is run when charging should be active. I'm not sure what Trefor would think about that and I'm not sure I like it either as that would be getting Predbat to react to the behaviour of the online service - which has changed.

gcoan, 2024-12-27:
In general we are trying to reduce the number of inverter commands that Predbat sends to avoid potential issues with excessive flash memory writes https://springfall2008.github.io/batpred/caution/#flash-memory

Would seem better to try to understand what the online portal is doing and work with that - option 1?

jrviz, 2024-12-27:
Good point. I knew I didn't like option 3. I have set up 1 to control charging now and 2 as a watchdog to make sure everything behaves as expected. I will report back.

PieBeard, 2024-12-27:
I think if there was a check done at every compute that the entities doing the inverter control are in the state they are supposed to be at based on predate state then that can be the control on whether or not to re-write the inverter service command.  For Solaredge inverters there is a storage_command_mode which is linked to a timeout till default which is 3600s, this can be changed but reverts back to 3600 when the inverter resets twice a day. There is also storage_default_mode which I have assigned as what predate alters. Its reset behaviour I'm not too sure on.

gcoan, 2024-12-27:
> I think if there was a check done at every compute that the entities doing the inverter control are in the state they are supposed to be at based on predate state then that can be the control on whether or not to re-write the inverter service command

That is what Predbat does with my GivEnergy inverters, if you manually make an inverter config change then Predbat will change it back on the next run to match what it should be according to the plan.

Might be a bit more complex for other inverter types if the changes are made via service calls, but would resolve your problem

jrviz, 2024-12-27:
I have set up an automation that is triggered by mode entity changes from Charging to Self Consumption AND if predbat is signalling charging then the default mode will be set back to charging.

buttash, 2024-12-31:
Hi - I've fallen down this rabbit hole.  My battery is charging for an hour and then discharging.  I was wondering how you got on the service mode change and automation.  I've just my system up to follow this strategy.

jrviz, 2025-01-01:
Changing the service calls to change the Default Storage Mode rather than Storage Command Mode has worked for me. I have the automation described above as a backstop but it does not seem to be needed at the moment but I will leave it there to pick up any odd changes that might come from the SE online service making changes

buttash, 2025-01-02:
This also works for me.  The automations get called and default storage mode is set back correctly.

Thanks for the help!
Martin

PieBeard, 2025-01-03:
Things do seem to be working again now and reliably end the night on a full charge. Did set up a backup automation which I’m trying to work out if it does kick in. Need to wait till I have more time for that but it’s looking positive.

richadlam, 2025-01-05:
Agreed with everyone else now, things seem to be working correctly for the last few days / predbat updates. 
Thanks for your help, happy to close this issue now.

cyberkryten, 2025-01-05:
Aha, just done an upgrade of HA and Integrations with tons of changes and assumed I'd broken something - I'm seeing the same and couldn't figure out why it was resetting after the hour when it never used to!

Can you confirm what the 'working correctly' fix is - I see two options referenced: 

a) An automation that triggers the change back after a remote change from SE online or inverter reset (I can't see an example)
b) Set config to use Default Storage Mode instead of Storage Command Mode

I don't think this should be closed as the Docs should be updated to either 
  for a) include the suggested automation or 
  for b) change the example yaml which still references storage_command_mode

Without updating the docs, everyone using SolarEdge is going to get failed charge/discharge cycles.

As for the reduction in flash memory writes, whilst I understand it (mine was one of the early Model S cars that was bricked when flash failed), is it an issue for SolarEdge?  I am seeing this in the logs, so register writes aren't being incremented:

```
2025-01-05 11:25:18.238845: Inverter 0 count register writes 0
2025-01-05 11:25:18.263014: Total inverter register writes now 0
```

richadlam, 2025-01-05:
The automation I set is as follows, this accounts for the change in solaredge inverters resetting to defaults 2 times per day:
[image]

cyberkryten, 2025-01-05:
I also have the issue with the inverter resetting twice per day.  Before I did all the upgrades Predbat detected the reset and sorted it out (I was on 8.7.2).  Now, it doesn't, so I'm wondering if this is an intentional change in the more recent versions?

The irony here is that it is those sort of automations and trying to factor in lots of possibilities are what led me to Predbat in the first place!  Whilst your automation will sort the resets, it won't account for the reset after 3600 seconds when the timeout triggers (or if there are changes made from the SE portal side).

I do think the correct way to resolve this is for Predbat to check the current Inverter mode and if it is different to the 'Plan' mode, then set it to the correct value. 

Here's what used to happen from the Log of the Storage Command Mode on the Inverter after those resets with no automations running - unfortunately I don't have Predbat logs going back this far, but you can see that at the next 5min run, Predbat used to correct the mode to what it was set to before the reset occurred:

```
Old Behaviour:

23Dec
18:10:35 Solaredgemodbus I1 Storage Command Mode  changed to Charge from Solar Power and Grid triggered by action Select: Select
18:06:38 Solaredgemodbus I1 Storage Command Mode  changed to Maximize Self Consumption
18:06:17 Solaredgemodbus I1 Storage Command Mode  became unavailable
18:00:19 Solaredgemodbus I1 Storage Command Mode  changed to Charge from Solar Power and Grid

23Dec
18:25:44 Solaredgemodbus I1 Storage Command Mode  changed to Discharge to Maximize Export
18:22:26 Solaredgemodbus I1 Storage Command Mode  changed to Maximize Self Consumption
18:21:54 Solaredgemodbus I1 Storage Command Mode  became unavailable
17:25:43 Solaredgemodbus I1 Storage Command Mode  changed to Discharge to Maximize Export

22Dec
18:25:43 Solaredgemodbus I1 Storage Command Mode  changed to Discharge to Maximize Export
18:22:09 Solaredgemodbus I1 Storage Command Mode  changed to Maximize Self Consumption
18:21:38 Solaredgemodbus I1 Storage Command Mode  became unavailable
16:55:43 Solaredgemodbus I1 Storage Command Mode  changed to Discharge to Maximize Export
```

buttash, 2025-01-05:
If it helps, here are my two automations, one when predbat charging starts and one when it stops.  I put the logic into a loop with a 5min delay as I has a SolarEdge outage on my modbus plugin a couple of nights ago.

```
- id: '1735644669306'
  alias: 'SolarEdge: Set SDM Maximize Self Consumption'
  description: Set to maximise if Predbat is nolonger charging
  triggers:
  - trigger: state
    entity_id:
    - binary_sensor.predbat_charging
    to: 'off'
  conditions: []
  actions:
  - repeat:
      sequence:
      - action: select.select_option
        metadata: {}
        data:
          option: Maximize Self Consumption
        target:
          entity_id:
          - select.solaredge_i1_storage_default_mode
      - delay:
          hours: 0
          minutes: 5
          seconds: 0
          milliseconds: 0
      while:
      - condition: state
        entity_id: select.solaredge_i1_storage_default_mode
        state: Charge from Solar Power and Grid
      - condition: state
        entity_id: binary_sensor.predbat_charging
        state: 'off'
  mode: single
- id: '1735891498751'
  alias: 'SolarEdge: Set SDM Charge from Solar and Grid'
  description: Set SolarEdge to charge if Predbat is charging
  triggers:
  - trigger: state
    entity_id:
    - binary_sensor.predbat_charging
    to: 'on'
  conditions: []
  actions:
  - repeat:
      sequence:
      - action: select.select_option
        metadata: {}
        data:
          option: Charge from Solar Power and Grid
        target:
          entity_id:
          - select.solaredge_i1_storage_default_mode
          - select.solaredge_i1_storage_command_mode
      - delay:
          hours: 0
          minutes: 5
          seconds: 0
          milliseconds: 0
      while:
      - condition: state
        entity_id: binary_sensor.predbat_charging
        state: 'on'
      - condition: state
        entity_id: select.solaredge_i1_storage_default_mode
        state: Maximize Self Consumption
  mode: single
```

jrviz, 2025-01-05:
@cyberkryten I think @buttash and I are using both approaches to make sure we pick up all behaviours. 

The service calls in apps.yaml are:
  # Services to control charging/discharging
  charge_start_service:
    service: select.select_option
    entity_id: "select.solaredge_i1_storage_default_mode"
    option: "Charge from Solar Power and Grid"
  charge_stop_service:
    service: select.select_option
    entity_id: "select.solaredge_i1_storage_default_mode"
    option: "Maximize Self Consumption"
  discharge_start_service:
    service: select.select_option
    entity_id: "select.solaredge_i1_storage_default_mode"
    option: "Discharge to Maximize Export"

This is a modification of the original document to change Storage Default Mode rather than Storage Command Mode. This makes the inverter keep charging if there are some types of change. 

The automations are there to pick up conditions which are not as expected. I had something very similar to the ones above but I like the loop to cover Modbus outages so I have adopted that :-)

Yes, it is a bit of a pain to have the extra automations but that is because of the behaviour of the SolarEdge Online service/Inverter and not a fault of Predbat. I'm not sure Predbat can check the status of every Inverter type. It makes the services call to stop and start charging but assumes those will not be changed from elsewhere - which does happen for SolarEdge unfortunately. 

I would expect that to build this into Predbat there would need to be a general model that could support all inverter types to read state and be told what the expected state should be in charge, discharge and idle conditions.

cyberkryten, 2025-01-07:
If this is specific to SolarEdge and doesn't affect any other inverters at all after resets/updates, then maybe it is best handled with extra automations instead of adding anything to Predbat.

My suggestion of Predbat checking/changing was really so that others could use Predbat easily without having to mess about with more complexity and setting up automation watchdogs - if that's what we recommend for SolarEdge users then I think we ought to update the Docs and provide examples.  Given that Predbat runs every X minutes anyway, seemed  to me to make sense to define a 'inverter_check_service' call, or similar, that would let Predbat check the inverter is in the state that it wants and then allow it to make the change if it is not.

One other thought is around the recent issues relating to register writes and wear, do we know if there's a difference in writing to Storage Command Mode (which seems to be transient, so probably RAM) and Storage Default Mode (which seems to be persistent so more likely to be a more permanent storage which may be flash)?  I can't find anything in the SE Docs saying how/where it is written and what any wear characteristics are for changes, but there could be an issue lurking down the road (register writes always shows zero for my inverter in Predbat so I'm not sure if that means SE isn't affected or it just isn't being counted).

jrviz, 2025-01-07:
It's now clear that the problem is caused by SE either as a timeout or some action of the online platform. I had a few nights when the automation did not trigger but it has on the last two nights.

If there is this automation to watch for the Storage Command Mode not being in the charging state when Predbat is indicating charging then if there is a worry about register writes then perhaps its worth an experiment where the service calls could revert to Storage Command Mode from Storage Default Mode or will the setting of Storage Default Mode, if different, immediately overwrite out automation?

jrviz, 2025-01-07:
That's an interesting point. I did wonder if that change might be related to the issues which SE inverters started to have. Maybe the old Predbat behaviour was masking the SE timeouts, resets etc. by calling the charge service regularly in the charge/discharge(?) window?

If that was the case then a switch to allow repeated charge/discharge calls in the charge/discharge window rather than call once and skip could be useful to keep all the commands inside Predbat rather than having supporting automations - Would depend on that being OK for other inverter types as a reversion of #1575 (which removed "duplicate" calls) and be backwards compatible as a non breaking change. @springfall2008 ?

In the meantime I have changed my apps.yaml service calls back to being command_mode based again and the same for my monitoring automations to see if those will work.

cyberkryten, 2025-01-07:
Yes, I do think that, unintentionally, #1575 was a breaking change for pretty much all SE Inverters.  The previous behaviour definitely resolved the issues with the SE inverter timeouts and resets and also would have quickly fixed any interference from the SE online platform.

Predbat regularly changing the mode back to the correct one was one of the great benefits I found when I switched to it, as I then deleted all my own automations to check the mode and change it to the correct Charge/Export status.

springfall2008, 2025-01-07:
Should I make a change to allow repeated service calls for some inverters?

Also could your service call change the mode to the correct mode?

jrviz, 2025-01-07:
The service call for charge/discharge would put the inverter into the correct mode if it is called regularly after the start of a period of activity. This would recover from resets, timeouts etc. on the SolarEdge inverter. Would it be specific inverter code or a switch to allow repeat service calls rather than skipping during a period of activity? What would happen after the end of a period of charge/discharge when the stop_service is called? For SolarEdge repeat calling the stop services would be the right mode for the inverter but that would go on all day which seems unnecessary. @cyberkryten any builds on this?

springfall2008, 2025-01-07:
I've just put in a pull request for this, it will allow you to add repeat to charge_start_service and discharge_start_service but you can skip putting repeat on the stop service if you like.

springfall2008, 2025-01-07:
Code on main if you want to test it:
```yaml
charge_start_service:
   service: select.select_option
   entity_id: "select.solaredge_i1_storage_default_mode"
   option: "Charge from Solar Power and Grid"
   repeat: True
```

jrviz, 2025-01-07:
I don't know how to test from main. Do I just copy in the modified python files?

gcoan, 2025-01-07:
> I don't know how to test from main. Do I just copy in the modified python files?

you can copy the python files, or in Home Assistant, open the control select.predbat_version and select 'main' from the list and predbat will update itself to that version

jrviz, 2025-01-07:
Thanks. All set up for testing tonight.

cyberkryten, 2025-01-08:
OK, an hour after the inverter was set to Charge, it's still there with no automation triggered so the 'always' flag is working :)

However, I'm seeing duplicate calls and some are failing with a Warn message.  I'd noticed two calls happening earlier today when I added all the logging so it isn't new from this change.  Earlier today I'd seen it change to Self Consumption on the first and then Charge/Export on the second.

```
2025-01-07 23:45:05.486883: Inverter 0 Skipped service charge_stop_service domain discharge service_name select.select_option as it was previously called.
2025-01-07 23:45:05.486897: Inverter 0 Calling service charge_start_service domain charge service_name select/select_option with data {'entity_id': 'select.solaredgemodbus_i1_storage_command_mode', 'option': 'Charge from Solar Power and Grid'}
2025-01-07 23:45:06.725903: Inverter 0 Skipped service charge_stop_service domain discharge service_name select.select_option as it was previously called.
2025-01-07 23:45:16.262848: Inverter 0 Skipped service charge_stop_service domain discharge service_name select.select_option as it was previously called.
2025-01-07 23:45:16.262862: Inverter 0 Calling service charge_start_service domain charge service_name select/select_option with data {'entity_id': 'select.solaredgemodbus_i1_storage_command_mode', 'option': 'Charge from Solar Power and Grid'}
2025-01-07 23:45:16.430707: Inverter 0 Skipped service charge_stop_service domain discharge service_name select.select_option as it was previously called.
2025-01-07 23:50:06.128618: Inverter 0 Skipped service charge_stop_service domain discharge service_name select.select_option as it was previously called.
2025-01-07 23:50:06.128631: Inverter 0 Calling service charge_start_service domain charge service_name select/select_option with data {'entity_id': 'select.solaredgemodbus_i1_storage_command_mode', 'option': 'Charge from Solar Power and Grid'}
2025-01-07 23:50:09.137902: Warn: Service call select/select_option data {'entity_id': 'select.solaredgemodbus_i1_storage_command_mode', 'option': 'Charge from Solar Power and Grid'} failed with response None
2025-01-07 23:50:09.139435: Inverter 0 Skipped service charge_stop_service domain discharge service_name select.select_option as it was previously called.
```

Debug from the SolarEdgeModbusMulti integration shows calls are made to set the state every run, twice on some and only once on the ones with the failure:

```
2025-01-07 23:45:05.491 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.select] set SE8000H-RWBMNBF54_750AF470_storage_command_mode to Charge from Solar Power and Grid
2025-01-07 23:45:06.340 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0x3 <class 'int'>
2025-01-07 23:45:16.267 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.select] set SE8000H-RWBMNBF54_750AF470_storage_command_mode to Charge from Solar Power and Grid
2025-01-07 23:45:17.401 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0x3 <class 'int'>
2025-01-07 23:45:33.195 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0x3 <class 'int'>
2025-01-07 23:45:49.173 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0x3 <class 'int'>
2025-01-07 23:46:05.290 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0x3 <class 'int'>
2025-01-07 23:46:21.063 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0x3 <class 'int'>
2025-01-07 23:46:37.059 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0x3 <class 'int'>
2025-01-07 23:46:53.053 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0x3 <class 'int'>
2025-01-07 23:47:09.152 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0x3 <class 'int'>
2025-01-07 23:47:25.145 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0x3 <class 'int'>
2025-01-07 23:47:41.140 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0x3 <class 'int'>
2025-01-07 23:47:57.069 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0x3 <class 'int'>
2025-01-07 23:48:13.121 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0x3 <class 'int'>
2025-01-07 23:48:29.398 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0x3 <class 'int'>
2025-01-07 23:48:45.154 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0x3 <class 'int'>
2025-01-07 23:49:01.148 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0x3 <class 'int'>
2025-01-07 23:49:17.124 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0x3 <class 'int'>
2025-01-07 23:49:33.221 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0x3 <class 'int'>
2025-01-07 23:49:49.085 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0x3 <class 'int'>
2025-01-07 23:50:05.590 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0x3 <class 'int'>
[... cut ...]
2025-01-07 23:50:22.497 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0x3 <class 'int'>
2025-01-07 23:50:38.164 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0x3 <class 'int'>
2025-01-07 23:50:54.290 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0x3 <class 'int'>
2025-01-07 23:51:10.145 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] I1: command_mode 0x3 <class 'int'>
```

jrviz, 2025-01-08:
I had a successful charge last night with my automations off and using "always". I just had one switch back to not charging during the night which was recovered by a charge start.

I'm seeing two calls to charge_start on 10 minute boundaries  (h:00, h:10, h:20...) but only one on 5 minute boundaries (h:05, h:15...). Are there two passes running on these boundaries?

The only times I have seen Warnings in the logs for service calls no response  have been on Predbat restarts after I made changes to apps.yaml but I will keep an eye open

buttash, 2025-01-08:
Glad to hear that worked for you - before I switch my service calls to storage_command_mode from storage_default_mode can you point to me to what  'always: True' means?  I couldn't see that in the documentation - apologies if it is there and I didn't look hard enough :(

jrviz, 2025-01-08:
It's an update to allow service calls to be repeated at each Predbat cycle during a charge/discharge active window. This is only in a test version of Predbat in main currently and not yet released. Documented above by @springfall2008  but there was a slight typo. For service calls you want to repeat to cope with Inverter reset, timeout etc. (start_charge, start_discharge) you would have:

charge_start_service:
   service: select.select_option
   entity_id: "select.solaredge_i1_storage_default_mode"
   option: "Charge from Solar Power and Grid"
   always: True

For those that don't need it (stop_charge stop_discharge) you leave out the always clause.

cyberkryten, 2025-01-08:
Overnight things worked fine, mode stayed as expected for the long charging period so I'm pretty sure this has fixed the issue, though my automation caught the inverter reset before Predbat had a chance to fix it (though I'm sure it would have).  

I've disabled all automations relating to checks/fixing the inverter state now and will monitor.

As this was a breaking change for systems already setup, is it possible to default to 'always: True' for charge_start_service and discharge_start_service IF the Inverter type is SE?  I'm not sure if it can be done selectively like that - if not, then having it for all services would be fine and those of use that know can disable it for the Stop services.

I think this will have affected people who aren't active here, and may have had Predbat working for some time with a SolarEdge Inverter - this change will have broken long charge/discharge periods and unless they re-read the docs they won't know how/where to fix it, so if we can make it the default for SE inverters then it will fix those systems automatically.  It doesn't matter for those of us reading this thread, so if it can't be done then that's fine, but being able to revert all the other SE systems to how they used to work seems like it is worth considering.

I'm not sure what happens with all the other inverter types, I don't see other issues raised so I'm guessing they don't suffer from the resets/timeouts that SE has.

jrviz, 2025-01-08:
@cyberkryten are you seeing two start_charge calls on round 10 minute run boundaries and only one on the interim 5 minute ones?

cyberkryten, 2025-01-08:
> @cyberkryten are you seeing two start_charge calls on round 10 minute run boundaries and only one on the interim 5 minute ones?

Ah, good point.  Yes I'm seeing one on 5min when the plan is not recalculated and two on 10min.

The first appears early on and then there's a second after the plan is recalculated - that explains why at times I've seen different results with the first being Self Consumption and the second Export - it first sets it to 'what my current intent it' and then after the recalculation, sets it to 'what my new plan is'. 

The second one, after a new plan is created, always needs to be done, but the problem with removing the first one is that the first call happens before the check to see if the plan is going to be recalculated - it needs to be done (otherwise we'll be waiting 10min for the inverter to reset to the correct mode instead of five).

I guess maybe the check could be added earlier, but that's a BIG change and could cause a lot of problems!

```
2025-01-08 14:05:05.705079: Inverter 0 Calling service charge_start_service domain charge service_name select/select_option with data {'entity_id': 'select.solaredgemodbus_i1_storage_command_mode', 'option': 'Charge from Solar Power and Grid'}
2025-01-08 14:05:07.080456: Will not recompute the plan, it is 5.0 minutes old and max age is 10 minutes
2025-01-08 14:10:05.482593: Inverter 0 Calling service charge_start_service domain charge service_name select/select_option with data {'entity_id': 'select.solaredgemodbus_i1_storage_command_mode', 'option': 'Charge from Solar Power and Grid'}
2025-01-08 14:10:06.622899: Will recompute the plan as it is now 10.0 minutes old and will exceed the max age of 10 minutes before the next run
2025-01-08 14:10:26.842257: Inverter 0 Calling service charge_start_service domain charge service_name select/select_option with data {'entity_id': 'select.solaredgemodbus_i1_storage_command_mode', 'option': 'Charge from Solar Power and Grid'}
2025-01-08 14:15:05.623853: Inverter 0 Calling service charge_start_service domain charge service_name select/select_option with data {'entity_id': 'select.solaredgemodbus_i1_storage_command_mode', 'option': 'Charge from Solar Power and Grid'}
2025-01-08 14:15:06.933015: Will not recompute the plan, it is 5.0 minutes old and max age is 10 minutes
2025-01-08 14:20:05.321200: Inverter 0 Calling service charge_start_service domain charge service_name select/select_option with data {'entity_id': 'select.solaredgemodbus_i1_storage_command_mode', 'option': 'Charge from Solar Power and Grid'}
2025-01-08 14:20:06.626341: Will recompute the plan as it is now 10.0 minutes old and will exceed the max age of 10 minutes before the next run
2025-01-08 14:20:26.798039: Inverter 0 Calling service charge_start_service domain charge service_name select/select_option with data {'entity_id': 'select.solaredgemodbus_i1_storage_command_mode', 'option': 'Charge from Solar Power and Grid'}
```

jrviz, 2025-01-08:
Just a bit of thinking out loud :-) I guess we go to this point because Predbat does not check inverter status as a condition for the service call. The current once and skip model causes issues with SE timeouts/resets. The always model result in redundant calls and raises a possible concern about inverter writes. 

Is there an approach where the service api calls are not directly to selects on the SE Modbus entity but to a script which has the logic to check current status and minimise writes? Yes, it is logic outside of Predbat but could be better than automations that are "snooping" from the sidelines for unexpected states.

jrviz, 2025-01-08:
I was thinking something like this called for the  start_charge service api to check status before a write to the SE Multi entity (apologies, very crude draft as this is new to me so this may need more work)

sequence:
  - condition: not
    conditions:
      - condition: state
        entity_id: select.solaredge_i1_storage_command_mode
        state: Charge from Solar Power and Grid
  - action: select.select_option
    metadata: {}
    data:
      option: Charge from Solar Power and Grid
    target:
      entity_id: select.solaredge_i1_storage_command_mode
alias: SE Inverter Start Charge
description: ""

cyberkryten, 2025-01-08:
Here's an example where it isn't behaving as I'd expect - the plan changed from Export to Charge - this may be a legacy of how other inverters work, but as we're only changing the one parameter in SE there is no need to make the charge_stop_service call to reset to Maximise Self Consumption before making the charge_start_service call to go into Charge from Solar Power and Grid mode:

```
2025-01-08 15:25:05.420180: Inverter 0 Skipped service charge_stop_service domain charge service_name select.select_option as it was previously called.
2025-01-08 15:25:05.420201: Inverter 0 Calling service discharge_start_service domain discharge service_name select/select_option with data {'entity_id': 'select.solaredgemodbus_i1_storage_command_mode', 'option': 'Discharge to Maximize Export'}
2025-01-08 15:25:06.585630: Inverter 0 Skipped service charge_stop_service domain charge service_name select.select_option as it was previously called.

2025-01-08 15:30:42.133205: Inverter 0 Calling service charge_stop_service domain discharge service_name select/select_option with data {'entity_id': 'select.solaredgemodbus_i1_storage_command_mode', 'option': 'Maximize Self Consumption'}
2025-01-08 15:30:44.108976: Inverter 0 Calling service charge_start_service domain charge service_name select/select_option with data {'entity_id': 'select.solaredgemodbus_i1_storage_command_mode', 'option': 'Charge from Solar Power and Grid'}
2025-01-08 15:30:44.328025: Inverter 0 Skipped service charge_stop_service domain discharge service_name select.select_option as it was previously called.
2025-01-08 15:30:55.497719: Inverter 0 Skipped service charge_stop_service domain discharge service_name select.select_option as it was previously called.
2025-01-08 15:30:55.497732: Inverter 0 Calling service charge_start_service domain charge service_name select/select_option with data {'entity_id': 'select.solaredgemodbus_i1_storage_command_mode', 'option': 'Charge from Solar Power and Grid'}
2025-01-08 15:30:55.881931: Inverter 0 Skipped service charge_stop_service domain discharge service_name select.select_option as it was previously called.
```

leacho73, 2025-01-08:
Just 'updated' to the main branch, and now I have lost all control, the battery should be charging now ready for the saving session and keeps going back to 'maximise self consumtion' I've added the following in my apps.yaml (which can I confirm is still located in the root of the predbat folder, or does this now need to move to the config folder?)

` # Services to control charging/discharging
  charge_start_service:
    service: select.select_option
    entity_id: "select.solaredge_i1_storage_command_mode"
    option: "Charge from Solar Power and Grid"
    always: true
  charge_stop_service:
    service: select.select_option
    entity_id: "select.solaredge_i1_storage_command_mode"
    option: "Maximize Self Consumption"
    always: true
  discharge_start_service:
    service: select.select_option
    entity_id: "select.solaredge_i1_storage_command_mode"
    option: "Discharge to Maximize Export"
`

Nothing in the log to say its even attempted to do anything

**** Starting Standalone Predbat ****
2025-01-08 16:41:30.334660: Loading apps.yaml
2025-01-08 16:41:30.378706: Info: Connected to Home Assistant at http://supervisor/core
2025-01-08 16:41:30.381444: Info: Add-on slug is 6adb4f0d_predbat
2025-01-08 16:41:30.381509: Creating task: <coroutine object HAInterface.socketLoop at 0x7f9016c4dfc0>
2025-01-08 16:41:30.381864: Info: Web Socket task started
2025-01-08 16:41:30.382079: Creating task: <coroutine object WebInterface.start at 0x7f9016c4df50>
2025-01-08 16:41:30.382399: Info: Start socket for url http://supervisor/core/api/websocket
Web interface started
2025-01-08 16:41:30.398837: Info: Web Socket active
2025-01-08 16:41:30.493858: Warn: Regular expression argument: octopus_intelligent_slot unable to match re:(binary_sensor.octopus_intelligent_slot|re:binary_sensor.octopus_energy_a_680df942_intelligent_dispatching), now will disable
2025-01-08 16:41:30.505295: Warn: Regular expression argument: octopus_free_session unable to match re:(event.octopus_energy_([0-9a-z_]+|)_octoplus_free_electricity_session_events), now will disable
Watching ['./download.py', './unit_test.py', './predbat.py', './config.py', './gecloud.py', './hass.py', './ui.py', './predheat.py', './futurerate.py', './output.py', './solcast.py', './execute.py', './plan.py', './apps.yaml', './userinterface.py', './utils.py', './inverter.py', './web.py', './octopus.py', './prediction.py', './fetch.py', './ha.py', './unzip/batpred-7.22.2/apps/predbat/predbat.py', './unzip/batpred-7.22.2/apps/predbat/config/apps.yaml'] for changes
2025-01-08 16:41:47.223113: Warn: Historical day 7 has 205 minutes of gap in the data, filled from 25.33 kWh to make new average 29.53 kWh (percent 86%)
2025-01-08 16:41:47.235671: Warn: Historical day 14 has 450 minutes of gap in the data, filled from 19.92 kWh to make new average 28.97 kWh (percent 69%)
2025-01-08 16:42:14.344711: Info: record_status Charging target 1%-100%
2025-01-08 16:45:14.185617: Warn: Historical day 7 has 205 minutes of gap in the data, filled from 25.32 kWh to make new average 29.52 kWh (percent 86%)
2025-01-08 16:45:14.200717: Warn: Historical day 14 has 450 minutes of gap in the data, filled from 19.99 kWh to make new average 29.07 kWh (percent 69%)
2025-01-08 16:45:26.480263: Info: record_status Charging target 2%-100%
2025-01-08 16:50:14.324305: Warn: Historical day 7 has 205 minutes of gap in the data, filled from 25.31 kWh to make new average 29.51 kWh (percent 86%)

cyberkryten, 2025-01-08:
@leacho73 Was this all working before you updated?  From the look of it, Predbat can't determine your IOG slots and that means it won't know when to charge.  This change kicks in later and so you have other problems not caused by this update (unless there were other changes pushed to main with this one).  

I'd try a HA restart, check your Octopus integration is working and if not then open a new ticket with full debug logs.

The 'failed with response None' has been seen by a few of us - not sure what it is yet, I'm away for a couple of days so won't get chance to debug until next week now.

leacho73, 2025-01-09:
> @leacho73 Was this all working before you updated?  From the look of it, Predbat can't determine your IOG slots and that means it won't know when to charge.  This change kicks in later and so you [...]

Hi @cyberkryten - yes all working before - no issues apart from the SE inverter mode disconnect - woke up to 25% battery this morning rather than 100% due to the inverter resetting overnight and predbat not seeing it.

Re: octopus IOG - not sure why that logs - as it picks up my slots absolutely fine.

Cheers

buttash, 2025-01-09:
Everything worked well last night.  Command mode set to charge from 23:30 to 5:30 and default mode unchanged from maximize. Battery fully charged by 2:30. I had automations turned off to set mode. The two automations to detect a change for 30s by @cyberkryten did not trigger.

[8 more replies]

closed 2026-09-09 by chalfontchubby