springfall2008/batpred #1753: Documentation updates for missing or out of date customisation controls

nbullus, 2024-12-18:
**Describe the bug**

There are two switches which are undocumented.  An explanation of their behaviour would be useful for users:

- switch.predbat_calculate_secondary_order
- switch.predbat_inverter_set_charge_before

The following controls are described in the documentation but cannot be found under export mode controls (may be obsolete or hardcoded):

- switch.predbat_calculate_regions
- input_number.predbat_charge_low_power_margin

**Expected behaviour**

Updated documenation

**Predbat version**

v8.8.9 onwards

**Environment details**

N/A

**Screenshots**

N/A

**Log file**

N/A

**Predbat debug yaml file**

N/A

springfall2008, 2024-12-18:
Updated on main, will be in the next release

nbullus, 2024-12-19:
Latest documentation looks fine now and splitting the secondary order settings is really useful. Tried this new setting and helping push exports as late as possible now. Issue can be closed.

closed 2024-12-19 by gcoan