springfall2008/batpred #1761: Getting more from Solar Assistant

xhemp, 2024-12-19:
Decided to open a feature request from Solar Assistant.
@sparkyhughes managed to enable has_target_soc and support_charge_freeze.
I have been testing on my side and now got just one error which is weird: 

`
Warn: Inverter 0 write to charge_limit failed
`

Although it's weird as I am able to change it on Home Assistant and Solar Assistant accepts it.

xhemp, 2024-12-19:
Checking if @springfall2008 can have a look at this.

xhemp, 2024-12-19:
@sparkyhughes I can see my inverter trying to charge the battery with the grid after a few minutes even though battery first grid charge is set to disabled, can you please check if you see the same?

sparkyhughes, 2024-12-19:
Does Solar-Assistant Inverter settings show "Battery first grid charge"  as  "Disabled"? - in SA under Power menu -> Inverter Settings
Enabled and Disabled start with a capital letter, so went with that in the apps.yaml file

xhemp, 2024-12-19:
Yes, it's showing Disabled on HA and SA, it works at first but then it starts charging from the battery after a few minutes which is strange, I am retesting now and will record a video.

sparkyhughes, 2024-12-19:
I know without the stop charge value being changed from 100, if there is enough solar, and the inverter is on Battery first mode, it will charge the battery.  It shouldn't from the grid though

xhemp, 2024-12-19:
Can't replicate, will wait until a Charge Freeze again to test!

sparkyhughes, 2024-12-20:
One other SA / Growatt issue I've seen overnight relates to Charge Hold.
When this was activated last night, it stops the charge (changes the Growatt mode to Load First), sets the target SOC to 10% and sets the discharge rate to 0W.  
However, Growatt Inverters (mine is SPH model) doesn't have a discharge rate for Load First mode, so essentially the battery discharges into the car.

```
2024-12-20 03:00:28.523696: Inverter 0 current discharge rate is 3000.0W and new target is 0W
2024-12-20 03:00:33.586022: Inverter 0 Wrote 0 to discharge_rate, successfully now 0
2024-12-20 03:00:33.994493: Disabling battery discharge while the car 0 is charging
2024-12-20 03:00:33.994965: Inverter 0 Calling service charge_stop_service domain charge with data {'device_id': ''}
2024-12-20 03:00:34.029136: Not setting charging SoC as we are not within the window (now 12-20 03:00:00 target set_soc_minutes 30 charge start time 12-21 00:30:00)
2024-12-20 03:00:34.029371: Inverter 0 adjust target soc for export to 0.0% based on requested all inverter soc 0%
2024-12-20 03:00:34.029493: Inverter 0 Current charge limit is 100% and new target is 10%
2024-12-20 03:00:44.140764: Inverter 0 Wrote 10 to charge_limit, successfully now 10
2024-12-20 03:00:44.452011: Inverter 0 count register writes 1
```

I think from SA perspective, certainly with Growatt is that we use Charge Freeze instead.  This I know works, as the Target SOC is changed to the current level.  This in my apps.yaml file with SA changes the inverter Battery First Grid Charge to Disabled, but doesn't change the mode to Load First.  The effect is the Battery SOC remains at the current level and the load is taken from the Grid - just like if the battery reaches the lower limit of 10% towards the end of the day (I only have a small battery)

I'm not sure how Batpred is choosing to use Charge Hold over Charge Freeze.  I've turned off Control Slider "Car Charging Hold" which was previously set to On to see if this makes a difference.  Can't find in the documentation much about Charge Hold vs Charge Freeze

xhemp, 2024-12-21:
I don't have an electric car so I can't test this, unfortunately.
I've been getting a lot of messages saying it can't change the charge_limit which is very odd as I can change it via HA with no issues, not sure what's going on. Tried to set it to debug mode but can't see exactly what's going on.

```
2024-12-21 02:01:09.359404: Warn: Inverter 0 Trying to write 10 to charge_limit didn't complete got 100
2024-12-21 02:01:09.470906: Warn: record_status Warn: Inverter 0 write to charge_limit failed
2024-12-21 02:02:52.644705: Warn: Inverter 0: Using inverter type SA - not all features are available
2024-12-21 02:03:44.012834: Warn: Inverter 0 Trying to write 15 to charge_limit didn't complete got 10
2024-12-21 02:03:44.133856: Warn: record_status Warn: Inverter 0 write to charge_limit failed
2024-12-21 02:04:54.865189: Warn: Inverter 0: Using inverter type SA - not all features are available
2024-12-21 02:07:08.004834: Warn: Inverter 0: Using inverter type SA - not all features are available
2024-12-21 02:07:38.788568: Warn: Inverter 0 Trying to write 14 to charge_limit didn't complete got 15
2024-12-21 02:07:38.839731: Warn: record_status Warn: Inverter 0 write to charge_limit failed
2024-12-21 02:08:12.849160: Warn: Inverter 0: Using inverter type SA - not all features are available
2024-12-21 02:08:51.256066: Warn: Inverter 0 Trying to write 14 to charge_limit didn't complete got 15
2024-12-21 02:08:51.322642: Warn: record_status Warn: Inverter 0 write to charge_limit failed
2024-12-21 02:10:11.770721: Warn: Inverter 0: Using inverter type SA - not all features are available
2024-12-21 02:12:45.456445: Warn: Inverter 0: Using inverter type SA - not all features are available
2024-12-21 02:13:07.027792: Info: record_status Freeze charging target 14%
2024-12-21 02:15:15.985962: Warn: Inverter 0: Using inverter type SA - not all features are available
2024-12-21 02:15:52.965944: Warn: Inverter 0 Trying to write 16 to charge_limit didn't complete got 15
2024-12-21 02:15:53.030086: Warn: record_status Warn: Inverter 0 write to charge_limit failed
2024-12-21 02:20:14.309894: Warn: Inverter 0: Using inverter type SA - not all features are available
2024-12-21 02:20:51.725835: Warn: Inverter 0 Trying to write 17 to charge_limit didn't complete got 16
2024-12-21 02:20:51.784122: Warn: record_status Warn: Inverter 0 write to charge_limit failed
2024-12-21 02:23:22.476683: Warn: Inverter 0: Using inverter type SA - not all features are available
2024-12-21 02:24:13.907828: Warn: Inverter 0 Trying to write 10 to charge_limit didn't complete got 17
2024-12-21 02:24:13.949810: Warn: record_status Warn: Inverter 0 write to charge_limit failed
2024-12-21 02:25:14.278755: Warn: Inverter 0: Using inverter type SA - not all features are available
2024-12-21 02:25:38.806418: Info: record_status Demand
2024-12-21 02:30:17.517903: Warn: Inverter 0: Using inverter type SA - not all features are available
2024-12-21 02:31:06.820397: Warn: Inverter 0 Trying to write 100 to charge_limit didn't complete got 10
2024-12-21 02:31:06.927059: Warn: record_status Warn: Inverter 0 write to charge_limit failed
2024-12-21 02:34:37.379964: Warn: Inverter 0: Using inverter type SA - not all features are available
2024-12-21 02:35:17.785934: Warn: Inverter 0: Using inverter type SA - not all features are available
2024-12-21 02:35:49.966656: Info: record_status Charging target 19%-100%
```

sparkyhughes, 2024-12-21:
Do you have charge_limit defined in apps.yaml pointing at the HA entity for the Growatt battery first target soc, and in config.py you have has_target_soc set to True?

sparkyhughes, 2024-12-21:
@springfall2008 I think for Growatt inverters via SA, they do not support a discharge rate setting whilst in Battery First or Load First modes. This means that charge hold does not work. 
I would suggest you expose a charge hold service like the charge start, stop and freeze. I can then set the inverter to Battery First, disable grid charge and if you set the target soc as the current level that should work perfectly. 
I know this works for charge freeze 

is this possible?

xhemp, 2024-12-21:
> Do you have charge_limit defined in apps.yaml pointing at the HA entity for the Growatt battery first target soc, and in config.py you have has_target_soc set to True? 

I do, yes. Only difference I believe between you and mine is tat I have an SPA and not a SPH.

xhemp, 2024-12-22:
Yes, need help on this one, there's something odd happening that sometimes doesn't set the inverter to what it needs to do, no idea why, though, I went through every single setting, set debug to see if I could find where the issue lies but had no luck.

sparkyhughes, 2024-12-22:
Can we assign this? I am not able to

xhemp, 2024-12-22:
Don't think I can, unfortunately.

sparkyhughes, 2024-12-22:
It's because we are not collaborators.  You're a contributor and I'm a nobody!
Reliant on @springfall2008 to see this and pick it up, unless anybody else can assign?

xhemp, 2024-12-22:
Maybe an issue instead of an enhancement gets assigned to him?

springfall2008, 2024-12-28:
Sorry what is the question here?

xhemp, 2024-12-28:
> Sorry what is the question here?

@sparkyhughes managed to enable has_target_soc and support_charge_freeze.
I have been testing on my side and now got just one error which is weird:

2024-12-19 13:10:51.958903: Warn: Inverter 0 Trying to write 67 to charge_limit didn't complete got 66
2024-12-19 13:10:52.024017: Warn: record_status Warn: Inverter 0 write to charge_limit failed

Although it's weird as I am able to change it on Home Assistant and Solar Assistant accepts it.
And it looks to me it's indeed being accepted but PredBat says it is not, interesting.

sparkyhughes, 2024-12-29:
@springfall2008 We’ve got two enhancement requests on here which isn’t probably very helpful! Sorry. 
I think for Growatt inverters via SA, they do not support a discharge rate setting whilst in Battery First or Load First modes. This means that charge hold does not work.
I would suggest you expose a charge hold service like the charge start, stop and freeze. I can then set the inverter to Battery First, disable grid charge and if you set the target soc as the current level that should work perfectly.
I know this works for charge freeze

is this possible?

I did a HA workaround which has a trigger on the Inverter 0 discharge rate. When it is set to 0 I change the Load First discharge stop value to 100 (100%). When it is set to the max value (3000 in my case) I change the load first discharge stop value to 10 (10%)

xhemp, 2024-12-29:
The one I described above looks to me more of an issue seeing HA accepts the change but predbat says it doesn't. Maybe a different response?  I've tried to set debug to on but it doesn't capture responses, AFAIK. Any debug logs I can send, please, let me know

xhemp, 2025-01-22:
Any chance you can look at this, @springfall2008 ? I can test anything you'd like on my side.
Thank you!

xhemp, 2026-03-09:
Looks like the SPA doesn't have, I found an old screenshot of the work mode on SA that didn't show that, unfortunately.
With regards to the RS485 cable, looks like there's a connector that covers all those connections, including the RS485 so I am using a RS232 to connect to the PI and I had to get rid of the ShineWiFi one, it's not an issue.
I will try to play a bit more with your configuration and see how it goes, I will let you know!

chalfontchubby, 2026-09-18:
Going to suggest we close this issue as it's unclear what (or how many) the requests are. 

Can you open issues for any specific asks we can fix please.

xhemp, 2026-09-18:
It's working well on my side, no worries!

closed 2026-09-18 by xhemp