springfall2008/batpred #1762: Ignore intelligent slots if car is unplugged switch not working

mpartington, 2024-12-19:
**Describe the bug**
Predbat uses the low cost IOG slots when car is unplugged instead of ignoring them

**Expected behaviour**
To ignore IOG slots if car is unplugged i.e. keep the battery in 'Demand' mode

switch.predbat_octopus_intelligent_ignore_unplugged (expert mode) - used with Octopus Intelligent to prevent Predbat from assuming the car will be charging when the car is unplugged

**Predbat version**
8.8.10

**Environment details**

- Inverter and battery setup
- Standard HAOS installer or Docker
- Anything else?

**Screenshots**
Event can be seen in logs at 19:03. I then manually overrode

[image]

[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
[predbat1.log](https://github.com/user-attachments/files/18202305/predbat1.log)
[predbat.log](https://github.com/user-attachments/files/18202308/predbat.log)

**Predbat debug yaml file**
[predbat_debug_19_00_00.yaml.txt](https://github.com/user-attachments/files/18202316/predbat_debug_19_00_00.yaml.txt)
[predbat_debug_19_05_00.yaml.txt](https://github.com/user-attachments/files/18202320/predbat_debug_19_05_00.yaml.txt)

springfall2008, 2024-12-20:
Oh I see, the switch doesn't do what you expected it to do - it still overrides the rates it just doesn't charge the car. Maybe it should also ignore the rates?

mpartington, 2024-12-20:
It does look like I've misunderstood. I thought maybe the car needed to be both plugged in and charging, so this was a way around some kind of bug. Why would you have slots with car unplugged? I assumed it also ignored the rates.

On the plus side, my slots that are being set with car not plugged in have been charged off peak.

springfall2008, 2024-12-20:
I guess the assumption is that Octopus is reporting the prices correctly. I could ignore future low rate slots if the car isn't plugged in also?

mpartington, 2024-12-20:
I thought I'd seen that the car had to be plugged in and charging to be billed at the off peak rate, but maybe wrong.  My worry is everything fires up, but gets back billed at peak rates after post processing. However that seems incorrect based on my recent experience. 

It comes down to whatever that feature was trying to provide I suppose. I thought it would fully ignore the rate for slot if the car was unplugged. Your proposal would be great for me, but does it break another use case.  Also would have to be careful to only ignore extra slots if car is unplugged and not the overnight ones

springfall2008, 2024-12-21:
Fix on main for testing

mpartington, 2024-12-21:
Not sure this is fixed, unless I've got a dodgy setting? Car is currently unplugged, I've forced charge til 16.30 as I have already partially charged (so it counts).

However the later slots are still planning to charge, rather than be ignored

[image]
[image]
[predbat_debug_16_15_00.yaml.txt](https://github.com/user-attachments/files/18219210/predbat_debug_16_15_00.yaml.txt)
[predbat.log](https://github.com/user-attachments/files/18219212/predbat.log)

springfall2008, 2024-12-21:
Can you show the raw octopus_energy_xxxx_intelligent_dispatching sensor attributes?

mpartington, 2024-12-21:
do you mean this?
[octopus_energy-55f8547e56ef4b1e850cb253dfbfda9c-Vehicle-a1275da831ade5e5e53b72b91a697dc9.json](https://github.com/user-attachments/files/18219394/octopus_energy-55f8547e56ef4b1e850cb253dfbfda9c-Vehicle-a1275da831ade5e5e53b72b91a697dc9.json)

mpartington, 2024-12-21:
or:

Planned dispatches
- start: '2024-12-21T12:11:00+00:00'
end: '2024-12-21T18:00:00+00:00'
charge_in_kwh: -13.96
source: smart-charge
location: null
Completed dispatches
- start: '2024-12-20T10:00:00+00:00'
end: '2024-12-20T10:30:00+00:00'
charge_in_kwh: -1.37
source: null
location: AT_HOME
- start: '2024-12-20T09:30:00+00:00'
end: '2024-12-20T10:00:00+00:00'
charge_in_kwh: -0.1
source: null
location: AT_HOME
- start: '2024-12-20T11:00:00+00:00'
end: '2024-12-20T11:30:00+00:00'
charge_in_kwh: -1.48
source: null
location: AT_HOME
- start: '2024-12-20T10:30:00+00:00'
end: '2024-12-20T11:00:00+00:00'
charge_in_kwh: -0.71
source: null
location: AT_HOME
- start: '2024-12-20T11:30:00+00:00'
end: '2024-12-20T12:00:00+00:00'
charge_in_kwh: -1.12
source: null
location: AT_HOME
- start: '2024-12-21T12:30:00+00:00'
end: '2024-12-21T13:00:00+00:00'
charge_in_kwh: -1.13
source: null
location: AT_HOME
- start: '2024-12-21T12:00:00+00:00'
end: '2024-12-21T12:30:00+00:00'
charge_in_kwh: -0.47
source: null
location: AT_HOME
- start: '2024-12-21T13:30:00+00:00'
end: '2024-12-21T14:00:00+00:00'
charge_in_kwh: -0.4
source: null
location: AT_HOME
- start: '2024-12-21T13:00:00+00:00'
end: '2024-12-21T13:30:00+00:00'
charge_in_kwh: -0.6
source: null
location: AT_HOME
- start: '2024-12-21T14:30:00+00:00'
end: '2024-12-21T15:00:00+00:00'
charge_in_kwh: -0.45
source: null
location: AT_HOME
- start: '2024-12-21T14:00:00+00:00'
end: '2024-12-21T14:30:00+00:00'
charge_in_kwh: -0.59
source: null
location: AT_HOME
Provider
JEDLIX
Vehicle battery size in kwh
Unknown
Charge point power in kw
Unknown
Current start
21 December 2024 at 12:00:00
Current end
21 December 2024 at 18:00:00
Next start
21 December 2024 at 23:30:00
Next end
22 December 2024 at 05:30:00

springfall2008, 2024-12-21:
No I mean find the slot sensor in home assistant and how its attributes:
[image]

mpartington, 2024-12-21:
[image]

springfall2008, 2024-12-21:
Do you sure you have the right code, can you update to the latest release I just made?

springfall2008, 2024-12-21:
Oh wait the logfile doesn't show the extra slots:
```
2024-12-21 16:19:12.660161: Car 0 using Octopus Intelligent is unplugged
2024-12-21 16:19:12.660241: Cars: SoC kWh: [20.28] Charge limit [52.0] plan time ['18:00:00'] battery size [52.0]
2024-12-21 16:19:12.660260: Fetch Octopus rates from sensor.octopus_energy_electricity_21m0191930_1470001439853_export_current_rate
2024-12-21 16:19:12.666387: Standing charge is set to 61.561080000000004 p
2024-12-21 16:19:12.670511: Octopus Intelligent slot at 12-20 10:00:00-12-20 10:30:00 assumed price 7.0 amount 1.37 kWh location AT_HOME source None octopus_slot_low_rate True
2024-12-21 16:19:12.670614: Octopus Intelligent slot at 12-20 09:30:00-12-20 10:00:00 assumed price 7.0 amount 0.1 kWh location AT_HOME source None octopus_slot_low_rate True
2024-12-21 16:19:12.670687: Octopus Intelligent slot at 12-20 11:00:00-12-20 11:30:00 assumed price 7.0 amount 1.48 kWh location AT_HOME source None octopus_slot_low_rate True
2024-12-21 16:19:12.670758: Octopus Intelligent slot at 12-20 10:30:00-12-20 11:00:00 assumed price 7.0 amount 0.7099999999999999 kWh location AT_HOME source None octopus_slot_low_rate True
2024-12-21 16:19:12.670825: Octopus Intelligent slot at 12-20 11:30:00-12-20 12:00:00 assumed price 7.0 amount 1.12 kWh location AT_HOME source None octopus_slot_low_rate True
2024-12-21 16:19:12.670890: Octopus Intelligent slot at 12-21 12:30:00-12-21 13:00:00 assumed price 7.0 amount 1.13 kWh location AT_HOME source None octopus_slot_low_rate True
2024-12-21 16:19:12.670958: Octopus Intelligent slot at 12-21 12:00:00-12-21 12:30:00 assumed price 7.0 amount 0.47 kWh location AT_HOME source None octopus_slot_low_rate True
2024-12-21 16:19:12.671023: Octopus Intelligent slot at 12-21 13:30:00-12-21 14:00:00 assumed price 7.0 amount 0.4 kWh location AT_HOME source None octopus_slot_low_rate True
2024-12-21 16:19:12.671083: Octopus Intelligent slot at 12-21 13:00:00-12-21 13:30:00 assumed price 7.0 amount 0.6 kWh location AT_HOME source None octopus_slot_low_rate True
2024-12-21 16:19:12.671179: Octopus Intelligent slot at 12-21 14:30:00-12-21 15:00:00 assumed price 7.0 amount 0.45 kWh location AT_HOME source None octopus_slot_low_rate True
2024-12-21 16:19:12.671259: Octopus Intelligent slot at 12-21 14:00:00-12-21 14:30:00 assumed price 7.0 amount 0.59 kWh location AT_HOME source None octopus_slot_low_rate True
``
Is the Octopus Integration passing on the rates? It maybe you need to change to manual rates for import.

mpartington, 2024-12-21:
It just updates to 8.8.11, still the same

springfall2008, 2024-12-21:
Is the Octopus Integration passing on the rates? It maybe you need to change to manual rates for import.

mpartington, 2024-12-21:
Hmmm, this is fro the octopus integration - seems OK, I need to do some digging
[image]

springfall2008, 2024-12-21:
It's showing low rate now, expand the attributes for the future data. If you disable the octopus import rate setting in apps.yaml and instead set it by time that should fix it

mpartington, 2024-12-21:
I managed to snip the attributes before the slot ended. Is this different to yours then?  If I do a manual override on rates in apps.yaml, will it still  see the extra slots as 7p for charging the battery?

[image]

springfall2008, 2024-12-21:
I don't have a problem with my Octopus slots so don't see this issue.

Don't use rate override, more manually specify the rates, when the IOG slots are active then Predbat will override them.

```
#metric_octopus_import: 're:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_current_rate)'

  rates_import:
    -  start: "23:30:00"
       end: "05:30:00"
       rate: 7.5
    -  start: "05:30:00"
       end: "23:30:00"
       rate: 25.58
```

mpartington, 2024-12-21:
Got you, thanks for looking into it. Sounds like it's going to be a manual solution, so I'll either take the slots (hopefully low risk of being charged at peak), or force to demand (which is what I'm currently doing).

It's back with the octopus team who integrate these tariffs, so hopefully it will eventually get fixed. It just needs octopus and Jedlix to maintain a flow of data and not close the connection far as I can see. Otherwise Hypervolt promised IOG integration on their V2 chargers by end of year (at worst) ..... cough....

mpartington, 2024-12-21:
These are the slots they have given me 😂. I can't risk that

[image]

springfall2008, 2024-12-22:
> Got you, thanks for looking into it. Sounds like it's going to be a manual solution, so I'll either take the slots (hopefully low risk of being charged at peak), or force to demand (which is what I'm currently doing). Presumably with your suggestion, I'd also need to manually add the extra ones as well once they are set

No Predbat will add the extra slots if the car is plugged in so it should work fine.
 
> It's back with the octopus team who integrate these tariffs, so hopefully it will eventually get fixed. It just needs octopus and Jedlix to maintain a flow of data and not close the connection far [...]

Seems like their code is buggy somehow, yes.

mpartington, 2024-12-25:
Manually specifying the rates does make it ignore the extra slots when car is unplugged

closed 2024-12-25 by mpartington