springfall2008/batpred #1764: Charging at inappropriate times

andyrew66, 2024-12-20:
**Describe the bug**
So, i think i reported this bug under the wrong repo, anyway, its happened again, I've discovered it charging when it reallydoesn'tneed to be, there is currently enough Pv to cover the house load and maintain the battery, it was charging to 90% from 82% at a high rate of 10p, when in the next 20 mins wouldve atleast been a lower rate.

**Expected behaviour**
Ideally, if shoulsnt have been charging at all in this period, but at the minimum it should be charginf at the cheaper slot in the next 20 mins. A d even now in my plan i am have it charging in the next 2 mins when there is a charge slot in 30 mins thay is cheaper..

**Predbat version**

8.8.10

**Environment details**

- Inverter and battery setup
- Standard HAOS installer or Docker
- Anything else?

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.
    [image]

[image]

[image]

springfall2008, 2024-12-20:
I put a potential fix onto 'main' for testing, I can't be quite sure as I found something wrong with set_export_freeze turned off

springfall2008, 2024-12-20:
BTW: metric_min_improvement_export: 75.0 -> will probably mean no exports?

springfall2008, 2024-12-20:
Oh wait:

>calculate_second_pass: true
>calculate_tweak_plan: true

You maybe want to turn these off, they can often result in this type of odd behaviour.

andyrew66, 2024-12-20:
Yeah, I've put the export quite high as its one of those things I like to control manually. 

I will turn those settings off and see if it makes a difference

andyrew66, 2024-12-22:
Just want to add, been running on main and changed those settings as suggested and everything seems to be running perfect now.

Thank you

closed 2024-12-22 by springfall2008