springfall2008/batpred #1766: Predbat will not discharge to min soc

Eeebygum23, 2024-12-21:
**Describe the bug**
Predbat used to discharge the battery every night to 4% (0.5kwh). Now it seems to be happy to discharge to about 26%.

**Expected behaviour**
I am hoping that PB will discharge the battery down to minimum keep.

**Predbat version**

8.8.10

**Environment details**

Givenergy 135kw AIO
- installed directly onto Home assistant Yellow
- 

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.
[PREDBAT CHANGE LOG.txt](https://github.com/user-attachments/files/18217741/PREDBAT.CHANGE.LOG.txt)

springfall2008, 2024-12-21:
setting 'calculate_export_low_import' to True seems to fix this, I think I should put the default back

springfall2008, 2024-12-21:
Fix on main to change this default

gcoan, 2024-12-21:
> Fix on main to change this default

Could the documentation be updated to match this change please @springfall2008 ?

Thanks

mpartington, 2024-12-21:
Bit confused on this, was the switch working the opposite way around or does enabling stop it being sorted by time order?

Eeebygum23, 2024-12-21:
I have no idea how to update these files. Are there replacement .py files anywhere that i can copy and paste?

springfall2008, 2024-12-21:
> > Fix on main to change this default
> 
> Could the documentation be updated to match this change please @springfall2008 ?
> 
> Thanks

Pretty sure I have done, docs only get published on release however (which I just did)

springfall2008, 2024-12-21:
Oh wait I didn't.... will update the docs

gcoan, 2024-12-21:
> > > Fix on main to change this default
> 
> > Could the documentation be updated to match this change please @springfall2008 ?
> 
> > Thanks
> 
> 
> 
> Pretty sure I have done, docs only get published on release however (which I just did)

I looked at the commits referenced in the above change and didn't see any changes to the documentation files.  But if done, great

springfall2008, 2024-12-23:
Docs update now done

closed 2024-12-23 by springfall2008