springfall2008/batpred #1770: Storm watch

browiw, 2024-12-21:
Hello,

On you tube I have just watched a video by speak to the geek in which he points out that Tesla have  in the app the ability to have the batteries fully charged when there is a risk of a severe storm and you may suffer a power cut. He suggested Givenenergy followed suit and add a similar thing. Could this be added into Predbat to carry out a similar thing to Tesla.

SwiftRR, 2024-12-21:
I have watched the video and this could be a useful addition as a configurable switch (as Geek mentions in the video).
I get very few power cuts and, when they do happen, it's the result of underground streams full of flood water shorting out the main electricity cables to my road. This can happen several days following storm as flood water comes down from the Pennines.
So yes, but as a switch.

Rob

cyberkryten, 2024-12-22:
This is less of an issue in the UK as the grid is generally far more stable than in many other countries and it is also hard to define what type and severity of storm is likely to need this sort of protection.  Many UK installed battery systems are not capable of operating independently of the Grid - if you don't have a specific backup gateway that disconnects your house from the grid entirely then your batteries won't supply the house in a grid outage, so many systems would be reserving battery for no useful reason (most installers don't suggest this extra cost because of the grid stability).

There's no need for any Predbat changes though, you can already do this if you want to as there is a way to instruct Predbat to keep the batteries charged if you set input_number.predbat_set_reserve_min to a value then it will not go below that battery level.  

You could setup automations to check forecast weather/wind speeds/flood levels or whatever else you want to trigger this on (eg Met office Yellow warnings) and then set the figure to an appropriate value based on your own consumption and battery size.

I've taken a simple approach and have put that variable on a Dashboard and if I know there's a storm coming, I just increase it so 30% for a day or so.

browiw, 2024-12-22:
Hello.

Thank you for the reply which is most helpful. I have a manual EPS fitted
but I will increase the level to 30% when we have a prediction for a named
storm. Thank you for all the work you are doing and at the moment I am busy
importing and exporting electricity with all the cheap power thanks to
Predbat. Thanks again.

Best Regards ian Brown