springfall2008/batpred #1775: Don't export unless import price is greater than export price

M0LTE, 2024-12-21:
**Is your feature request related to a problem? Please describe.**

I am on Octopus Agile import, with half-hourly variable import prices. I get paid 15p/kWh fixed to export energy, irrespective of time.

I sometimes see Predbat importing for 30 minutes at say 6p/kWh and then immediately exporting it at 15p/kWh, while the import price remains below the (fixed) export price.

That feels to me like I could be open to being accused of gaming the system.

**Describe the solution you'd like**

I would like an option which allows me to prevent export while the current export price is above the current import price.

**Describe alternatives you've considered**

A fixed threshold configurable via apps.yml ("prevent export if the import price is under 10p")

springfall2008, 2024-12-21:
I'm pretty sure most users export when the Agile import rates are low and have no issues.

However, you could try a few things:

1. Turn off calculate_export_oncharge (expert mode)
2. Set metric_battery_cycle to something non-zero (e.g. 2, also expert mode)

This will likely reduce this sort of behaviour.

gcoan, 2024-12-21:
@M0LTE   I'm on Agile and 15p fixed export and yes, at times like this on cheap periods, predbat can be quite enthusiastic importing and exporting through the night.  I'll end up on a net negative total electricity bill today.

There's been quite a bit of discussion about this on the GivEnergy customer forum and the conclusion is that if Octopus felt that this was such a problem they wouldn't pay us and offer the tariff they do.  Octopus are quite happy to charge us high prices in the peak that don't reflect the actual underlying cost, this is one of the few occasions that Predbat enables us to take a little back.

As well as the suggestions @springfall2008 makes, you can also turn on switch.predbat_set_export_freeze_only or set the Predbat mode to Control Charge which will stop Predbat doing brown exports

springfall2008, 2026-10-29:
Closing as stale. This was answered at the time with existing settings, and there has been no follow-up since December 2024:

- turn off `switch.predbat_calculate_export_oncharge` (expert mode), and/or
- set `input_number.predbat_metric_battery_cycle` to a non-zero value (e.g. 2p).

Both make Predbat less willing to import cheaply and export straight back out. Please reopen if those don't do what you need.

closed 2026-10-29 by springfall2008