springfall2008/batpred #1779: Irregular Load Subtraction

nbullus, 2024-12-22:
**Is your feature request related to a problem? Please describe.**
The existing functionality allows you to subtract EV and iBoost loads from the load calculation. However, if you have other variable ad-hoc high loads these cannot be subtracted which can skew the load calculations. This would be useful for Agile customers who turn on additional ad-hoc loads when price plunge comes into effect.

For example, I have a garden room which has 2kW electric heating. Heating of this room only occurs when we either occupy it, or when the electricity price is negative (heats up by default). I also have electric towel rails in main house that are heated by electric when their is a plunge price. This additional electric heating is controlled by my HA automation with each heater having its own energy sensor.

**Describe the solution you'd like**
Extra section in Apps.yaml to specify additional list of load subtraction energy sensors to prevent skewing of base load predictions on any given day.

**Describe alternatives you've considered**
None as additional loads can be variable, sporadic with no fixed or regular pattern, and I cannot see a way of doing this accurately.

**Additional context**
None

nickgee31, 2024-12-22:
Apologies if I have misunderstood but I believe this is already possible. You can add extra items to the car_charging_energy in apps.yaml like this:

[image]

nbullus, 2024-12-22:
This would be fine if I don't need to configure EV charging, as don't have an electric car.

nickgee31, 2024-12-22:
Can you not just use it for your other energy uses though? As you can see I have my hot tub usage in there

springfall2008, 2024-12-22:
Yeh car_charging_energy can be used for anything not just cars

gcoan, 2024-12-22:
Another way you can do it is to create a custom home load sensor and subtract the loads you don't want to include.

e.g. in templates.yaml:

```
- trigger:
    - platform: time_pattern
      minutes: "/5"
  sensor:
    - name: "House Load ex ASHP Today"
      unique_id: "house_net_load_today"
      unit_of_measurement: kWh
      state_class: total
      device_class: energy
      state: >
        {% set x=(states('sensor.house_load_today')|float(0) 
          - states('sensor.ashp_energy_today')|float(0) )
        %}
        {{ max(x,0)|round(1) }}
```

The above gives me a sensor that excludes the heat pump energy.  It only updates every 5 minutes which is good enough granularity for Predbat and reduces the amount of unnecessary HA database writes that you'd get with an ordinary template sensor as the underlying sensors change

nbullus, 2024-12-22:
Thanks for all the tips, I'll have a play.

closed 2024-12-26 by springfall2008