springfall2008/batpred #1781: Repeated and unnecessary setting pause end time slot

gcoan, 2024-12-22:
**Describe the bug**
I have this afternoon upgraded from 8.5.3 to 8.8.13.  Upgrade process went smoothly but afterwards noticed that the Predbat status was often being set to a warning of a rest failure when setting pause slot on each inverter:
[image]

Looking at the logfile I can see repeated logs about setting the pause time on each Predbat run, some of which fail.
In the sensor history I can see that battery pause start time sensor is set to 00:00:00 but battery pause end time is flipping between 00:00:00 and 23:59:00 on each Predbat run - i.e. being changed to firstly 23:59:00 and then 00:00:00 on each 5 minute execution of Predbat:
[image]

The pause end time only started changing at 17:21, corresponding to when I upgraded to 8.8.13.

Total inverter writes is already at 32 in the last hour of running this Predbat version, and the battery has been in Eco/Demand status for all that time so no writes really were necessary.

I also noticed confusing log messages about whether the scheduled discharge is being turned off or not:

```
8587 | 2024-12-22 18:41:15.015490: Inverter 1 Turning off scheduled export
8586 | 2024-12-22 18:41:15.015475: Inverter 1 No write needed for scheduled_discharge_enable as False == False
```

Is this repeatedly turning off scheduled export or not?  Confusing.

**Expected behaviour**
Battery pause start and end times only ever need to be set once and shouldn't be repeatedly changing.

**Predbat version**
8.8.13

**Environment details**
HAOS 2024.12.3
GivEnergy Gen 1 hybrids with fast response 191/193 firmware (includes battery pause support)
GivTCP 3.0.4

**Log file**
[predbat.log](https://github.com/user-attachments/files/18222903/predbat.log)

**Predbat debug yaml file**
[predbat_debug_18_40_00.yaml.txt](https://github.com/user-attachments/files/18222925/predbat_debug_18_40_00.yaml.txt)

springfall2008, 2024-12-22:
I think you don't have the pause start/end time registers and so you need to comment out the pause_start and pause_end settings in apps.yaml to tell Predbat this (as GivTCP seems not to know)

gcoan, 2024-12-22:
> I think you don't have the pause start/end time registers and so you need to comment out the pause_start and pause_end settings in apps.yaml to tell Predbat this (as GivTCP seems not to know)

I was pretty sure I did have the pause start/end time registers as these controls appeared with GivTCP 2 when I turned the 'old firmware' switch off and I had to manually set them to 00:00:00 and 23:59:00 to start using the pause feature.

And if I look at the sensor history, it shows as being on 23:59:00 up to the point I upgraded Predbat this afternoon
[image]

But if I set the end time now then after a short while it reverts back to 00:00:00 and reading the end time in the portal it comes back as 00:00:00 all the time.

So maybe I don't have these controls supported and GivTCP 2 thought I had.

I had already come to the conclusion about commenting the start and end times out of apps.yaml already to reduce the inverter writes, so done that, and now am seeing the following warnings in the logfile:

2024-12-22 20:05:07.023639: Warn: Out of range index 1 within item pause_start_time value None
2024-12-22 20:05:07.024007: Warn: Out of range index 1 within item pause_end_time value None

SwiftRR, 2024-12-22:
I raised this in our old thread #3122.
On my AC3.0 inverter from mid 2022 I was able to 'new firmware' configuration in givTCP 2.4.9. I think an update to givTCP inverter firmware D0.205-A0.205 allowed this.

My apps.yaml for the pause slots is:
```
# Pause mode is not supported by all firmware's and will be ignored if not present
  pause_mode:
    - select.givtcp_{geserial}_battery_pause_mode
#    - select.givtcp_{geserial2}_battery_pause_mode

# Not all firmwares support pause start/end time, delete these if not supported
# to avoid spurious writes/warnings
#  pause_start_time:
#    - select.givtcp_{geserial}_battery_pause_start_time_slot
#    - select.givtcp2_{geserial2}_battery_pause_start_time_slot
#  pause_end_time:
#    - select.givtcp_{geserial}_battery_pause_end_time_slot
#    - select.givtcp2_{geserial2}_battery_pause_end_time_slot
```
I uncommented the two lines for pause_start_time and pause_end_time.
I then got an error:
```
2024-12-16 22:01:05,128 - GivTCP - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2024-12-16 22:01:05,174 - GivTCP - write       -  [ERROR   ] - Setting Battery Pause Slot failed: ('KeyError', 'write.py', 610)
```
I then commented out the two pause time lines and the ERROR line disappeared.

I think I was running v8.8.10 to 8.8.11 at the time.

My conclusion was that a mid 2022 AC3.0 inverters allows 'pause mode' but not start and end time pause slots. Pause mode is always refaced by the 'gracefully received' line which is supposed to be a givTCP bug.

But the errors from @gcoan seem to be different.

Is this all linked to the two recent switches (which remain a complete mystery to me)?:
```
switch.calculate_import_low_export
switch.calculate_export_high_import  
```

Rob

gcoan, 2024-12-22:
I have a Gen 1 hybrid inverter which in the latest firmware provided battery pause capability.

As I said I thought that the pause start and end times were implemented, even though we're not using them in Predbat.

Curiously I get different messages in GivTCP when the pause end time is set, I get the same critical error 86 but the GivTCP log claims the action worked fine:

```
2024-12-22 20:05:12,850 - H - mqtt        -  [INFO    ] - Requesting Control Action: setPauseEnd - {'finish': '00:27'}
2024-12-22 20:05:13,631 - H - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2024-12-22 20:05:13,663 - H - write       -  [INFO    ] - Setting Pause Slot End to: 00:27 was a success
2024-12-22 20:11:03,927 - H - mqtt        -  [INFO    ] - Requesting Control Action: setPauseEnd - {'finish': '00:09'}
2024-12-22 20:11:04,573 - H - transparent -  [CRITICAL] - Function code 86 recieved. Gracefully handled
2024-12-22 20:11:04,599 - H - write       -  [INFO    ] - Setting Pause Slot End to: 00:09 was a success
```

I've now commented the pause start and end times out  of my apps.yaml and it's no longer erroring in Predbat.  Other than the Predbat warning above I'm getting it seems OK

SwiftRR, 2024-12-22:
In #3114 @dpe got this error (on AC3.0 inverter):

```
2024-12-16 10:35:41.741872: Warn: Inverter 0 set pause slot {'start': '00:00', 'finish': '23:59'} via REST failed
2024-12-16 10:35:41.781956: Warn: record_status Warn: Inverter 0 REST failed to setPauseSlot
```

And in #3116, after restarting HA:
```
2024-12-16 11:13:55,677 - GivTCP - transparent - [CRITICAL] - Function code 86 recieved. Gracefully handled
2024-12-16 11:13:55,711 - GivTCP - write - [ERROR ] - Setting Battery Pause Slot failed: ('KeyError', 'write.py', 610)
```

Following this, I tried uncommenting the pause time slot lines (on givTCP 3.0.4) and I then got different errors.

How much of this is linked to givTCP rather than predbat?

Rob

mpartington, 2024-12-23:
Rob, are you sure you've got the pause times hashed out? It's still trying to.set the slot times, which the AC3 doesn't have.

I'm also running AC3 with GivTCP 3.0.4 and do not get those errors

springfall2008, 2024-12-23:
I'm afraid gen1 inverters don't have the pause start/end time registers but GivTCP seems blissfully unaware of this....

SwiftRR, 2024-12-23:
To confirm, predbat, when running givTCP 3.0.4, I have pause mode **uncommented**.
My AC3.0's serial number start CE2218, so week 18 in 2022. I get 3000 W charge and discharge.  
```
  pause_mode:
    - select.givtcp_{geserial}_battery_pause_mode
```
and the pause start_time and end_time lines commented out.
```
#  pause_start_time:
#    - select.givtcp_{geserial}_battery_pause_start_time_slot
#    - select.givtcp2_{geserial2}_battery_pause_start_time_slot
#  pause_end_time:
#    - select.givtcp_{geserial}_battery_pause_end_time_slot
#    - select.givtcp2_{geserial2}_battery_pause_end_time_slot
```
I did experiment by uncommenting the pause start_time and end_time lines and got errors in givTCP log. Is I put the comment #s back in apps.yaml.
Those of us who have tried this seem to have got different errors in the givTCP log!

Rob

SwiftRR, 2024-12-23:
For info, I have just submitted what may be a related issue here following a weird 'crash' on givTCP v3.0.4:
on https://github.com/britkat1980/giv_tcp/issues/316

Rob

mpartington, 2024-12-23:
hi Rob, I also have an AC3 on latest firmware.and on givtcp 3.0.4 and am not seeing the errors in the GivTCP logs (except the one related to the firmware bug). What message are you still getting with the pause slot times hashed out?

SwiftRR, 2024-12-23:
I normally get **no errors** in givTCP logs. The one this afternoon came out of the blue and may be nothing to do with Pause/Freeze. But the last successful instruction before the givTCP error sequence was a Pause.

In predbat apps.yaml, the only line that is uncommented is the one for the various Pause modes
Pausecharge, Pausedischarge, etc.

Rob

gcoan, 2025-11-22:
Have fixed the code to stop the unnecessary warning messages when pause_start and end_time are not configured in apps.yaml will be in my next PR