springfall2008/batpred #1783: Debug from Johan Ä

JohanAlvedal, 2024-12-23:
**Expected behaviour**

Why does my battery only last one pass and not two on a day

**Predbat version**

8.8.13

**Predbat debug yaml file**

[debug_yaml.txt](https://github.com/user-attachments/files/18232692/debug_yaml.txt)

springfall2008, 2024-12-23:
I'm afraid your debug YAML is empty?

JohanAlvedal, 2024-12-23:
I’ll fix it when I get home; I tried from my phone, and it didn’t go so well.

JohanAlvedal, 2024-12-23:
[predbat_debug.txt](https://github.com/user-attachments/files/18233350/predbat_debug.txt)

springfall2008, 2024-12-24:
I put some fixes on 'main' for testing that may help, but you might also want to try a lower 'metric_battery_cycle' setting and turn off set_charge_freeze_only

JohanAlvedal, 2024-12-28:
I've tried adding some settings and switches, but it's easy to get lost in all the configurations. I also use öre instead of pence. When I add a cost of 44 öre for Metric Battery Cycle Cost, it doesn't charge at all right now, and I get the feeling that this value has too much of an impact and isn't directly comparable to öre/kWh. If you have the time, I'd really appreciate it if you could review my settings and let me know if there's anything I should adjust.

[https://drive.google.com/drive/folders/1l9HmCDXdma6lSIqjySedXKROBAWn58Cr?usp=drive_link](url)

springfall2008, 2024-12-28:
The drive link doesn't work but if you attach another predbat debug YAML I can view all the settings

JohanAlvedal, 2024-12-28:
[predbat_plan.html.txt](https://github.com/user-attachments/files/18266094/predbat_plan.html.txt)
[predbat.log](https://github.com/user-attachments/files/18266095/predbat.log)
[apps.yaml.txt](https://github.com/user-attachments/files/18266096/apps.yaml.txt)
[predbat_debug.yaml.txt](https://github.com/user-attachments/files/18266097/predbat_debug.yaml.txt)

JohanAlvedal, 2024-12-28:
In normal mode, the plan looks better, but I lose access to all the expert settings. In expert mode it seems that the planning still ends up completely different, even when I reset all costs to zero and set all switches to default mode.

springfall2008, 2024-12-28:
I'm looking now, there's some odd stuff here.

One thing I notice wrong right away (but changing it didn't fix it) is that 'forecast_hours' in apps.yaml should be 48 and not 24

springfall2008, 2024-12-28:
So the difference between expert mode and non-expert mode was around the plan hours, I suggest you change both forecast_hours in apps.yaml to 48 and also plan_forecast_hours setting in expert mode.

And also update to the latest release, and then report back

JohanAlvedal, 2024-12-28:
I have updated and fixed 48 hours. Try it for a couple of days and we'll see

closed 2025-01-12 by springfall2008