springfall2008/batpred #1793: 8.8.15 plan not stable, oscillating between freeze and demand

gcoan, 2024-12-26:
**Describe the bug**
My plan on 8.8.14 was to let the SoC drop to empty (one battery is already empty) and then recharge overnight in the few cheap slots there are:
[image]

Upgraded to 8.8.15 at 19:40 and Predbat decided to instead freeze the charge at the current level until the overnight charging (I was fairly OK with this plan):
[image]

But by 20:00 the plan had changed again to let the battery now discharge to empty:
[image]

And by 20:15 it had changed again to now freeze charge through to tonight:
[image]

If I look across the different plans at a fixed 100% SoC point, say 05:00, I can see that there isn't a lot financially in the different plans, so I can understand why Predbat comes up with different plans but this instability isn't good, not least for inverter writes where each plan change writes more inverter register commands.

**Expected behaviour**
Ideally a bit of stability in the plan, not changing the plan unless there is a measurable financial benefit, and then sticking to that changed plan, not changing back again within half an hour.

**Predbat version**
8.8.15

**Environment details**
HAOS 14.1, 2x Gen 1 GiVenergy hybrid inverters

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**
I tried to download the debug yaml file using the predbat web console option using the HA Companion app, but the file appears on the screen and didn't give any option to save it. It also appeared to hang the app and after taking the screenshot I couldn't do anything without closing and clearing the app from memory:
[image]

gcoan, 2024-12-26:
Logfile:
[predbat.log](https://github.com/user-attachments/files/18255315/predbat.log)

Debug YAML:
[predbat_debug_20_20_00.txt](https://github.com/user-attachments/files/18255317/predbat_debug_20_20_00.txt)

gcoan, 2024-12-26:
20:35, plan is now to hold the SoC for this slot when the import rate is 22.05p, and then release it in the subsequent slot when the import rate is 21.95p which doesn't make sense at all as the current slot is more expensive

[image]

gcoan, 2024-12-26:
20:50 changed back to freeze charge through the evening again:
[image]

Latest logfile:
[predbat.log](https://github.com/user-attachments/files/18255408/predbat.log)

gcoan, 2024-12-27:
Here's another bit of data on the "mode instability", through the night the inverters charged in the (few) low rate periods, but there is sometimes quite a few mode swaps during a half hour period, e.g. from 4:00 to 5:00 (2 periods, 21.43p and 20.3p) Predbat swapped between Freeze Charging (20 minutes), Charging (10 minutes), Hold Charging (20 minutes), Charging (5 minutes), Hold Charging (5 minutes) then Freeze Charging (10 minutes).

[image]

The target SoC was 100% throughout the entire period.

Functionally Hold Charging and Charging are I think doing the same thing?

Inverter writes going up during the period.  I have two inverters so the writes count will be doubled
[image]

springfall2008, 2024-12-27:
I'll play around with it a bit more but I'd say the most obvious things to do in order to stop the instability are either:

a) Turn off best_soc_keep (set to 0)
- This means you will get the cheapest plan

b) Turn up best_soc_keep_weight to 1.0 
- This means your constraint of 1.0 keep will likely be kept to, but at the expensive of higher rate charging

gcoan, 2024-12-27:
I'll set best_soc_keep to zero and see what that does for the plan.

I had it set to the battery réserve from some time ago as the plan needed that to stop some other behaviour, but that was a while ago and predbat has changed a lot since then. In the winter the best_soc_keep is immaterial, need the whole battery range

gcoan, 2024-12-27:
Plan with best_soc_keep set to 1.0:

[image]

There's a spurious freeze charge at 19:00 and a charge at 20:30 that shouldn't really be there

Setting best_soc_keep to 0.0:

[image]

Both of these go away and the battery just runs down to empty in the peak period.

No charging back up overnight but I think that's down to the length of the forecast plan, and as it rolls forward to include tomorrow's peak rates, it will include the overnight charge

gcoan, 2024-12-27:
I changed best_soc_keep to 0 at 12:09, plan as above

Since then it's again cycled through Charging/Hold Charging/Freeze Charging with some of these only being executed for 5 minutes.
[image]

I know I can increase the duration between runs but that doesn't seem the right fix, more consistency between activities in a 30 minute slot, such as not changing between Hold Charging and Freeze Charging (which are almost the same) would be better

springfall2008, 2024-12-27:
I did look at the log you posted earlier, but it didn't really cover the time when the plan changed.

I have an idea why this could be, as in dual inverter systems I changed the way the SOC was calculated and so the algorithm to pick the 'same again' if not much changed would be broken.

Please can you try the new code on 'main' and see if to helps?

springfall2008, 2024-12-27:
Hold on I found an issue with this change

springfall2008, 2024-12-27:
Fix has been updated on 'main' ready for testing

gcoan, 2024-12-27:
I have updated to main (twice!), the before and after plans looked the same and were what I would expect of charging in the few cheaper periods and holding in-between

[image]

I'll see how it executes tonight, thanks

gcoan, 2024-12-27:
Slightly weird, after letting the SoC drop to empty, one battery on 5% and the other dropping to 2% (probably needs a recalibration), was running on grid import with no charging planned until 01:30

But its just swapped to Hold Charging 3% then NoCharge 3% as I was writing this even though I don't see that there is financial benefit in this:
[image]

And both batteries are filling up:
[image]

I am setting a manual force demand to stop this charging

Here's the logfile for just now:
[predbat.log](https://github.com/user-attachments/files/18264174/predbat.log)

springfall2008, 2024-12-28:
There's something odd here about when the battery percentage level is below the reserve, I think Predbat is confused in this case. I'll have to see what the model actually does.

springfall2008, 2024-12-28:
A proposed fix is on 'main' where I treat a freeze if the SOC is below the reserve the same as a charge

gcoan, 2024-12-28:
Thanks, the freeze when below reserve shouldn't happen because it should never get below reserve, but occasionally it does do so.

It looks like the charging behaviour last night was better with Predbat sticking to just Charging mode from 01:30-02:00
[image]

and from 2:32 to 3:02 charging then freeze for half hour to 3:32 then charging to 4:02
[image]

unfortunately other problems with my predbat sensors prevent me sharing a more complete set of results.  I accidentally purged the import, export and house load sensors too much so Predbat crashed at 02:30 which I was able to fix by changing the sensor, and then crashed again at 04:15 and stayed crashed all night.
I'm still trying to fix that issue.  Changing days_previous to 1 and trying other sensors and it still doesn't work 

[predbat.log](https://github.com/user-attachments/files/18265975/predbat.log)

gcoan, 2024-12-28:
Have updated to main.  Manually set forced charge for the 12:00 and 12:30 slots to get the battery topped up, looks like it is still cycling between different charging/hold charging statii:

[image]

[predbat.log](https://github.com/user-attachments/files/18266385/predbat.log)

springfall2008, 2024-12-28:
Can you save the debug YAML?

springfall2008, 2024-12-28:
It looks like the hold charging isn't holding and the SOC drops back down and it has to charge again.

Something odd here, I think there is a bug when SOC is out of balance, will check

2024-12-28 12:45:02.243737: Inverter 0 SOC: 8.85kW 93% Current charge rate 2392W Current discharge rate 2704W Current power -1733.0W Current voltage 52.0V
2024-12-28 12:45:12.343506: Inverter 1 SOC: 4.18kW 100% Current charge rate 2392W Current discharge rate 2704W Current power 0.0W Current voltage 52.0V
2024-12-28 12:45:24.702921: Charge window will be: 2024-12-28 12:00:00+00:00 - 2024-12-28 15:30:00+00:00 - current soc 93 target 100

...
2024-12-28 12:45:34.761667: Inverter 0 adjust target soc for charge to 100.0% based on requested all inverter soc 100%
2024-12-28 12:45:34.761698: Inverter 0 Current Target SOC is 100%, already at target
2024-12-28 12:45:34.761716: Inverter 0 Current reserve is 4.0 already at target

2024-12-28 12:45:44.829737: Inverter 1 adjust target soc for charge to 100.0% based on requested all inverter soc 100%
2024-12-28 12:45:44.829839: Inverter 1 Current Target SOC is 100%, already at target
2024-12-28 12:45:44.829868: Inverter 1 Current reserve is 4.0 already at target
2024-12-28 12:45:44.829895: Inverter 1 count register writes 0

2024-12-28 12:45:45.055276: Info: record_status Hold charging target 100%-100%

gcoan, 2024-12-28:
Here’s the debug file.  Freeze and hold charge do I think work on my inverter and both of them are the same gen 1 hybrid’s
[1450.txt](https://github.com/user-attachments/files/18266532/1450.txt)

springfall2008, 2024-12-28:
So the 'bug' is that if inverter 0 has an SOC that means it can hold charge or freeze charge but the second inverter SOC is too low then Predbat get confused and ends up in the wrong state. I've changed it so that it won't freeze or hold until all SOCs are above the minimum.

gcoan, 2024-12-28:
I was just looking at the graph of SoC vs status, the yellow status being when Predbat was in hold charge, red is freeze charge,  and can see that both there is some charge leakage

[image]

Let me know when there's a new main version and I'll install it.  Next charging period looks to be about 4am tonight

springfall2008, 2024-12-28:
Okay, I've put some code on 'main' for testing - that was quite complex but I managed to write some unit tests for it so hopefully it will work.

Now if you get a freeze charge but one inverters SOC is below the reserve it will charge that one up to the reserve level while holding the other at the current level.

And hold charge will not be active if one inverters SOC is below the hold level even if on average its above, instead it will charge to the target level which should start to balance them.

gcoan, 2024-12-28:
@springfall2008 I updated to the main version about 3 hours ago, it was running fine but has just started crashing with the following errors:

```
12374 | 2024-12-28 20:05:04.464910: Error: 'PredBat' object has no attribute 'soc_percent'
12373 | 2024-12-28 20:05:04.464632: Info: record_status Error: Exception raised 'PredBat' object has no attribute 'soc_percent'
12371 | AttributeError: 'PredBat' o bject has no attribute 'soc_percent'
12370 | if self.set_soc_enable and can_hold_charge and self.soc_percent >= self.charge_limit_percent_best[0]:
12369 | File "/config/execute.py" , line 172, in execute_plan
12368 | status, status_extra = self.execute_plan()
12367 | File "/config/predbat.py" , line 564, in update_pred
12366 | self.update_pred(schedu led=True)
12365 | File "/config/predbat.py" , line 973, in run_time_loop
12364 | 2024-12-28 20:05:04.437490: Error: Traceback (most recent call last):
12363 | 2024-12-28 20:05:04.437198: Error: Exception raised 'PredBat' object has no attribute 'soc_percent'
12374	2024-12-28 20:05:04.464910: Error: 'PredBat' object has no attribute 'soc_percent'
12373	2024-12-28 20:05:04.464632: Info: record_status Error: Exception raised 'PredBat' object has no attribute 'soc_percent'
12371	AttributeError: 'PredBat' o bject has no attribute 'soc_percent'
12370	if self.set_soc_enable and can_hold_charge and self.soc_percent >= self.charge_limit_percent_best[0]:
12369	File "/config/execute.py" , line 172, in execute_plan
12368	status, status_extra = self.execute_plan()
12367	File "/config/predbat.py" , line 564, in update_pred
12366	self.update_pred(schedu led=True)
12365	File "/config/predbat.py" , line 973, in run_time_loop
12364	2024-12-28 20:05:04.437490: Error: Traceback (most recent call last):
12363	2024-12-28 20:05:04.437198: Error: Exception raised 'PredBat' object has no attribute 'soc_percent'
```

Had been in Demand mode all that time, but has just changed to freeze charge:
[image]

Also looks like the failure wasn’t caught correctly as the error attribute of predbat.status is false.

Logfile:
[predbat.log](https://github.com/user-attachments/files/18267529/predbat.log)

I haven’t upgraded yet to the release you have just pushed out, it’s not yet on my predbat versions dropdown list.  When it arrives I will try upgrading to see if this fixes this crash

gcoan, 2024-12-28:
OK the problem is with the freeze charge logic on this version I am running.

After half hour when Predbat changed the plan to Demand, the error cleared.

Logfile:
[predbat.log](https://github.com/user-attachments/files/18267577/predbat.log)

Still not showing the new released version to upgrade to, have to wait until the github cache expires

springfall2008, 2024-12-28:
You can just update to main again

gcoan, 2024-12-28:
Sorry Trefor, the 8.8.17 release crashes in freeze charge mode.

The update appeared on the list but thought I was already on that version so I couldn’t select it.
Downgraded to 8.8.16, set a manual charge freeze, worked OK.
Upgraded to 8.8.17 and executing the charge freeze crashes Predbat same as the main version.

Here’s the logfile:
[predbat.log](https://github.com/user-attachments/files/18267782/predbat.log)

springfall2008, 2024-12-28:
Fix in 8.8.18 - https://github.com/springfall2008/batpred/releases/tag/v8.8.18

gcoan, 2024-12-28:
> Fix in 8.8.18 - https://github.com/springfall2008/batpred/releases/tag/v8.8.18

thanks for fixing so quickly.

8.8.18 wasn't showing on the drop-down list yet so I upgraded to main with a manual freeze charge that was crashing on 8.8.17.  Doesn't crash now but does report the freeze charge as being a status of hold charge, maybe a side effect of the recent code changes?

[image]

Anyway, not crashing which is the important thing, I'll see how it works tonight

gcoan, 2024-12-29:
Much more stable overnight with 8.8.18, charging from 3:30-4:00, 4:30-6:00 and 7:30-8:30 when rates were lowest and no intermediate charging/freeze charging/hold charging changes in those charging slots

[image]
[image]

Thanks Trefor

springfall2008, 2024-12-29:
There's a couple of extra tweaks on 'main' that you could test if you get a chance?

gcoan, 2024-12-29:
No problem, I’ll install it now and let you know how it operates in tonight’s charging

gcoan, 2024-12-30:
Hi Trefor,

I ran with the version on main yesterday evening and night, sorry for the delay in not feeding back.  Have just upgraded to 8.8.19 which I assume includes the main changes.

TBH it looked to me as if the version on main was a step backwards yesterday evening.  The batteries were empty, rates were high-ish and dropping overnight:
[image]

But Predbat chopped and changed between demand, freeze charging, hold charging and even 5 minutes of charging:
[image]
[image]

So I think something isn't quite right here still

Overnight the rates were comparatively low and there was a power-up event from 11, Predbat did what I expected, imported for a bit and exported for a bit, leaving the batteries with just enough space to fill up in the powerup.
[image]
[image]

springfall2008, 2024-12-30:
Do you have a logfile?

springfall2008, 2024-12-30:
I push some changes to 'main' under a pre-release of .20, can you please test those and save the logfile?

gcoan, 2024-12-30:
> Do you have a logfile?

I knew you needed a logfile but stupidly forgot to attach it.

I just checked and I only have predbat.1.log and predbat.log with the earliest records starting at 13:20 today, so last night is lost.

I pushed through a PR a few months ago https://github.com/springfall2008/predbat_addon/pull/11 which changed the backup logfile and increased the number of logfile generations to 9, and I can see that this is incorporated in the codebase, but the for loop in lines 98-103 of hass.py clearly isn't working as I only have 1 old logfile.

```
        # maximum number of historic logfiles to retain
        max_logs = 9
        
        log_size = self.logfile.tell()
        if log_size > 10000000:
            # check for existence of previous logfiles and rename each in turn
            for num_logs in range(max_logs - 1, 1, -1):
                filename = "predbat." + format(num_logs) + ".log"
                if os.path.isfile(filename):
                    newfile = "predbat." + format(num_logs + 1) + ".log"
                    os.rename(filename, newfile)
```

Can you see if I have done something stupid?  I couldn't test it fully running as an addon without building a new addon repository myself.

Tonight I have seen one weird flip, same situation, batteries empty after changing to Demand at the start of the peak, but then at 20:36 changed to Hold Charge for 25 minutes:
[image]

Logfile for tonight:
[predbat.log](https://github.com/user-attachments/files/18277864/predbat.log)

Meanwhile I will test the main branch tonight

springfall2008, 2024-12-31:
Okay, some more changes on 'main' as I realised I had a bug.

Also I fixed the logfile generation issue but you need to update your add-on for that.

gcoan, 2024-12-31:
Lots of fun and games last night with my inverters getting set to the wrong date set, breaking GivTCP and Predbat. Resulted in spending a couple of hours finding and eventually fixing the issue, see [318](https://github.com/britkat1980/giv_tcp/issues/318) on GivTCP

Lots of people affected on the GivEnergy community forum, github, etc,  both GivTCP and GivLocal broke with the bad date format.

So the status was a bit mucked up, but here it is up to midnight
[image]

And after I'd fixed everything:
[image]

Logfiles only go back to 3:45 am today

[predbat.log](https://github.com/user-attachments/files/18281550/predbat.log)
[predbat.1.log](https://github.com/user-attachments/files/18281551/predbat.1.log)

> Okay, some more changes on 'main' as I realised I had a bug.
> 
> Also I fixed the logfile generation issue but you need to update your add-on for that.

Thanks for fixing the logfile generation issue, I have updated the addon and then updated predbat to the main version (but I see there is a new 8.8.22 release that is presumably the same - should I be on that?)

Trefor, I'm not convinced about the fix you made to the logfile generation issue, I had tested the code loop running it in predbat.py not the addon hass.py, so was fairly confident it was correct.

Originally your code had:
```
            os.rename("predbat.log", "predbat.1.log")
            self.logfile = open("predbat.log", "w")
```

rename current logfile as .1 and open a new logfile.  I left that in place and added the preceding for loop.  I'm not convinced that it is actually executing rather than its stopping prematurely.

the pseudo logic of my code was:

max_logs=9
from num_logs = max_logs-1 to 1 in steps of -1
  if filename.[num_logs.log] exists, rename to filename.[numlogs+1].log

So when the loop gets down to 1 it should rename predbat.1.log to predbat.2.log - but it isn't.

You've changed the end of the loop to be 0, but this just means it will try to find predbat.0.log which it will never find

I'm sure there is something else wrong but I can't see what it is

gcoan, 2024-12-31:
Running main (pre-release of 8.8.20)….

In the peak period and normally my batteries basically just cover the peak period and I start grid importing at about 8pm.

Maybe we are running slightly behind average but not particularly because the cooking has just started, but I don’t like the plan that Predbat has come up with, exporting at 16:30 and part of 18:00 slot ahead of cheaper slots from 19:30.
Maybe this will come out fine, but the risk is that I run out of battery prematurely and end up grid importing.
Since Predbat has planned 1.5 slots of discharge I would have thought it should have scheduled them to be in the 18:30 and 19:00 slots rather than where it has placed them?

[image]

Debug file and logfile:
[1625.txt](https://github.com/user-attachments/files/18282716/1625.txt)
[predbat.log](https://github.com/user-attachments/files/18282718/predbat.log)

gcoan, 2024-12-31:
r/e the logfile generations, turning debug mode on of course produces more voluminous logs, and it looks like the generations are now being saved correctly:

[image]

Confused !

But good news

springfall2008, 2024-12-31:
For the python loop the end is exclusive.

If you grab the debug yaml I can have a look

gcoan, 2024-12-31:
> For the python loop the end is exclusive.
> 
> If you grab the debug yaml I can have a look 

Ah. Flipping Python.

The debug file from the peak period exports is attached 2 comments above 

Happy New Year

springfall2008, 2025-01-01:
If you turn off 'calculate_export_high_import' then you will get the plan you were more expecting. When that setting is on export slots are picked for the highest import rates which are earlier in the day.

[image]

gcoan, 2025-01-01:
> If you turn off 'calculate_export_high_import' then you will get the plan you were more expecting. When that setting is on export slots are picked for the highest import rates which are earlier in the day.

Thanks Trefor, I thought it might be one of those new switches that would do this but was unsure. I've turned it off now.

With all the changes you've made to multi-inverter support I think I'll close this now.  I don't see the chopping and changing of the plan which is the issue I first raised.

closed 2025-01-01 by gcoan