springfall2008/batpred #1794: 8.8.15 causing errors on load

andyrew66, 2024-12-27:
**Describe the bug**
The following error appears upon updating 
Status
Status	Error: Exception raised str.format() got multiple values for keyword argument 'charge_start_time

Reverting to 8.8.13 resolves the issue

**Predbat version**

8.8.15

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.
[image]

[image]

gcoan, 2024-12-27:
what inverter type and can you share a debug file or apps.yaml please.

I'm on 8.8.15 and not seen this error so it must be inverter-specific

andyrew66, 2024-12-27:
My log files are at the end of my comment

It's a lux inverter, as mentioned, it doesn't have an issue on previous versions.

springfall2008, 2024-12-27:
Fix on 'main', will be released soon

springfall2008, 2024-12-28:
Fix released

closed 2024-12-28 by springfall2008