springfall2008/batpred #1796: update to 8.8.16 not working after stop/start is says is still 8.8.15

eddysteurs, 2024-12-27:
**Describe the bug**
A clear and concise description of what the bug is.
update to 8.8.16 not working after stop/start is says is still 8.8.15
**Expected behaviour**
A clear and concise description of what you expected to happen.

**Predbat version**
should be 8.8.16
xxxx

**Environment details**

- Inverter and battery setup
- Standard HAOS installer or Docker : HAOS
- Anything else?

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.
Predbat files are installed correctly for version v8.8.15
**** Starting Standalone Predbat ****
2024-12-27 13:48:29.565764: Loading apps.yaml
2024-12-27 13:48:29.605640: Info: Connected to Home Assistant at http://supervisor/core
2024-12-27 13:48:29.608290: Info: Add-on slug is 6adb4f0d_predbat
2024-12-27 13:48:29.608396: Creating task: <coroutine object HAInterface.socketLoop at 0x7f28d375de00>
2024-12-27 13:48:29.608865: Info: Start socket for url http://supervisor/core/api/websocket
2024-12-27 13:48:29.609341: Info: Web Socket task started
2024-12-27 13:48:29.609859: Creating task: <coroutine object WebInterface.start at 0x7f28d375dd90>
Web interface started
2024-12-27 13:48:29.615199: Info: Web Socket active
2024-12-27 13:48:29.694867: Warn: Regular expression argument: metric_standing_charge unable to match re:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_current_standing_charge), now will disable
2024-12-27 13:48:29.697515: Warn: Regular expression argument: carbon_intensity unable to match re:(sensor.carbon_intensity_be), now will disable
2024-12-27 13:48:29.699744: Warn: Regular expression argument: octopus_saving_session unable to match re:(binary_sensor.octopus_energy([0-9a-z_]+|)_saving_session(s|)), now will disable
Watching ['./execute.py', './solcast.py', './web.py', './octopus.py', './predbat.py', './userinterface.py', './gecloud.py', './apps.yaml', './utils.py', './inverter.py', './unit_test.py', './hass.py', './ha.py', './predheat.py', './output.py', './fetch.py', './prediction.py', './download.py', './config.py', './plan.py', './futurerate.py'] for changes
2024-12-27 13:48:41.549395: Warn: Inverter 0: Using inverter type MINE - not all features are available
2024-12-27 13:48:56.852461: Warn: Inverter 0: Using inverter type MINE - not all features are available
2024-12-27 13:48:59.416106: Warn: Service call script/set_solis_inverter_config_predbat_pause_charge data {} failed with response None
2024-12-27 13:48:59.521222: Info: record_status Hold charging target 100%-100%
2024-12-27 13:50:09.490865: Warn: Inverter 0: Using inverter type MINE - not all features are available
2024-12-27 13:50:11.737313: Info: record_status Hold charging target 100%-100%
2024-12-27 13:55:08.859160: Warn: Inverter 0: Using inverter type MINE - not all features are available
2024-12-27 13:55:25.190938: Warn: Inverter 0: Using inverter type MINE - not all features are available
2024-12-27 13:55:25.327080: Info: record_status Hold charging target 100%-100%
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/predbat.py
Got data, writing to /config/predbat.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/config.py
Got data, writing to /config/config.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/prediction.py
Got data, writing to /config/prediction.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/gecloud.py
Got data, writing to /config/gecloud.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/utils.py
Got data, writing to /config/utils.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/inverter.py
Got data, writing to /config/inverter.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/ha.py
Got data, writing to /config/ha.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/download.py
Got data, writing to /config/download.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/unit_test.py
Got data, writing to /config/unit_test.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/web.py
Got data, writing to /config/web.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/predheat.py
Got data, writing to /config/predheat.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/futurerate.py
Got data, writing to /config/futurerate.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/octopus.py
Got data, writing to /config/octopus.py.v8.8.16
Downloading https://raw.githubusercontent.com/springfall2008/batpred/v8.8.16/apps/predbat/solcast.py
Got data, writing to /config/solcast.py.v8.8.16
[... 27 lines cut ...]
2024-12-27 13:58:55.593523: Loading apps.yaml
2024-12-27 13:58:55.631217: Info: Connected to Home Assistant at http://supervisor/core
2024-12-27 13:58:55.632957: Info: Add-on slug is 6adb4f0d_predbat
2024-12-27 13:58:55.632986: Creating task: <coroutine object HAInterface.socketLoop at 0x7f2aa7f75e00>
2024-12-27 13:58:55.633306: Info: Start socket for url http://supervisor/core/api/websocket
2024-12-27 13:58:55.634208: Info: Web Socket task started
2024-12-27 13:58:55.634336: Creating task: <coroutine object WebInterface.start at 0x7f2aa7f75d90>
Web interface started
2024-12-27 13:58:55.642861: Info: Web Socket active
2024-12-27 13:58:55.710861: Warn: Regular expression argument: metric_standing_charge unable to match re:(sensor.(octopus_energy_|)electricity_[0-9a-z]+_[0-9a-z]+_current_standing_charge), now will disable
2024-12-27 13:58:55.713059: Warn: Regular expression argument: carbon_intensity unable to match re:(sensor.carbon_intensity_be), now will disable
2024-12-27 13:58:55.714688: Warn: Regular expression argument: octopus_saving_session unable to match re:(binary_sensor.octopus_energy([0-9a-z_]+|)_saving_session(s|)), now will disable
Watching ['./execute.py', './solcast.py', './web.py', './octopus.py', './predbat.py', './userinterface.py', './gecloud.py', './apps.yaml', './utils.py', './inverter.py', './unit_test.py', './hass.py', './ha.py', './predheat.py', './output.py', './fetch.py', './prediction.py', './download.py', './config.py', './plan.py', './futurerate.py'] for changes
2024-12-27 13:59:06.855708: Warn: Inverter 0: Using inverter type MINE - not all features are available
2024-12-27 13:59:21.917675: Warn: Inverter 0: Using inverter type MINE - not all features are available
2024-12-27 13:59:22.548031: Warn: Service call script/set_solis_inverter_config_predbat_pause_charge data {} failed with response None
2024-12-27 13:59:22.651693: Info: record_status Hold charging target 100%-100%
2024-12-27 14:00:10.030658: Warn: Inverter 0: Using inverter type MINE - not all features are available
2024-12-27 14:00:12.363079: Info: record_status Hold charging target 100%-100%
**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

symonchester, 2024-12-27:
Likewise, also breaks open weather part of predheat..

2024-12-27 13:07:11.889752: WARN: Unable to fetch data for weather.openweathermap
2024-12-27 13:07:11.900906: Warn: record_status Warn - Unable to fetch data from weather.openweathermap
File ./predbat.py was modified
Stopping Predbat due to file changes....
2024-12-27 13:07:12.285918: Stopping Predbat
Web interface stop called
Web interface stopped

springfall2008, 2024-12-27:
Shoul dbe fixed now, please check

symonchester, 2024-12-27:
> Shoul dbe fixed now, please check

Installs ok for me now and openweather error gone.. No errors in the logs following install

closed 2024-12-28 by springfall2008