springfall2008/batpred #1798: Excessive charging/hold charge after 100% SOC reached

Yuffi17, 2024-12-27:
**Describe the bug**
Predbat sometimes plans to charge (or hold charge) well beyond the point where 100% SOC is reached, seemingly irrespective of price. 
This occurs for my system (GivEnergy Hybrid Inverter with two 9.5kWh batteries) and my son-in-law's system (two GivEnergy AIW units on a single gateway) so I don't think it's hardware dependant. We're both on Agile and using Predbat to charge up for the day.

The plan produced will charge at the lowest priced slots (as expected) and then sometimes but not always it continues to charge (or switches to hold charge) for several more hours despite the system being at 100% SOC and slot prices being higher.
I've also seen Predbat update my plan (as it did today) to switch from demand to charging (this should not have happened as my system was at 89% and had more than enough capacity to run the house for the rest of the day on expected demand) until the (overnight) lower prices.

Therefore I think there's something weird happening with calculating or allocating charging slots in the plan. Unfortunately I don't know what is causing this so can't offer any helpful insights.

**Expected behaviour**
Predbat plan should charge to 100% SOC using the lowest price sessions and then enter idle/demand mode until the next set of cheap charging slots (often the next day)

**Predbat version**
8.8.16 but it was doing this on 8.8.15 too. This behaviour has been around for a week or two now.

**Environment details**
- GivEnergy Hybrid inverter and two 9.5kWh batteries. 
- Predbat installed as addon in HA and kept up to date
- I'm still using GivTCP v2 (not updated to version 3 yet) but this should not matter?

I sometimes use force-demand and force-charge to optimise charging blocks (to reduce writes - I don't mind paying a little more to get a decent charge block). I don't think this manual over-ride has any influence beyond when the system reaches 100% SOC., but it may be worth knowing.

**Screenshots**
[image]

This plan was weird as it says it's charging after the hold charge, but the system SOC is decreasing...??
[image]

**Predbat debug yaml file**
[predbat_debug_14_45_00.yaml.txt](https://github.com/user-attachments/files/18261145/predbat_debug_14_45_00.yaml.txt)

springfall2008, 2024-12-27:
I don't see anything obviously wrong in the plan in that charging and discharging incurs losses where as running off grid avoids those losses.

But let me run your plan and see

springfall2008, 2024-12-27:
The plan you uploaded in the YAML looked like this to me, but it doesn't match the image - can you check?

[image]

springfall2008, 2024-12-27:
The plan you uploaded YAML to me looks like this, I don't see any issues with it? Did you fail to capture the time period with the problem?

[image]

springfall2008, 2024-12-27:
BTW: You may find updating to .16 resolves your issue, but keep in mind you have **best_soc_keep** set to 3.0 which means you are trying to keep your battery level higher at all times. Maybe reduce that either to something lower or to 0?

gcoan, 2024-12-27:
> Predbat plan should charge to 100% SOC using the lowest price sessions and then enter idle/demand mode until the next set of cheap charging slots (often the next day)

I think your expectation of how Predbat works isn't quite right.

Predbat always does (or tries to do) a least cost optimisation, its not as simple as 'charge the battery to 100% and then enter idle mode'

At the moment the Agile prices are very 'flat', not much variation overnight or even during the day, so once the battery is charged other than the peak period (where your battery can be seen to discharge) there is no real benefit to using that stored battery charge vs importing from the grid.
And in fact as @springfall2008 highlights, using the stored battery charge will incur the inverter and battery discharge losses, so it will actually cost you more than just importing from the grid.

That's why you are seeing quite a lot of hold or freeze charging, Predbat is letting you import from the grid rather than using the stored charge as this is most cost effective

Yuffi17, 2024-12-27:
Thanks for speedy responses - especially at this time of year.  Please note this issue isn't an emergency, but it is something that doesn't seem to make sense. 

The plan I uploaded above is indeed the one shown in Trefor's reply: although I disagree that there is nothing wrong with this specific plan because the midnight charge/export is not profitable and has no future chance of profitability at the declared prices, but that's not the problem I wanted to raise.

I am using Predbat to automate charging my system at the lowest available agile slots each day. I am more interested in buying a days worth of electricity at the lowest price I can than I am in maximising my income, so I use metric_self_sufficiency set to +5p. This should weight against importing unless there's a big price difference.
I have best_soc_keep set to 3kWh because this will allow me to run the freezer for a while in the unlikely event of a power cut. I reduce this value to 2 or 1 when Agile prices are below 10p to allow import/export of more battery capacity. Please note best_soc_keep at 3kWh is about 15% of my available capacity.

Typically I can charge the battery to 100% SOC in the Agile overnight sessions (which tend to be cheaper anyway) and run for the rest of the day from battery, with enough reserve to pick the best slots from the following day rather than having to charge at midnight (this is also part of thinking behind best_soc_keep 3kWh).

The problem I see is a recent change in the behaviour of Predbat, whereby it charges to 100% SOC and then continues to charge for long periods of time at prices higher than I've already paid to fully charge the system. I understand the idea that the energy stored in my batteries is not "free" (although it has already been paid for). Including the inverter losses - say around 2p per kWh to be pessimistic - each kWh in my battery is worth 13p to export and 17p to use... so I disagree that running from grid at over 20p is cost effective. As far as I can tell, Predbat should be using the battery for every slot over 17p (ish) unless there is insufficient charge to meet the days needs and then we need to look at the lowest prices available to import enough to make it to the next day.

I also agree that Agile prices have been flat recently, but I've seen this charging anomaly after charging overnight at prices around 16p and the plan continued to "charge" at 22p and higher... The first screenshot was from one such occasion.

The second screenshot shows what I think is the same problem but might be something else -  from 00:30 onwards even though Predbat is setting the system to charge, it also has the system SOC decreasing (as I would expect on idle/demand). I've only seen this a couple of times, but when it happened I thought it odd enough to grab a screenshot.

If this isn't a bug and my excessive charging issue can be solved by changing my settings then I'm happy to take advice (this is the key reason I included the yaml file as it details my Predbat parameters). I have used the same set of parameters for a couple of months now, and they have worked fine, and sometimes still do. This issue is fairly recent so it's not something I've changed, but it could be something a recent update has corrected that's thrown my setup out of whack? Or it could be a bug... hence asking.

SO here is my current plan: I have set some forced charges overnight to make a single block to bring the battery up to 100% SOC. Predbat has already included another hour and a half of hold charge (updated to "charge" while writing this) that is not needed and is taking place at higher prices than later slots. I've included the debug file for this plan too.

[image]

[predbat_debug_19_15_00.yaml.txt](https://github.com/user-attachments/files/18263004/predbat_debug_19_15_00.yaml.txt)

gcoan, 2024-12-27:
It is quite difficult sometimes to work out why Predbat has come up with the plan it has.

> Including the inverter losses - say around 2p per kWh to be pessimistic - each kWh in my battery is worth 13p to export and 17p to use... so I disagree that running from grid at over 20p is cost [...]

You're correct that each kWh in your battery is roughly worth 13p when exported, I tend to think it's 13.5p, so same ballpark.

But the 'cost to use' isn't quite right and isn't a flat 17p as it depends on a couple of other factors: what it cost to fill the battery and what it will cost in the future to refill the battery.

If it cost you say 18p/kWh overnight to fill the battery  (which is about as good as I have had for the last couple of days) then with about 7% inverter and charging losses, each stored kWh of battery power actually cost 18*1.07=19.26p as as you will incur similar 7% losses on discharge you'll only get 0.93kWh out from the 1kWh of stored charge so it will in effect have cost you 19.26/0.93=20.7p to use it in the home.
A direct grid import at 20p is therefore still cost effective.

But also you have to look at what the cost will be to refill the battery in the future. If the overnight Agile rates are at best 18p then that's the target (plus losses) that Predbat has to consider when deciding whether to lower the battery SoC or not.

> I also agree that Agile prices have been flat recently, but I've seen this charging anomaly after charging overnight at prices around 16p and the plan continued to "charge" at 22p and higher... The first screenshot was from one such occasion.

I don't know if this is the case for you or not, but it may be that Predbat was holding the charge for later on peak rates based on what your forecast load was, although your half-hour load figures in the screenshot don't look particularly high.

Do you gave combine_charge_slots turned on?  This might be why you are getting very long charge/hold charge slots, might be worth turning this off so Predbat plans the slots individually.

> The second screenshot shows what I think is the same problem but might be something else - from 00:30 onwards even though Predbat is setting the system to charge, it also has the system SOC [...]

The second is weird I agree.  I have occasionally seen similar behaviour before and it was due to the way Predbat does its actual planning and how it summarises it in the HTML plan view.
Predbat plans at 5 minute increments but summarises them to 30 minute slots.  It's possible therefore that a 30 minute slot may contain a mixture of 5-minute long Demand and Charge/Freeze Charge activity which is why the SoC is dropping but the overall status is shown as Charging.
Would be worth seeing what Predbat actually executes.

This is a guess, I have seen it before, it may be something else though

Yuffi17, 2024-12-27:
I understand the conversion losses, but I maintain that the energy stored in my battery is only worth what I can sell it for (unless there's a power cut!) and therefore it's worth a net 15p regardless of what I paid for it. Running off the grid at 20p so I don't need to use 15p worth of electricity that I paid 18p for is not sensible in my mind. 

I think the key difference in our viewpoints is one of philosophy: I'm looking for minimal import costs rather than maximum import/export return.

This however doesn't change the fact that Predbat has started to add charging and hold charge slots where it did not previously. 
If this is caused by a bug then hopefully it will be possible to fix it. If it's due to my settings then I'm happy to be guided as the changes required to achieve what Predbat used to do.

Yuffi17, 2025-01-01:
With a bit of trial and error I have possibly narrowed the issue down to best_SOC_keep - although I don't know why. 
I usually run best_SOC_keep of 3kWh which is 16% of my system capacity. Here are two plans, created one after the other, today, the first has best_SOC_keep 3kWh and the second has best_SOC_keep 2kWh (about 11% capacity).

Both plans will stay well above the 3kWh (16%) soft limit. In fact, neither plan gets even close (SOC > 30% for both), and both plans hit 99% or more at the end of charging. Therefore the best_SOC_keep value should have no effect: "_best_soc_keep is the minimum battery level in kWh that Predbat will to try to keep the battery above for the Predbat plan. This is a soft constraint only that's used for longer term planning and is ignored for the forthcoming first 4 hours of the plan._".

However, you can clearly see that with a value of 3kWh, Predbat plans an extra charge at 22:30 (which makes sense) and then two hold charges at 06:00 and 06:30 which are not needed (as per my original post) - all of which push the projected cost up by 18p.

3kWh best_soc_keep plan: projected import cost £2.86
[image]

2kWh best_soc_keep plan: projected import cost £2.68
[image]

Yuffi17, 2025-01-02:
Two more screen grabs to illustrate unnecessary charging...

This is my system status from my HA home screen, and the Predbat plan. We were generating enough solar to cover the house load, and the battery is so full it's doing a little discharge, but Predbat is trying to charge from grid despite there being cheaper slots within the next 4 hours.... 

I believe this charging is unnecessary and I can't see how it makes economic sense, hence reporting this as a potential bug.

[image]

[image]

Yuffi17, 2025-01-03:
And another plan showing the same behaviour. There are two earlier charge slots that are cheaper, and the charge slot runs on for an unnecessary length of time at prices that should not be used.
[image]

I know this plan is more than 4 hours ahead and may change, but it does illustrate the problem clearly and I want to be able to trust Predbat to make sensible decisions while I'm asleep.

I'll stop posting at this point: there should be enough evidence here that something is not right. I'm happy to provide further information and I'm willing to try different settings if you think this will help.

Any help sorting this would be much appreciated.

gcoan, 2025-01-03:
I'm sure Trefor will look at this, but could you please post a logfile and debug file corresponding to one of these recent screen grabs

also, I'd suggest turning combine charge slots off, I think it is on as you are getting one long charge period and may be part of the problem

Yuffi17, 2025-01-03:
Thanks for the response. I haven't changed any of my settings since the first file (although, of course, the prices and the plan change). 

I already have combine charge slots turned off.

springfall2008, 2025-01-04:
Can you please re-test with the version on 'main' which appears to fix this? I'm still not sure exactly why.

[image]

Yuffi17, 2025-01-04:
Thanks. Main now installed and will check it after 4pm update, and in the morning, then provide feedback.

Yuffi17, 2025-01-04:
*** First impression is this is much better. ***
Here is Predbat's plan for tonight:
[image]

And here is what I would do myself:
[image]

The key difference is I'm charging to (almost) 100% whereas Predbat is charging to 67% (which is about a day's expected load). The Predbat version is the lowest anticipated cost, my version is a pessimistic version (keeping a significant reserve).

The important thing to note is that Predbat has NOT added any extra charge slots after my forced charges, which it was doing on the previous version. I'll do a further check in the morning (when previously Predbat would try to sneak in a couple of charging sessions that were not expected).

Yuffi17, 2025-01-05:
All looks good. No excess charging yesterday or today, with nice predictable behaviour: selecting the lowest priced slots and charging to meet the predicted demand for the day.

Thanks for the help!

Yuffi17, 2025-01-05:
Still looks good with todays price updates. I'm happy this is sorted.

Thanks!

closed 2025-01-06 by gcoan