springfall2008/batpred #1801: Document sensor.predbat_xx_N_idle/scheduled_discharge sensors and rationalise their use?

gcoan, 2024-12-27:
**Is your feature request related to a problem? Please describe.**
I have noticed three sensors for each of my inverters that are not documented in the 'Output Data' predbat documentation so I do not know what they do.

Moreover (and what led me to this) is that Predbat is writing a lot of state change records to the database for these sensors, so the database is filling up with unnecessary data:
[image]

predbat_ge_N_scheduled_discharge_enable (repeated for both of my inverters) appears to be writing two state change records every time that predbat runs, an 'on' and then an 'off', so taking up 200,000+ bytes of database storage that isn't useful.

**Describe the solution you'd like**
Either add to the documentation to explain what these sensors do, or let me know and I am happy to add it.

Consider whether _schedule_discharge_enable needs to be set for every Predbat run or not?  Maybe only write a state change when its actually needed.  Should it actually be a binary sensor?

Are these actually needed as Predbat output sensors?

**Describe alternatives you've considered**
In the meantime I am going to exclude these sensors from the recorder in configuration.yaml, but would be useful to improve for other Predbat users

springfall2008, 2024-12-28:
Interesting the double change is a mistake for sure, I should look at that.

These are essentially dummy inverter registers for the functions your inverter is marked as not supporting.

springfall2008, 2024-12-28:
Oddly mine never changes, I wonder if its a bug with dual inverter:

[image]

gcoan, 2024-12-28:
I think yours probably is changing Trefor, the last update time is shown as 36 seconds ago

But if you look at state history or the logbook, there is no history of changes on mine either, but in the HA database there is definitely a load of history being recorded !

In the Predbat log I can see the control being turned on and off:

```
2024-12-28 02:34:45.764488: Inverter 0 Wrote scheduled_discharge_enable to True successfully and got on
2024-12-28 02:34:45.764521: Inverter 0 sensor.predbat_GE_0_scheduled_discharge_enable set to True
2024-12-28 02:34:45.764572: Inverter 0 scheduled discharge enable is True
...
2024-12-28 02:34:45.800511: Inverter 1 scheduled charge enable is True
2024-12-28 02:34:45.800547: Inverter 1 charge windows currently [{'start': 90, 'end': 450, 'average': 0}, {'start': 1530, 'end': 1890, 'average': 0}, {'start': 2970, 'end': 3330, 'average': 0}]
2024-12-28 02:34:45.800581: Inverter 1 Charge settings: 12-28 01:30:00-12-28 07:30:00 limit 100.0 power 2.392 kW
2024-12-28 02:34:55.850375: Switch sensor.predbat_GE_1_scheduled_discharge_enable is now on
2024-12-28 02:34:55.850537: Inverter 1 Wrote scheduled_discharge_enable to True successfully and got on
2024-12-28 02:34:55.850566: Inverter 1 sensor.predbat_GE_1_scheduled_discharge_enable set to True
2024-12-28 02:34:55.850608: Inverter 1 scheduled discharge enable is True
...
2024-12-28 02:35:42.025215: Switch sensor.predbat_GE_0_scheduled_discharge_enable is now off
2024-12-28 02:35:42.025363: Inverter 0 Wrote scheduled_discharge_enable to False successfully and got off
```

[predbat.log](https://github.com/user-attachments/files/18266065/predbat.log)

springfall2008, 2024-12-28:
Looks like you are right, this is a recent bug related to the reduction in register writes change I made, will fix it

gcoan, 2024-12-28:
Is the bug fixed version on main?  I just installed it and seems to still be updating scheduled discharge enable:
[image]

[predbat.log](https://github.com/user-attachments/files/18266381/predbat.log)

springfall2008, 2024-12-28:
Should be but it was only merged 5 minutes ago....

[image]

gcoan, 2024-12-28:
OK I'll update to main again and see what it does

thanks

gcoan, 2024-12-28:
That's better, its not updated for 10 minutes now 
[image]

```
2024-12-28 14:15:02.828609: Inverter 0 scheduled discharge enable is False
...
2024-12-28 14:15:05.467725: Inverter 0 No write needed for scheduled_discharge_enable as False == False
```

What's the purpose of these three sensors, you mentioned that they are dummy's for features my inverter doesn't have.
I noticed that the idle end time is set to 23:59:00 and at the moment idle start time to 13:00:00 which doesn't correspond to any idling in my plan (its been charging or freeze charge all day).
I can't disable the sensors in HA as they don't have unique id's, but I can exclude them from the recorder db.
Is there any value in users seeing what these contain?

springfall2008, 2024-12-28:
With some inverters the values are used for automations to trigger charge/export via the service API.

In this case they aren't really needed by Predbat so you can exclude them from the history if you like.

BTW: Idle === Eco mode in this context

gcoan, 2024-12-28:
> With some inverters the values are used for automations to trigger charge/export via the service API.
> 
> 
> 
> In this case they aren't really needed by Predbat so you can exclude them from the history if you like.
> 
> 
> 
> BTW: Idle === Eco mode in this context

thanks I'll add something to the documentation