springfall2008/batpred #1802: PredBat not discharging overnight when Car is plugged in and fully charged

leacho73, 2024-12-28:
**Describe the bug**
I have noticed when our car is plugged in overnight, and even when its fully charged - PredBat never discharges anything to the grid, like it does when the car is disconnected - this seems to happen when our zappi is either in ' Waiting for EV' or 'Completed' state.

**Expected behaviour**
I would still expect the battery to discharge to grid to make use of cheaper electricity overnight as it does when the car is disconnected - I can normally discharge 3-4kwh to the grid overnight when disconnected, but 0kwh when the car is connected. Not sure if I am missing a setting somewhere to allow this, but I can't see one - I have 'Allow Car to Charge from Battery' turned off, however as the car will already be full when I am expecting the battery to discharge, I don't think I need this?

**Predbat version**

8.8.16 - but has been doing it for a while.

**Environment details**

- Solaredge Inverter and Battery
- HAOS
- Octopus Intelligent Go, using a Zappi Charger.

**Screenshots**
[image].
As you can see, no discharging is planned.

**Log file**
N/A

**Predbat debug yaml file**
N/A

springfall2008, 2024-12-28:
What does binary_sensor.predbat_car_charging_slot show in its attributes?

springfall2008, 2024-12-28:
Or could you grab the Predbat debug YAML so I can look?

leacho73, 2024-12-28:
> Or could you grab the Predbat debug YAML so I can look?

I've had to unplug the car at the moment but it will be plugged in tonight and fully charged tonight so I'll pop the debug file in tomorrow AM so it captures overnight if that's ok? Unless it has historical data - as it didn't do anything last night

leacho73, 2024-12-28:
> What does binary_sensor.predbat_car_charging_slot show in its attributes?

planned: 
- start: 12-28 12:00:00
  end: 12-28 12:30:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 12:30:00
  end: 12-28 13:00:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 13:00:00
  end: 12-28 13:30:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 13:30:00
  end: 12-28 14:00:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 14:00:00
  end: 12-28 14:30:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 14:30:00
  end: 12-28 15:00:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 15:00:00
  end: 12-28 15:30:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 15:30:00
  end: 12-28 16:00:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 20:30:00
  end: 12-28 21:00:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 21:30:00
  end: 12-28 22:00:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 22:00:00
  end: 12-28 22:30:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 22:30:00
  end: 12-28 23:00:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 23:00:00
  end: 12-28 23:30:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 23:30:00
  end: 12-29 00:00:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-29 00:00:00
  end: 12-29 00:30:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-29 00:30:00
  end: 12-29 01:00:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-29 01:00:00
  end: 12-29 01:30:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-29 01:30:00
  end: 12-29 02:00:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-29 02:00:00
  end: 12-29 02:30:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-29 02:30:00
  end: 12-29 03:00:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-29 03:00:00
  end: 12-29 03:30:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-29 03:30:00
  end: 12-29 04:00:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-29 04:00:00
  end: 12-29 04:05:00
  kwh: 0.12
  average: 7
  cost: 0.85

cost: 570.65
kwh: 81.52
friendly_name: Predbat car charging slot
icon: mdi:home-lightning-bolt-outline

springfall2008, 2024-12-28:
I'm a little confused as this is showing a plan to charge your car and hence Predbat can't export, but your snapshot of the plan didn't show data in the car column?

leacho73, 2024-12-28:
I've just generated a new charging plan by plugging the car back in, here are the new results:

planned: 
- start: 12-28 11:45:00
  end: 12-28 12:00:00
  kwh: 1.85
  average: 7
  cost: 12.95
- start: 12-28 12:00:00
  end: 12-28 12:30:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 12:30:00
  end: 12-28 13:00:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 13:00:00
  end: 12-28 13:30:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 13:30:00
  end: 12-28 14:00:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 14:00:00
  end: 12-28 14:30:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 14:30:00
  end: 12-28 15:00:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 15:00:00
  end: 12-28 15:30:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 15:30:00
  end: 12-28 16:00:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 21:30:00
  end: 12-28 22:00:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 22:00:00
  end: 12-28 22:30:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 22:30:00
  end: 12-28 23:00:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 23:00:00
  end: 12-28 23:30:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-28 23:30:00
  end: 12-29 00:00:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-29 00:00:00
  end: 12-29 00:30:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-29 00:30:00
  end: 12-29 01:00:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-29 01:00:00
  end: 12-29 01:30:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-29 01:30:00
  end: 12-29 02:00:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-29 02:00:00
  end: 12-29 02:30:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-29 02:30:00
  end: 12-29 03:00:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-29 03:00:00
  end: 12-29 03:30:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-29 03:30:00
  end: 12-29 04:00:00
  kwh: 3.7
  average: 7
  cost: 25.9
- start: 12-29 04:00:00
  end: 12-29 04:20:00
  kwh: 1.97
  average: 7
  cost: 13.8

cost: 570.65
kwh: 81.52
friendly_name: Predbat car charging slot
icon: mdi:home-lightning-bolt-outline

[image]

It appears to have updated itself - but the car will be sat plugged in, not requiring any more power in an hour or so - so unless I unplug it tonight, the battery won't discharge to grid - is there any way to allow this if the car doesn't require any more power, or is it because it still thinks a charging plan is active and thus is disabled?

springfall2008, 2024-12-28:
Why is Octopus reporting the car will charge for such a long period if it isn't going to?

leacho73, 2024-12-28:
The car isn't supported, so it talks to the Zappi rather than the car.

springfall2008, 2024-12-28:
If you have car SOC sensor and target charge % then I could probably have predbat decide the car is full and drop the slots as an option.

leacho73, 2024-12-28:
Yes I do have a SoC and Target sensor - that would be really helpful

springfall2008, 2024-12-30:
Potential fix on 'main' for testing.

leacho73, 2024-12-30:
Thanks @springfall2008 - do I need to make any changes to reference the SoC of the vehicle?

springfall2008, 2024-12-30:
You would need to make sure the car soc and car battery size are set correctly based on the car in apps.yaml. Although the limit can come from the Octopus Intelligent. Here's mine:
```
 car_charging_battery_size:
    - 75
  car_charging_soc:
    - 're:sensor.tsunami_battery'
  ```

leacho73, 2024-12-30:
Cool thanks - I'll take a look now - does this take into account if the vehicle is charged to an amount other than 100% - such as 80% which is recommended by some manufacturers?

springfall2008, 2024-12-30:
So there's also car_charging_limit which sets this threshold, but for me it also gets reported via Octopus Intelligent GO as I set it in their app.

leacho73, 2024-12-30:
Ill set this up now and report back in the morning if it has sorted the issue - thanks for the hard work @springfall2008

springfall2008, 2024-12-30:
Great. What I didn't fix was the Car KWh column on the HTML, that will still show what Octopus plans but in theory Predbat will detect the full car

leacho73, 2024-12-30:
[image]

Initial plan is looking promising - will report back in the morning

leacho73, 2024-12-31:
Morning Trefor - some mixed success I would say - although I had no export between 03:30 and 04:30 as the plan suggested, I did have 2kw of export from 23:00 until 00:00 - but nothing thereafter.

springfall2008, 2024-12-31:
Did you keep a logfile?

leacho73, 2024-12-31:
[predbat.1.log](https://github.com/user-attachments/files/18281365/predbat.1.log)
I did :)

leacho73, 2025-01-15:
Hi Trefor - I was just wondering if the change you made here is in the current release - as I'm not getting any export again.

[image]

Cheers
Leacho

cyberkryten, 2025-01-31:
> Great. What I didn't fix was the Car KWh column on the HTML, that will still show what Octopus plans but in theory Predbat will detect the full car

I have a similar setup - IOG controlling Zappi so it has no knowledge of car SOC and the 'Car kWh' always shows some odd values at times.  Usually max rate but with gaps at times and sometimes lower figures.
  
I figured it isn't used by Predbat but it is a bit confusing!

Here's the current view.  The car target is 90% SOC (though IOG is told 100% so it plans for that when I do need the full charge).  Car is currently plugged in and at 89% so there's no way it is getting 10kWh added tonight on the current settings - I think Predbat is assuming octopus_charge_limit (100%) is where the car will charge to as I don't think Predbat has adjusted for the car_charging_limit (90%) figure

I'm also not sure why it is showing the charge at the end of the IOG period - if the car does charge, it always does so at the start of the period (I wish it did it at the end but there's no way to tell IOG to do it that way).

[image]

leacho73, 2025-01-31:
I still have no discharging happening overnight for some reason - similar setup - Octopus IOG - car doesn't tell the zappi how full it is - IOG set to 100% but car set to 80%.

[image]

chalfontchubby, 2026-08-20:
Is this resolved for you now? Closing for now, but please do open a new ticket if there's still something going on - some of this can be tricky since Predbat can't always control things directly due to how Octopus/the car interact.

closed 2026-08-20 by chalfontchubby