springfall2008/batpred #1812: Unable to install addon

Amigos2222, 2024-12-28:
**Describe the bug**
Cannot fresh install through the addon menu in HA after adding the repository. When clicking install the icon spins but nothing else happens
Unsure if this is related to the update issue being reported yesterday

**Expected behaviour**
Addon to install

**Predbat version**

Latest
**Log file**
2024-12-28 19:38:24.118 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/springfall2008/predbat_addon repository
2024-12-28 19:40:18.541 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2024-12-28 19:41:38.099 INFO (MainThread) [supervisor.store] Loading add-ons from store: 91 all - 0 new - 0 remove
2024-12-28 19:41:38.100 INFO (MainThread) [supervisor.store] Loading add-ons from store: 91 all - 0 new - 0 remove
2024-12-28 20:10:19.032 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2024-12-28 20:13:05.256 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
2024-12-28 20:17:41.311 INFO (MainThread) [supervisor.addons.addon] No 6adb4f0d_predbat addon Docker image 6adb4f0d/armv7-addon-predbat found
2024-12-28 20:17:41.316 INFO (MainThread) [supervisor.docker.addon] Starting build for 6adb4f0d/armv7-addon-predbat:1.2.1

gcoan, 2024-12-28:
It can take quite a long time to install the predbat add-on, depending on the speed of your HA machine.
How long did you leave it?

Amigos2222, 2024-12-28:
Left it till it timed out with HA throwing an error on the addon install screen

gcoan, 2024-12-28:
Predbat probably took 20 minutes or so to install in the HAOS VM I'm running on a Windows i5 PC.

But looking at the logfile above, it took half an hour to just to download the HA add-on store list, that seems very very slow.

What specification machine are you running HA on?

Maybe giving it a clean restart of everything from scratch and it may start cleaner and quicker because those run times which are before it even gets to the Predbat install don't look promising

Amigos2222, 2024-12-28:
I tried removing a few addons i was not using and then restarted. Now i cant even get the supervisor to load so bigger problems. Going probably reinstall and see where we get to

Amigos2222, 2024-12-29:
Gave up and left it overnight. Come back to it just now and all has loaded up and the addon installed correctly! Not sure why maybe it all queued up when i tried multiple times to install and that slowed everything down?

closed 2024-12-29 by Amigos2222

gcoan, 2024-12-29:
> Gave up and left it overnight. Come back to it just now and all has loaded up and the addon installed correctly! Not sure why maybe it all queued up when i tried multiple times to install and that slowed everything down?

Weird stuff.  Glad it's resolved now.  Do follow the instructions for setting up Predbat and let us know if you have any questions