springfall2008/batpred #1814: Agile Pricing Incorrect from URL

joshwillcock, 2024-12-29:
**Describe the bug**
_When creating this issue the plan appeared to update itself. The *only* change made during this time was enabling and disabling debug mode. - I'm going to still share this issue incase somebody else has a similar issue._

rates_import_octopus_url appears to be incorrect/not updating reliably.

Expected to see 5:30 as 16.64 when batpred's plan is showing it as 21.67 (30% more or 5.03p)
Expected to see 6:00 at 17.19 when batpred's plan is showing it as 23.37 (35% more or 6.18p)

**Sample from the endpoint** [Octopus API Endpoint URL](https://api.octopus.energy/v1/products/AGILE-24-10-01/electricity-tariffs/E-1R-AGILE-24-10-01-H/standard-unit-rates/) Saved API Response: [apiresponse.txt](https://github.com/user-attachments/files/18268069/apiresponse.txt)
 {
      "value_exc_vat": 16.38,
      "value_inc_vat": 17.199,
      "valid_from": "2024-12-29T06:00:00Z",
      "valid_to": "2024-12-29T06:30:00Z",
      "payment_method": null
    },
    {
      "value_exc_vat": 15.85,
      "value_inc_vat": 16.6425,
      "valid_from": "2024-12-29T05:30:00Z",
      "valid_to": "2024-12-29T06:00:00Z",
      "payment_method": null
    },

**Batpred's Plan for that time:**

Sun 05:30 | 21.67 | 15.00 | ↘ |   |   | 0.18 | 56↘ | → | £2.15
Sun 06:00 | 23.37 | 15.00 | ↘ |   |   | 0.19 | 54↘ | → | £2.15

**Expected behaviour**
Batpred should be using the correct rates_import_octopus_url's values.

**Predbat version**

v8.8.16

**Environment details**
- FoxESS
- Docker
- Anything else?

**Screenshots**
Screenshot of URL setting:
[image]
Screenshot of plan which was incorrect:
[image]
Screenshot of plan which corrected itself after debug was enabled:
[image]

**Log file**
[predbat (1).log](https://github.com/user-attachments/files/18268085/predbat.1.log)

**Predbat debug yaml file**
When creating the debug yaml log run the plan updated and the times began showing correctly
[predbat_debug_00_25_00.yaml.txt](https://github.com/user-attachments/files/18268071/predbat_debug_00_25_00.yaml.txt)

springfall2008, 2024-12-29:
Are you sure the URL rate data wasn't just out of date and hence showing yesterdays? The URL only updates every 4 hours.

joshwillcock, 2024-12-29:
Hi Trefor,

This theoretically could have happened, but I would be quite surprised.

I don't believe we have the ability to see at what time the data from Octopus actually got released at, the apiresponse shows that it was correct at 12am (at the latest).

- I looked at the batpred plan (10pm, 11pm and 12am) which was showing the incorrect prices.
- I checked agileprices.co.uk & Octopus' API at 12:15am and the API was returning the correct data.
- I then enabled debug (for this issue) and then it updated correctly.

This means that batpred would have ben using the cached/predicted/old data for 10pm-12am 'live'. I would have thought it shouldn't be possible for the cache ttl of 4 hours to be longer than the future data window (given it releases at around 4PM & we are given data to 11:30pm the following day).
Worst case scenario, the data is released at 5PM, the 4 hour cache would expire at 5:05 (next run delay) + 9:05 (4 hour cache) + 9:10pm (next run delay), yet the data was not correct at 10pm for 10pm or 12am for 12am.

I'm going to keep an eye on what happens today/tomorrow, but I'd say it's more likely that Octopus' API was incorrect than the 4hr caching window hadn't expired, wouldn't you agree?

Thank you for your hard work on this **ridiculously cool** project!

joshwillcock, 2024-12-29:
Today's run:

4:08 Octopus API Wrong
4:08: Batpred Estimating 2:30am to be 18.12p
4:17 Octopus API Correct (7.10p)
4:17 Battered Estimating 2:30 to be 18.12p
4:37 Battered updated 2:30am (7.10p) Correct.

I'm going to close this ticket as I can't work out how we could reproduce it, and as much as we hate it, anything random can go wrong once... I'll keep monitoring this much closer over the next few weeks and will reopen if I can reproduce it accurately.

closed 2024-12-29 by joshwillcock

gcoan, 2024-12-29:
I'm fairly sure that somewhere in the Octopus Agile terms and conditions there is some wiggle room as to when they have to publish the next day prices.

I don't think it is guaranteed to be 4pm, but that doesn't necessarily answer what you saw.  Certainly the other day I saw tomorrow's prices released before 4pm, at about 3:20pm I think which I was quite surprised about.

It might have been that Octopus gave an error when Predbat tried to retrieve the next day prices and so continued to use the prior data.  I use the Octopus integration to give the prices to Predbat and I do see occasional 502 or other errors in the HA log where Octopus has rejected the connection.

Have you turned the norddata pricing prediction on?  This will give a more accurate prediction of what the next day prices are after about 11am in the morning, that then get overwritten with the actual prices when they come out.  This is better than the 'roll forward yesterday prices' that Predbat does by default

joshwillcock, 2024-12-29:
> I'm fairly sure that somewhere in the Octopus Agile terms and conditions there is some wiggle room as to when they have to publish the next day prices.
>
> I don't think it is guaranteed to be 4pm, but [...]

I haven't explored norddata - something I'll check out now, although better predictions is good, I'm going to keep an eye on what happened, if it happens again I'll dive into the debug options!